letscontrolit/ESPEasy #1898: Modify analog plugin for piezo disc

Sammii203, 2018-10-13:
### Summarize of the problem/feature request
I can't get values of piezo disc with "Analog input - internal" plugin.

### Expected behavior
- Give the "highest" value between two poll of the analog plugin
   Add checkbox in "data acquisition" section on the configuration of "Analog input - internal" for specify if we want the highest value between two pools.
- Change data interval acquisition in milliseconds instead of seconds

### Actual behavior
I have a problem, because Piezo don't detect all pressures and knocks near it.
And without pressure or knocks, values are always between 10 and 20.

### Steps to reproduce
I explain my problem in [this post](http://www.letscontrolit.com/forum/viewtopic.php?f=2&t=5922&sid=5aece6f158338dbbb25e67854463bdd8)

### System configuration
Hardware : Adafruit HUZZAH ESP8266 breakout
ESP Easy version: ESP_Easy_mega-20181006_normal_ESP8266_4096.bin
ESP Easy settings/screenshots: 
[image]

TD-er, 2018-10-13:
I am not sure how we can do this on the analog pin.
Sampling of the analog input isn't fast enough to do this.

Maybe @ShardanX can explain it better?

Sammii203, 2018-10-14:
I don't understand; it's firmware or hardware problem ?
It seems to be feasible If I read [this article](http://designinformaticslab.github.io/productdesign_tutorial/2017/01/24/piezoelectric_sensor.html)

Grovkillen, 2018-10-14:
I believe they use the computer for the real time reading? Or why are they having it connected?

uzi18, 2018-10-14:
Did you connect MAE540 like in the article (with 1MOhm resistor)?
Code for blynk only reads analog input, convert and send it to blynk, it's simple job.
ESPEasy is more busy here.

TD-er, 2018-10-14:
The linked article uses a Piezo sensor with mass, which means there is some weight glued to it.
This weight does:
- Act as some kind of threshold for vibrations, a minimal force is required to make it vibrate
- Act as a filter, only lower frequencies will be giving enough output signal.
- Cause multiple vibration periods, giving the ESP more time to record the higher amplitude value.

Without such a weight, you have to sample at quite a large sample rate, or include some sample-and-hold circuit to keep the higher output values long enough for the ESP to sample it.

When using just a piezo element, you also have to protect the input of the ESP, since a piezo element can generate quite large voltages of over 50V.

tonhuisman, 2023-09-03:
This issue is probably no longer relevant, and can be closed.

closed 2023-09-04 by TD-er