letscontrolit/ESPEasy #1903: Saving controller Generic HTTP Advanced give reboot

glsf91, 2018-10-14:
I flashed ESP_Easy_mega-20181014_test_ESP8266_4096.bin.
Setup to connect to Wifi.
Then add generic HTTP Advanced controller. When adding ip-address and setting port, click submit.
Then the esp8266 will crash/reboot.

I'am also using ESP_Easy_v2.0-20180307_dev_ESP8266_4096.bin which don't have this problem.
I have also use some older versions then 20181014 which has the same problem.

Also erase complete esp8266 before or reset factory has no effect.

```
INIT : Booting version: mega-20181014 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2)
86 : INIT : Warm boot #8 - Restart Reason: Software/System restart
88 : FS   : Mounting...
111 : FS   : Mount successful, used 75802 bytes of 957314
430 : CRC  : program checksum       ...OK
436 : CRC  : SecuritySettings CRC   ...OK 
536 : INIT : Free RAM:21880
537 : INIT : I2C
537 : INIT : SPI not enabled
550 : INFO : Plugins: 72 [Normal] [Testing] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2)
552 : WIFI : No valid wifi settings
553 : WIFI : Could not connect to AP!
554 : WIFI : Set WiFi to AP
1472 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
2806 : WD   : Uptime 0 ConnectFailures 0 FreeMem 17496

Panic /home/travis/.platformio/packages/framework-arduinoespressif8266@1.20401.2/cores/esp8266/core_esp8266_main.cpp:99 __yield

ctx: sys 
sp: 3fff3840 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fff39f0:  3fff19fc 3fff3014 3fff1a24 4026867d  
3fff3a00:  3ffe92be 00000000 00002b58 40269528  
3fff3a10:  3fff19fc 3fff3014 00002b58 4022bbcb  
3fff3a20:  3ffe92be 3ffe9323 3fff1a24 40268877  
3fff3a30:  3fff3b00 3ffe9323 3fff3e54 402bd15a  
3fff3a40:  3fff1a24 0001c768 0001c768 00000000  
3fff3a50:  00000070 00000000 00000000 402bd15b  
3fff3a60:  3fff1a24 3fff3aa0 3fff19fc 4022dafd  
3fff3a70:  402bd1b1 00000001 3fff1b98 3fff3dec  
3fff3a80:  3fff436c 00000c1e 00000c1e 3fff3dec  
3fff3a90:  3fff2a44 3fff3e54 3fff19fc 4022dc32  
3fff3aa0:  00000400 00007000 3fff1dcc 40268410  
3fff3ab0:  3ffe9323 3fff3b00 402b4e4d 40218e0a  
3fff3ac0:  00000000 00000000 00000000 3fff3dec  
3fff3ad0:  00000000 3fff3b00 3fff3e54 4023a796  
3fff3ae0:  3fff2a44 3fff3e54 00000000 4024abae  
3fff3af0:  3fff22a4 3fff1ebc 3fff3f9c 40268410  
3fff3b00:  01a8c000 0000009f 00001f96 00000000  
3fff3b10:  00000000 00000000 00000000 00000000  
3fff3b20:  00000000 00000000 00000000 00000000  
3fff3b30:  00000000 00000000 00000000 00000000  
3fff3b40:  00000000 00000000 00000000 00000000  
3fff3b50:  00000000 00000000 00000000 00000000  
3fff3b60:  00000000 00000000 00000000 00000000  
3fff3b70:  00000000 00000000 00000000 00000000  
3fff3b80:  00000000 00000000 00000000 00000000  
3fff3b90:  00000000 00000000 00000000 00000000  
3fff3ba0:  00000000 00000000 00000000 00000000  
3fff3bb0:  00000000 00000000 00000000 00000000  
3fff3bc0:  00000000 00000000 00000000 00000000  
3fff3bd0:  00000000 00000000 00000000 00000000  
3fff3be0:  00000000 00000000 00000000 00000000  
3fff3bf0:  00000000 00000000 00000000 00000000  
3fff3c00:  00000000 00000000 00000000 00000000  
3fff3c10:  00000000 00000000 00000000 00000000  
3fff3c20:  00000000 00000000 00000000 00000000  
3fff3c30:  00000000 00000000 00000000 00000000  
[... 952 lines cut ...]

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
⸮U86 : 

INIT : Booting version: mega-20181014 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2)
86 : INIT : Warm boot #9 - Restart Reason: Software Watchdog
88 : FS   : Mounting...
111 : FS   : Mount successful, used 75802 bytes of 957314
431 : CRC  : program checksum       ...OK
461 : CRC  : SecuritySettings CRC   ...OK 
561 : INIT : Free RAM:21880
562 : INIT : I2C
562 : INIT : SPI not enabled
575 : INFO : Plugins: 72 [Normal] [Testing] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2)
577 : WIFI : Set WiFi to STA
608 : WIFI : Connecting NG24 attempt #0
1942 : WD   : Uptime 0 ConnectFailures 0 FreeMem 18784
4390 : WIFI : Connected! AP: NG24 (BC:EE:7B:34:72:D8) Ch: 1 Duration: 3781 ms
```

s0170071, 2018-10-14:
The stack dump is quite long... I guess the stack was too full when spiffs was accessed during save.
I suggest to check for at least 600 bytes of free stack before save and display a stack full warning otherwise. This way we get a decent hint on the source of the error. Otherwise it will happen on so many different occasions that we will hit the github issue limit ;-)

TD-er, 2018-10-14:
@s0170071 That issue limit is something we can hand over to others ;)
But I agree on adding stack checks in save routines.
At least because crashes during save may corrupt settings.

TD-er, 2018-10-15:
In the last build, I added a few delay() statements during SPIFFS interaction.
Can you test if it is still crashing?
Last night I also had a number of crashes on my node (before adding the delay statements) when saving something.

It just seems like there is something going on in the back which sometimes may take a lot more time when saving/loading files on SPIFFS.

s0170071, 2018-10-15:
see #1906. That may fix this as well.

glsf91, 2018-10-15:
> In the last build, I added a few delay() statements during SPIFFS interaction.
> Can you test if it is still crashing?
> Last night I also had a number of crashes on my node (before adding the delay [...]

Should it be in ESPEasy_mega-20181015.zip on https://github.com/letscontrolit/ESPEasy/releases?

TD-er, 2018-10-20:
Could you test if this is still an issue, or is it fixed since this PR got merged: https://github.com/letscontrolit/ESPEasy/issues/1906
It was merged on Okt 16. so it must be included since the 17th.

glsf91, 2018-10-21:
I used ESP_Easy_mega-20181021_dev_ESP8266_4096.bin for testing.
Now I don't see the problem anymore.
Thanks.

closed 2018-10-21 by glsf91