letscontrolit/ESPEasy #1904: Framed Oled Display mixing up multiple values in different lines

Wiki591, 2018-10-14:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Displaying readings of different sensors in one line works. But if the following line refers to different readings, too, the lines are mixed up.
Example: Filling the lines as shown ans submitting them

[image]

Returns the mixed up lines as shown below:

[image]

The values are shown on the display but mixed exactly as in the configuration shown.

### Expected behavior

The lines should stay as they are before submitting

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1.  Fill in different sensor values in two line 
2.  Submit the configuration
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-10-14:
Can you see if the ESP node has crashed/rebooted during saving of the settings?
It looks like it stopped writing during the save action and as result the data got corrupted.

We're experiencing a lot of non-reproducible errors lately and it looks like there is some data corrupted during operation.

I've seen crashes during save of this plugin, but those were on ESP32.

v-a-d-e-r, 2018-10-14:
You are using 'H' for humidity, so take 'T' for temperature. Done.

TD-er, 2018-10-14:
I'm now making a `safe_strncpy` function and will apply it to all locations where similar copies are being made.
Just to be sure the same issue cannot corrupt other settings.

Wiki591, 2018-10-14:
Hi,
I renamed '°C' to 'C' and everything works fine, problem - for me - solved. Thank you!

closed 2018-10-14 by Wiki591