letscontrolit/ESPEasy #1910: Notification settings extra strings on stack

s0170071, 2018-10-15:
in webserver.ino, handle_notification() there is 

 ```
 String domain = WebServer.arg(F("domain"));
  String server = WebServer.arg(F("server"));
  String sender = WebServer.arg(F("sender"));
  String receiver = WebServer.arg(F("receiver"));
  String subject = WebServer.arg(F("subject"));
  String user = WebServer.arg(F("user"));
  String pass = WebServer.arg(F("pass"));
  String body = WebServer.arg(F("body"));
  String notificationenabled = WebServer.arg(F("notificationenabled"));

```

those strings are only used once more and are not manipulated in between. 

```     strncpy(NotificationSettings.Domain, domain.c_str(), sizeof(NotificationSettings.Domain));
        strncpy(NotificationSettings.Server, server.c_str(), sizeof(NotificationSettings.Server));
        strncpy(NotificationSettings.Sender, sender.c_str(), sizeof(NotificationSettings.Sender));
        strncpy(NotificationSettings.Receiver, receiver.c_str(), sizeof(NotificationSettings.Receiver));
        strncpy(NotificationSettings.Subject, subject.c_str(), sizeof(NotificationSettings.Subject));
        strncpy(NotificationSettings.User, user.c_str(), sizeof(NotificationSettings.User));
        strncpy(NotificationSettings.Pass, pass.c_str(), sizeof(NotificationSettings.Pass));
        strncpy(NotificationSettings.Body, body.c_str(), sizeof(NotificationSettings.Body));
```
Instead of copying the args to strings, WebServer.arg should be used directly.

TD-er, 2018-10-15:
I am not entirely sure the webserver argument will remain constant.
If so, then there must already be a deep copy available and then you're absolutely right.
In Webserver.ino a lot of variables can then be removed.

But first we have to look into the WebServer.arg() call, to see what it does and how long the data is valid.

s0170071, 2018-10-16:
There is no way I can think of for the argument to become invalid. There are no system calls between the arg() to String and the String to struct commands. I did a google search and found no indication about it either.

TD-er, 2018-10-19:
Will be fixed when #1923 is merged
Maybe you could do some tests on that PR before I merge it?

s0170071, 2018-10-20:
Will do this weekend and let you know.

TD-er, 2019-10-10:
The Webserver code has been changed a lot since this issue, so this one is no longer valid.

closed 2019-10-10 by TD-er