letscontrolit/ESPEasy #1912: MQTT Import payload

redskinhu, 2018-10-15:
### Summarize of the problem/feature request

MQTT Import payload problem 

### Expected behavior

I expected to get the sent payload value.

### Actual behavior

I got "Illegal payload" message. The import add some string to the original payload (it looks its get from the Rules?). And after i tried other values (22, 333, 4444, 55555, 666666, 000000, 1)  i got unexpected values. See the Log.

### Steps to reproduce

1.  Create MQTT Import
2. Send a payload value.

I tried to reinstall / reset to factory default / erase and rewrite the full flash.

### System configuration

Hardware: ESP12F

ESP Easy version:  ESP_Easy_mega-20181015_test_ESP8266_4096_VCC
Earlier versions are fine.

### Rules or log data
Log:

37705: IMPT : Bad Import MQTT Command home/Livingroom_ESP07/PWM/14/oo_command
37706: ERR : Illegal Payload 1mand MQTT_Import

Real payload: 1
"mand": text fragment from the rules

42205: IMPT : Bad Import MQTT Command home/Livingroom_ESP07/PWM/14/oo_command
42206: ERR : Illegal Payload 22and MQTT_Import

Real payload: 22

47505: IMPT : Bad Import MQTT Command home/Livingroom_ESP07/PWM/14/oo_command
47506: ERR : Illegal Payload 333nd MQTT_Import

Real payload: 333

52805: IMPT : Bad Import MQTT Command home/Livingroom_ESP07/PWM/14/oo_command
52806: ERR : Illegal Payload 4444d MQTT_Import

Real payload: 4444

58613: IMPT : [MQTT_Import#PWM_14_onoff_command] : 55555.00
58615: EVENT: MQTT_Import#PWM_14_onoff_command=55555.00

Real payload: 55555

67613: IMPT : [MQTT_Import#PWM_14_onoff_command] : 666666.00
67615: EVENT: MQTT_Import#PWM_14_onoff_command=666666.00

Real payload: 666666

**_75213: IMPT : [MQTT_Import#PWM_14_onoff_command] : 166666.00
75214: EVENT: MQTT_Import#PWM_14_onoff_command=166666.00

Real payload: 1_**

120513: IMPT : [MQTT_Import#PWM_14_onoff_command] : 0.00
120514: EVENT: MQTT_Import#PWM_14_onoff_command=0.00

Real payload: 000000

129718: IMPT : [MQTT_Import#PWM_14_onoff_command] : 100000.00
129720: EVENT: MQTT_Import#PWM_14_onoff_command=100000.00

Real payload: 1

Thanks
T

TD-er, 2018-10-15:
I guess something is wrong here with the safe_strncpy function I introduced yesterday.
I will revert it.

TD-er, 2018-10-16:
Has it been fixed now?

redskinhu, 2018-10-16:
Mega-20181016 is ok.

TD-er, 2018-10-17:
Thanks for replying

closed 2018-10-17 by TD-er