letscontrolit/ESPEasy #1915: Formulas don't work %value%+100

evg26, 2018-10-16:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

evg26, 2018-10-16:
mega-20181016

evg26, 2018-10-16:
Generic - Pulse counter

TD-er, 2018-10-16:
> mega-20181016

What version was the last one you know that was still working?

evg26, 2018-10-16:
No information

evg26, 2018-10-16:
3 days testing

evg26, 2018-10-16:
[image]

TD-er, 2018-10-16:
Does it show the expected value on the Devices tab?

evg26, 2018-10-16:
pulse count only, formula does not work

evg26, 2018-10-16:
[image]

TD-er, 2018-10-16:
Just as an test, can you try `100.0 + %value%`
 ?

evg26, 2018-10-16:
100.0+%value% It's not working

evg26, 2018-10-16:
Since the power outage may be short in time and the meter changes readings rarely, there is a need to keep the last reading.Save after changing the reading or by time.

evg26, 2018-10-16:
Корректировка показаний счетчиков желательно иметь возможность с сервера. Очень актуально для счетчиков с импульсным выходом.

evg26, 2018-10-16:
Correction of meter readings is desirable to be able to the server. It is very important for counters with pulse output.

evg26, 2018-10-16:
mega-20180801 It's not working

evg26, 2018-10-17:
mega-20180506 Formulas don't work %value%+100 to Pulse Counter

evg26, 2018-10-17:
#1876 Pulse Counter Formula error