letscontrolit/ESPEasy #1916: Webif not loading/unresponsive in Mega

adrianmihalko, 2018-10-16:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I am experiencing a problem with Web interface that is not loading/unresponsive after 1-2 click on the interface. Until now I tried mega 20181016 on Sonoff with 1MB Flash and 20181011 on Wemos D1 with 4MB flash. Same problem.

Maybe it is a memory -RAM- problem on the ESP8266? Maybe Mega it is too big for these device?

Video:

https://youtu.be/LyGrUvNz0wc

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-10-16:
Can you please give a bit more information:

- ESP Easy version (normal/test/dev)
- ESP Easy settings (plugin/crontroller active)
- Free memory

adrianmihalko, 2018-10-16:
Normal version.

ESP Easy settings (plugin/crontroller active) - right after first boot, with any plugin/controller added.

```
Unit:	0
GIT version:	mega-20181016
Local Time:	NTP disabled
Uptime:	0 days 0 hours 19 minutes
Load:	54.60% (LC=10545)
Free Mem:	4520 (3688 - sendContentBlocking)
Free Stack:	3520 (1168 - sendContentBlocking)
IP:	192.168.1.81
Wifi RSSI:	-49 dB (xy)
```

TD-er, 2018-10-16:
Hmm that amount of free memory is not what I see when using the 'normal' builds here.
Are you sure this is the "normal" build? You can see more on the sysinfo page.
When the memory is almost depleted, like in your setup, the web interface will be very slow, since it is actively trying to free some memory in the background.

Can you also check if the (serial) log level is set to none?

adrianmihalko, 2018-10-16:
Here it is:

```

Firmware
--
Build | 20102 - Mega
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2
GIT version | mega-20181016
Plugins | 46 [Normal]
Build Md5 | 2c4c42e8a02c836a57bafb44255584
Md5 check | passed.
Build time | Oct 16 2018 02:20:02
Binary filename | ESP_Easy_mega-20181016_normal_ESP8266_1024.bin
```

Serial log Level: None

TD-er, 2018-10-16:
I see.
This is clearly not normal, or at least not how it should be.

adrianmihalko, 2018-10-16:
I restarted, now I see more Free Memory:

[image]

...but it's still unusable. All tried all browsers: Safari, Chrome, Firefox. Mac + Windows.

TD-er, 2018-10-17:
The latest nightly build does change memory allocation while loading settings.
Can you test if that will help?
Not really sure what's going on with your setup, since it shouldn't be unusable slow

uzi18, 2018-10-17:
try fo disable antivirus software for 10minutes and reload page

adrianmihalko, 2018-10-17:
Hi, I have no antivirus on my Mac.

adrianmihalko, 2018-10-17:
I am trying on another Sonoff S20. I started from v2.0.0-dev11:

[image]

I tried to upgrade with built in upgrade tool, but upgrade fails  in this case & not booting anymore so I must manually flash. First I flashed with blank 1MB FW then with normal.

Right after first boot:

[image]

I am clicking on the menus from left to right, first it stopped working at Tools menu. Waited few minutes, refreshed the page then finally got the menu.

I clicked on System Info, blank screen, waiting. Clicking on address bar in the browser, then enter -> page loads finally. 

```
Build
20102  - Mega
Libraries
ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2
GIT version
mega-20181017
Plugins
46 [Normal]
Build time
Oct 17 2018 02:17:59
Binary filename
ESP_Easy_mega-20181017_normal_ESP8266_1024.bin

Unit | 0
-- | --
Uptime | 0 days 0 hours 5 minutes
Load | 3.00% (LC=13307)
Free Mem | 10584 (4968 - sendContentBlocking)
Free Stack | 3584 (1584 - sendContentBlocking)
Boot | Cold boot (0)
Reset Reason | Power on

```

You can see how slow it is:

https://youtu.be/C1wrHY8eN6E

TD-er, 2018-10-17:
Can you see if it is maybe crashing/rebooting between click and the loading of the page?
And please can you give the flash ID?
Can you store any settings ?

adrianmihalko, 2018-10-17:
No crash/reboot when waiting for page load.

[image]

```

Firmware
--
Build | 20102 - Mega
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2
GIT version | mega-20181017
Plugins | 46 [Normal]
Build Md5 | edae249585968ef1d457f96d2647f74
Md5 check | passed.
Build time | Oct 17 2018 02:17:59
Binary filename | ESP_Easy_mega-20181017_normal_ESP8266_1024.bin
System Status
Syslog Log Level: | None
Serial Log Level: | None
Web Log Level: | None
ESP board
ESP Chip ID | 1951683 (0x1DC7C3)
ESP Chip Freq: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID | Vendor: 0xE0 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DOUT
Flash Writes | 0 daily / 0 boot
Sketch Size | 742 kB (132 kB free)
SPIFFS Size | 110 kB (36 kB free)

```

TD-er, 2018-10-17:
Changing settings are stored correct on that node?

adrianmihalko, 2018-10-17:
Yes, they are (hardware settings, device settings, controller, rules).

vteh01, 2018-10-22:
Same thing with latest build. This problem has appeared after build 20181003.

TD-er, 2018-10-22:
Have you tried using another computer to access the ESP node?
Perhaps a phone?

Are the ESP and the computer you use on the same network, or is there some kind of router in-between?
For example do you share the same subnet? (e.g. 192.168.1.x / 255.255.255.0 )

adrianmihalko, 2018-10-22:
I tried everything and not working. I flashed Tasmota, Espurna on the weekend and both working fine on the same device.

s0170071, 2018-10-24:
@ adrianmihalko: Your sketch/spiffs sizes are different from my S20

I use the S20 with these Arduino IFE settings: 
Spiffs size 256MB
Module: D1 mini lite
Lwip 2 low BW
Core 2.4.2
46 Plugins, version from Oct 16 
running for 7 days  now, not a single reboot.

Uptime: | 7 days 7 hours 44 minutes
Free Mem: | 18144 (15824 - sendContentBlocking)
Free Stack: | 3520 (1348 - rulesProcessing)

adrianmihalko, 2018-10-24:
I use pre built binaries.

TD-er, 2018-10-24:
@adrianmihalko The pre-built binary determines the SPIFFS size.
For 1M modules it is 128k SPIFFS if I remember correctly.

For these 1M modules, I think we should go back to core 2.4.2 and LWIP 2.
These are not likely to use PWM for applications where PWM stability (low jitter) is important.

closed 2019-10-11 by Grovkillen