letscontrolit/ESPEasy #1917: Bug in _P005 DHT

v-a-d-e-r, 2018-10-17:
2 times switch 'Par3' with different arguments:

Perform the actual reading + interpreting of data.
......
  switch (Par3) {
    case P005_DHT11:
    case P005_DHT22:
    case P005_DHT12:  delay(18); break;  // FIXME TD-er: Must this be so long?
    case P005_AM2301: delayMicroseconds(900); break;
    case P005_SI7021: delayMicroseconds(500); break;
  }
  switch (Par3) {
    case P005_DHT11:
    case P005_DHT22:
    case P005_DHT12:
    case P005_AM2301:
      pinMode(Plugin_005_DHT_Pin, INPUT);
      delayMicroseconds(50);
      break;
.....

TD-er, 2018-10-17:
The only change in the code is that for the Si7021 it no longer does the "change pin to input" and "delay 50 usec".

For that one, the next case is being used:

```C++
    case P005_SI7021:
      // See: https://github.com/letscontrolit/ESPEasy/issues/1798
      digitalWrite(Plugin_005_DHT_Pin, HIGH);
      delayMicroseconds(20);
      pinMode(Plugin_005_DHT_Pin, INPUT);
      break;
  }
```

I changed it to switch statements to make it easier to read. If...else...else... statements are very hard to read.
See: https://github.com/letscontrolit/ESPEasy/issues/1798 for a description of the requested changes.

Does this plugin actually behave buggy, or is it a comment on suspected code to pinpoint something that's very likely to be wrong?

v-a-d-e-r, 2018-10-17:
It looks just a bit senseless for me, 2 switch (Par3) after each other. Why not in one block?

switch (Par3) {
case P005_DHT11:
case P005_DHT22:
case P005_DHT12: delay(18); break; // FIXME TD-er: Must this be so long?
case P005_AM2301: delayMicroseconds(900); pinMode(Plugin_005_DHT_Pin, INPUT); delayMicroseconds(50); break;
case P005_SI7021: delayMicroseconds(500); break;
}

TD-er, 2018-10-17:
Your block example is different.
A switch statement in which the cases do not end with a `break;`, all use the same code until the `break;` statement.

```C++
  switch (Par3) {
    case P005_DHT11:
    case P005_DHT22:
    case P005_DHT12:  delay(18); break;  // FIXME TD-er: Must this be so long?
    case P005_AM2301: delayMicroseconds(900); break;
    case P005_SI7021: delayMicroseconds(500); break;
  }
```
means:
```C++
if (Par3 == P005_DHT11 || Par3 == P005_DHT22 || Par3 == P005_DHT12) {
  delay(18);
} else if (Par3 == P005_AM2301) {
  delayMicroseconds(900); 
} else if (Par3 == P005_SI7021) {
  delayMicroseconds(500); 
}
```

In the current implementation of P005, there is a difference on how `P005_AM2301` is handled and the `P005_DHTxx`
In your given example block, the pinmode is not being changed for the `P005_DHTxx` modules and in my 2 switch statements it is.

I use switch statements a lot to make complex if-then-else constructions readable, since it is easier to see what is the same and what differs.

v-a-d-e-r, 2018-10-17:
I made it in one block for me and all works. :-) BTW: The default delay for DHT22 is 1ms (Range: 0.8-20ms) and works here with 5 sensors set with 1ms without any issue...

TD-er, 2018-10-17:
Can you post what has to change?
It can be done in one block, but either it is not the same code, or it is less readable.

So if you have some changes which work for DHT22, and do not include the long delay of 18 msec, please share it.

v-a-d-e-r, 2018-10-17:
My code block is this (all data are default (typical) from data sheet):

  switch (Par3) {
    case P005_DHT11: delayMicroseconds(80); pinMode(Plugin_005_DHT_Pin, INPUT); delayMicroseconds(80); break;
    case P005_DHT12: delayMicroseconds(80); pinMode(Plugin_005_DHT_Pin, INPUT); delayMicroseconds(80); break;
    case P005_DHT22: delay(1); pinMode(Plugin_005_DHT_Pin, INPUT); delayMicroseconds(30); break;
    case P005_AM2301: delay(1); pinMode(Plugin_005_DHT_Pin, INPUT); delayMicroseconds(30); break;
    case P005_SI7021: delayMicroseconds(500); digitalWrite(Plugin_005_DHT_Pin, HIGH); delayMicroseconds(20); pinMode(Plugin_005_DHT_Pin, INPUT); break;
  }

TD-er, 2018-10-17:
So your code can be stripped down to:

```C++
switch (Par3) {
  case P005_DHT11: 
  case P005_DHT12: 
    delayMicroseconds(80); 
    pinMode(Plugin_005_DHT_Pin, INPUT); 
    delayMicroseconds(80); 
    break;
  case P005_DHT22: 
  case P005_AM2301:
    delay(1); 
    pinMode(Plugin_005_DHT_Pin, INPUT); 
    delayMicroseconds(30); 
    break;
  case P005_SI7021: 
    delayMicroseconds(500); 
    digitalWrite(Plugin_005_DHT_Pin, HIGH); 
    delayMicroseconds(20); 
    pinMode(Plugin_005_DHT_Pin, INPUT); 
    break;
}
```

In this switch statement the behavior of   P005_DHT22 &  P005_AM2301  are now the same, and thus it makes sense to have it in one switch statement.
For the current implementation of P005 these differ in 2 stages of the process, so that's why I split it into 2 switch statements.

TD-er, 2018-10-17:
It is a matter of taste about the newlines or oneliners. (I have multiple columns of source code next to eachother on my screens)
But at least you can delete half of the lines you have, since the whole idea of switch statements is that they allow for a fall-through when you don't insert a break.
This allows to see very easily if the same code can be executed for several cases.

And I really hate code duplication, so I will do my best to avoid code duplication and still keep it readable. 

So the question remains, is the current code incorrect and should it be changed?

v-a-d-e-r, 2018-10-17:
For me you can keep it as you want. Let's see what other self-compilers mean to that new structure....

TD-er, 2018-10-17:
Is it a bug?
I really want to get the number of issues down, since we're having close to 500 issues open and just leaving issues open to see what someone else will think of it is just clutter.

v-a-d-e-r, 2018-10-17:
Maybe 'bug' was not the right word for that. It looked a bit curious to me. So let's close this 'issue' and we have one less. ;-)

closed 2018-10-17 by v-a-d-e-r