letscontrolit/ESPEasy #1920: CSE7766  strange low voltage

michass, 2018-10-19:
I have Sonoff POW R2 to control home ventilation sytem with domoticz. 
I use ROM:
ESP_Easy_mega-20181017_dev_ESP8266_4096

Time to time I have:
[image]
[image]

I know, when i go to ESPEasy control panel -> Devices -> CSE7766 and submit, then everything go back to normal.
How to repair this issue?
Best Regards
Michał

uzi18, 2018-10-19:
try to catch error/debug log, so. we can find something

uzi18, 2018-10-19:
@TD-er looks like serial communication problems

TD-er, 2018-10-19:
The number looks like the power factor, not voltage.
So indeed issues with serial communication

michass, 2018-10-19:
It is soft issue or this unit hardware problem or maybe who knows?

uzi18, 2018-10-19:
@TD-er maybe controller sends wrong values?

michass, 2018-10-19:
Controller you mean Sonof device?
In Device tab parameters are also wrong

michass, 2018-10-19:
[debug.txt](https://github.com/letscontrolit/ESPEasy/files/2497125/debug.txt)
My log

uzi18, 2018-10-19:
@TD-er 

8692145: CSE: packet received
8692145: CSE: Checksum Failure
8692147: CSE: Header received
8692248: CSE: packet received
8692249: CSE: Checksum Failure
8692250: CSE: Header received
8692333: sendcontent free: 10920 chunk size:400
8692338: sendcontent free: 9536 chunk size:353
8692341: sendcontent free: 9536 chunk size:0
8692353: CSE: packet received
8692354: CSE: Checksum Failure
8692355: CSE: Header received
8692458: CSE: packet received
8692459: CSE: Checksum Failure
8692462: CSE: Header received
8692561: sendcontent free: 10920 chunk size:400
8692746: CSE: plugin read
8692749: EVENT: PoborWent#Voltage=0.92
8692774: CSE: packet received
8692775: CSE: Checksum Failure
8692777: EVENT: PoborWent#Voltage=0.92 Processing time:28 milliSeconds
8692778: CSE: Header received
8692780: EVENT: PoborWent#Power=0.00
8692812: DEBUG DEV: Parsed String='SendToHTTP 192.168.0.109,8084,/json.htm?type=commandśm=udevice&idx=78&nvalue=0&svalue=0.92'
8692816: ACT  : SendToHTTP 192.168.0.109,8084,/json.htm?type=commandśm=udevice&idx=78&nvalue=0&svalue=0.92
8692843: sendcontent free: 10064 chunk size:400
8692847: sendcontent free: 8464 chunk size:400
8693029: GET /json.htm?type=commandśm=udevice&idx=78&nvalue=0&svalue=0.92 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2
8693030: GET /json.htm?type=commandśm=udevice&idx=78&nvalue=0&svalue=0.92 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2
8693030: HTTP : Command_HTTP_SendToHTTP closing connection
8693049: Command: sendtohttp
8693050: SendToHTTP: Host: 192.168.0.109 port: 8084
8693050: SendToHTTP: Path: /json.htm?type=commandśm=udevice&idx=79&nvalue=0&svalue=0.00
8693055: GET /json.htm?type=commandśm=udevice&idx=79&nvalue=0&svalue=0.00 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2
8693055: GET /json.htm?type=commandśm=udevice&idx=79&nvalue=0&svalue=0.00 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2
8693056: HTTP : Command_HTTP_SendToHTTP closing connection
8693070: sendcontent free: 9808 chunk size:400

TD-er, 2018-10-20:
Should be fixed by @uzi18 
See PR #1924

michass, 2018-10-20:
I will check after package update

Pobierz aplikację Outlook dla systemu Android<https://aka.ms/ghei36>

uzi18, 2018-10-20:
@michass mozesz wejsc na gitter.im i napisac do mnie na priv?

TD-er, 2018-10-20:
Patch is reverted, see comments @ #1924

uzi18, 2018-10-20:
definitely we need to log buffer when checksum is wrong, so we can investigate what is wrong here, mayne protocol. is different in new devices?

uzi18, 2018-10-21:
@TD-er thx to @michass we investigate problem and it is indeed trivial.
if someone is interested #1934 is ready to test

michass, 2018-10-22:
I have latest Log:
`35429684: GET /json.htm?type=command¶m=udevice&idx=75&nvalue=0&svalue=1 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2010
35429684: GET /json.htm?type=command¶m=udevice&idx=75&nvalue=0&svalue=1 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2010
35429685: HTTP : Command_HTTP_SendToHTTP closing connection
35429735: CSE: B0: 90+2+230+248+0+3+47+0
35429735: CSE: B8: 64+31+6+75+183+80+78+152
35429735: CSE: B16: 79+29+147+225+127+0+14+242
35429735: CSE: B24: 0+0+0+0+0+0+0+0
35429735: CSE: s: 26
35429736: CSE: Checksum Failure
35429862: CSE: B24: 0+0+0+0+0+0+0+0
35429862: CSE: s: 24
35429863: CSE: Checksum Failure
35429945: CSE: B0: 90+2+230+248+0+3+47+0
35429945: CSE: B8: 64+31+6+75+183+80+78+152
35429945: CSE: B16: 82+81+92+225+127+0+14+242
35429945: CSE: B24: 0+0+0+0+0+0+0+0
35429945: CSE: s: 26
35429946: CSE: Checksum Failure
35430052: CSE: B24: 0+0+0+0+0+0+0+0
35430053: CSE: s: 24
35430053: CSE: Checksum Failure
35430155: CSE: B0: 90+2+230+248+0+3+47+0
35430155: CSE: B8: 64+31+6+75+183+80+78+152
35430155: CSE: B16: 85+132+38+225+127+0+14+242
35430155: CSE: B24: 0+0+0+0+0+0+0+0
35430155: CSE: s: 26
35430156: CSE: Checksum Failure
35430260: CSE: B24: 0+0+0+0+0+0+0+0
35430260: CSE: s: 26
35430261: CSE: Checksum Failure
35430365: CSE: B0: 90+2+230+248+0+3+47+0
35430365: CSE: B8: 64+31+6+75+183+80+78+152
35430365: CSE: B16: 88+181+241+225+127+0+13+242
35430365: CSE: B24: 0+0+0+0+0+0+0+0
35430365: CSE: s: 24
35430366: CSE: Checksum Failure
35430524: EVENT: PoborWent#Power=1.02
35430560: EVENT: PoborWent#Power=1.02 Processing time:36 milliSeconds
35430562: EVENT: PoborWent#Current=0.02
35430576: CSE: B0: 90+2+230+248+0+3+47+0
35430576: CSE: B8: 64+31+6+75+183+80+78+152
35430576: CSE: B16: 91+233+186+225+127+0+13+242
35430576: CSE: B24: 0+0+0+0+0+0+0+0
35430577: CSE: s: 24
35430577: CSE: Checksum Failure
35430751: GET /json.htm?type=command¶m=udevice&idx=75&nvalue=0&svalue=1 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2010
35430751: GET /json.htm?type=command¶m=udevice&idx=75&nvalue=0&svalue=1 HTTP/1.1^^Host: 192.168.0.109:8084^^User-Agent: ESP Easy/2010
35430752: HTTP : Command_HTTP_SendToHTTP closing connection
35430783: CSE: B0: 90+2+230+248+0+3+47+0
35430783: CSE: B8: 64+31+6+75+183+80+78+152
35430783: CSE: B16: 95+28+132+225+127+0+14+242
35430783: CSE: B24: 0+0+0+0+0+0+0+0
35430784: CSE: s: 26
35430784: CSE: Checksum Failure
35430889: CSE: B0: 90+2+230+248+0+3+47+0
35430889: CSE: B8: 64+31+6+75+183+80+78+152
35430889: CSE: B16: 96+181+233+225+127+0+13+242
35430889: CSE: B24: 0+0+0+0+0+0+0+0
35430889: CSE: s: 24
35430890: CSE: Checksum Failure
35431937: CSE: B24: 0+0+0+0+0+0+0+0
35431937: CSE: s: 228
35431937: CSE: Checksum Failure
35432041: CSE: B0: 90+2+230+248+0+3+47+0
35432041: CSE: B8: 64+31+5+88+17+80+78+152
35432042: CSE: B16: 114+77+64+225+127+0+116+242
35432042: CSE: B24: 0+0+0+0+0+0+0+0
35432042: CSE: s: 230
35432042: CSE: Checksum Failure
35432149: CSE: B24: 0+0+0+0+0+0+0+0
35432149: CSE: s: 228
35432150: CSE: Checksum Failure
35432251: CSE: B0: 90+2+230+248+0+3+47+0
35432251: CSE: B8: 64+31+5+88+17+80+78+152
35432252: CSE: B16: 117+127+11+225+127+0+116+242
35432252: CSE: B24: 0+0+0+0+0+0+0+0
35432252: CSE: s: 230
35432252: CSE: Checksum Failure
35432549: EVENT: PoborWent#Current=0.02
35432567: CSE: B0: 90+2+230+248+0+3+43+0
35432567: CSE: B8: 64+31+5+88+17+80+78+152
35432567: CSE: B16: 122+76+57+225+127+0+112+242
35432567: CSE: B24: 0+0+0+0+0+0+0+0
35432567: CSE: s: 222
35432568: CSE: Checksum Failure
35432571: EVENT: PoborWent#Current=0.02 Processing time:22 milliSeconds
35432573: EVENT: PoborWent#Pulses=65018.00
35432742: Delay queue C001 stats: Count: 656 Avg/min/max 14496.87/8987/465201 usec
35432742: timeDiff() stats: Count: 935908450 - CPU cycles per call: 3.86
35432743: Scheduler stats: (called/tasks/max_length/idle%) 164465792/2160447/9/63.80
35432776: CSE: B0: 90+2+230+248+0+3+43+0
35432776: CSE: B8: 64+31+5+88+17+80+78+152
35432777: CSE: B16: 125+126+4+225+127+0+112+242
35432777: CSE: B24: 0+0+0+0+0+0+0+0
35432777: CSE: s: 222`

uzi18, 2018-10-22:
@TD-er have found another issue related to this one, basically all data are in the buffer but shifted so checksum is wrong

michass, 2018-10-25:
Hi,
after update to new firmware dedicated for Sonoff POW R2
Problem came back. Maybe you don't merge @uzi18 changes.

```
2018-10-25 10:45:00 | 1
-- | --
2018-10-25 10:50:00 | 1
2018-10-25 10:55:00 | 1
2018-10-25 11:00:00 | 1
2018-10-25 11:05:00 | 1
```

[image]

TD-er, 2018-10-25:
@michass Indeed the latest fixes of @uzi18 were not merged yet.
He mentioned he would add some changes, so I waited.
And my last merges were around 2:30 am, so I thought going to bed was probably a better idea.
At least I now have one POW R2 running ESPeasy too, so I can test myself.

michass, 2018-10-25:
I can't get logs.
Before update I had possibility to log from VPN (other IP range), now I can't connect via VPN.
I there default setting "Local only"?

TD-er, 2018-10-25:
Yep, the default is to allow local network only.

michass, 2018-10-25:
Another strange behaviour:
[image]
[image]
[image]

As I sayd, no acces to logs

michass, 2018-10-25:
Strange also is, then on @uzi18 i can connect via VPN, on official test rom for POW R2 i can't. Even if I allow for that.

uzi18, 2018-10-25:
I have build for You standard dev_ESP8266_4096, with only changes from #1934 , branch is 20181020 with revert of #1924 
If you want I can pull last changes and rebuild it one more time

michass, 2018-10-25:
Yes, sure

michass, 2018-10-25:
@uzi18 thanks!

uzi18, 2018-10-25:
@michass is it work for You?

michass, 2018-10-25:
Looks nice ;)
`1810009: CSE: B0: 242+90+2+230+248+0+3+41
1810009: CSE: B8: 0+64+31+4+61+126+80+78
1810009: CSE: B16: 152+250+232+156+97+202+229+238
1810009: CSE: B24: 242+90+0+0+0+0+0+0
1810009: CSE: s: 238
1810010: CSE voltage: 235.11
1810010: CSE power: 0.00
1810010: CSE current: 0.00
1810010: CSE pulses: 51941
1811269: CSE: B0: 242+90+2+230+248+0+3+41
1811269: CSE: B8: 0+64+31+4+50+155+80+78
1811269: CSE: B16: 152+0+0+128+97+202+229+2
1811269: CSE: B24: 242+90+0+0+0+0+0+0
1811269: CSE: s: 2
1811269: CSE voltage: 235.11
1811270: CSE power: 0.00
1811270: CSE current: 0.00
1811270: CSE pulses: 51941
1812318: CSE: B0: 242+90+2+230+248+0+3+43
1812319: CSE: B8: 0+64+31+4+50+155+80+78
1812319: CSE: B16: 152+31+23+74+97+202+229+4
1812319: CSE: B24: 242+90+0+0+0+0+0+0
1812319: CSE: s: 4
1812319: CSE voltage: 234.53
1812319: CSE power: 0.00
1812319: CSE current: 0.00
1812320: CSE pulses: 51941
1812525: CSE: B0: 242+90+2+230+248+0+3+43
1812525: CSE: B8: 0+64+31+4+50+155+80+78
1812526: CSE: B16: 152+34+75+19+97+202+229+4
1812526: CSE: B24: 242+90+0+0+0+0+0+0
1812526: CSE: s: 4
1812526: CSE voltage: 234.53
1812526: CSE power: 0.00
1812526: CSE current: 0.00
1812527: CSE pulses: 51941
1812841: CSE: B0: 242+90+2+230+248+0+3+43
1812841: CSE: B8: 0+64+31+4+50+155+80+78
1812841: CSE: B16: 152+39+22+67+97+202+229+4
1812842: CSE: B24: 242+90+0+0+0+0+0+0
1812842: CSE: s: 4
1812842: CSE voltage: 234.53
1812842: CSE power: 0.00
1812842: CSE current: 0.00
1812842: CSE pulses: 51941
1813887: CSE: B0: 242+90+2+230+248+0+3+43
1813887: CSE: B8: 0+64+31+3+130+70+80+78
1813887: CSE: B16: 152+55+21+52+97+202+229+254
1813887: CSE: B24: 242+90+0+0+0+0+0+0
1813888: CSE: s: 254
1813888: CSE voltage: 234.53
1813888: CSE power: 0.00
1813888: CSE current: 0.00
1813888: CSE pulses: 51941
1815042: CSE: B0: 242+90+2+230+248+0+3+43
1815042: CSE: B8: 0+64+31+3+130+70+80+78
1815042: CSE: B16: 152+72+172+139+97+202+229+253
1815042: CSE: B24: 242+90+0+0+0+0+0+0
1815042: CSE: s: 253
1815043: CSE voltage: 234.53
1815043: CSE power: 0.00
1815043: CSE current: 0.00
1815043: CSE pulses: 51941
1816195: CSE: B0: 242+90+2+230+248+0+3+43
1816196: CSE: B8: 0+64+31+3+130+70+80+78
1816196: CSE: B16: 152+90+67+226+97+202+229+253
1816196: CSE: B24: 242+90+0+0+0+0+0+0
1816196: CSE: s: 253
1816196: CSE voltage: 234.53
[... 213 lines cut ...]
1834443: CSE: B24: 242+90+0+0+0+0+0+0
1834443: CSE: s: 57
1834444: CSE voltage: 235.11
1834444: CSE power: 0.00
1834444: CSE current: 0.00
1834444: CSE pulses: 51942
1835699: CSE: B0: 242+90+2+230+248+0+3+41
1835700: CSE: B8: 0+64+31+3+167+156+80+78
1835700: CSE: B16: 152+173+119+218+97+202+230+246
1835700: CSE: B24: 242+90+0+0+0+0+0+0
1835700: CSE: s: 246
1835700: CSE voltage: 235.11
1835700: CSE power: 0.00
1835700: CSE current: 0.00
1835701: CSE pulses: 51942
1835909: CSE: B0: 242+90+2+230+248+0+3+41
1835910: CSE: B8: 0+64+31+3+167+156+80+78
1835910: CSE: B16: 152+176+170+164+97+202+230+246
1835910: CSE: B24: 242+90+0+0+0+0+0+0
1835910: CSE: s: 246
1835910: CSE voltage: 235.11
1835910: CSE power: 0.00
1835910: CSE current: 0.00
1835911: CSE pulses: 51942
`

michass, 2018-10-25:
No :(
If I try, then Unit don't respond, only  switch power off can resolve problem.

uzi18, 2018-10-25:
So something went wrong between 20-25 october

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er