letscontrolit/ESPEasy #1921: deep sleep and irregular interval

szwigiel, 2018-10-19:
Hello,
My ESP is connect to mobile wifi router. For all sensors I have interval = 60s. 
I care about the power savings that's why I use the deep sleep mode (awake=5s; sleep=55).

Unfortunately, the frequency of data transfer (to domoticz) is not constant. 
The value: (Time_devicechanged - Time_lastupdate) i got from ~62s to ~140s.

This irregular time is due to the fact that my mobile router remembers the previous connection. The next connection after deep sleep is crashed.
That's why I'm asking if it's possible to apply disconnection WIFI before deep sleep? Can you implement it?

My English is terrible, sorry for that. I hope you will understand me ;)

Budman1758, 2018-10-19:
2 possible solutions.
Use a static IP address.
Use a rule to disconnect WIFI before sleep.

szwigiel, 2018-10-21:
I used the second solution. Thanks 👍

TD-er, 2018-10-21:
So this one is resolved?
Then we can close this issue.
If it should still be open, please re-open it.

closed 2018-10-21 by TD-er