letscontrolit/ESPEasy #1929: ESP32 Interrupt/millis() issues

thehijjt, 2018-10-21:
I am using an Espeasy to read some flow meters, the esp8266 works perfectly fine for this, and doesn't have this bug. However I needed more I/O, and tried to use the ESP32 with the same devices. They are hall effect flow sensors with 800 pulses/Liter. Any time there is flow on the flow meter using ht Generic Pulse Counter plugin the ESP32 will crash, and reboot
with this message:
`Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed)`

I researched this and found that I can get them to count pulses and not crash by adding IRAM_ATTR to the pulse check ISR, and removing the call to the millis() function, however this also removes any debouncing.
```
void IRAM_ATTR Plugin_003_pulsecheck(byte Index)
{
  //const unsigned long PulseTime=timePassedSince(Plugin_003_pulseTimePrevious[Index]);
  //if(PulseTime > (unsigned long)Settings.TaskDevicePluginConfig[Index][0]) // check with debounce time for this task
  //{
        Plugin_003_pulseCounter[Index]++;
        Plugin_003_pulseTotalCounter[Index]++;
  //    Plugin_003_pulseTime[Index] = PulseTime;
  //    Plugin_003_pulseTimePrevious[Index]=millis();
  //}
}
```
Simply adding the IRAM_ATTR will still crash on the flow meter calling the interrupt, the millis() function removal was also necessary. 

### System configuration
Hardware:
Lolin D32

ESP Easy version: latest git esp32test_1M8_partition, built using platformio.

TD-er, 2018-10-21:
I know the ESP32 has a lot more options for reading GPIO and perhaps it even has a hardware debounce-option.
So I guess we should add a separate function for the ESP32 reading the GPIO state.

By the way, good to see such a well descriptive issue report :)

viorels, 2018-12-30:
I've hit this bug also. Thanks for the workaround @thehijjt !
@TD-er , any tips on how to fix the timer also? I have little experience with ESP32, but maybe I can contribute in some way...

TD-er, 2018-12-30:
@viorels At this moment I don't have any tip... yet.
Fixing GPIO handling is one of the things we really have high on the to-do list and as easy/simple as it sounds, it apparently isn't. :(
But the last weeks/months I've been really busy trying to fix the HW watchdog and WiFi issues (probably related) so there hasn't been as much activity as I would have liked to see on this topic.

thehijjt, 2019-05-08:
FYI, 
This is issue is exactly the same problem. 
https://github.com/espressif/arduino-esp32/issues/2758#issue-441733256
Their workaround uses micros() instead of millis(), and the problem goes away. 

The ESP32 has built in pulse counters (8 of them I believe)  #include "driver/pcnt.h".  I believe they include filtering and would be a better fit than using a soft debounce. I don't currently have an esp32 with me to test.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er