letscontrolit/ESPEasy #1937: MCP-23017 one Portnumber shifted

Sasch600xt, 2018-10-22:
Hello,

i did try the feedback for the gpio state for my MCP-23017 and found something confusing :

The feedback is alwas one gpio number too high.
For example i set gpio input 0 to "HIGH" (external by wire +3.3V) then i get as feedback "gpio 1 HIGH"
So i can never see the state of gpio 0.
So all gpios are shifted one up for feedback.
Also in the log output.

these are my rules:

``on System#Boot do
timerSet,1,5
endon

On Rules#Timer=1 do
Publish,%sysname%/MCP-GPIO/A0,[plugin#mcpgpio#pinstate#0]
Publish,%sysname%/MCP-GPIO/A1,[plugin#mcpgpio#pinstate#1]
Publish,%sysname%/MCP-GPIO/A2,[plugin#mcpgpio#pinstate#2]
Publish,%sysname%/MCP-GPIO/A3,[plugin#mcpgpio#pinstate#3]
Publish,%sysname%/MCP-GPIO/A4,[plugin#mcpgpio#pinstate#4]
Publish,%sysname%/MCP-GPIO/A5,[plugin#mcpgpio#pinstate#5]
Publish,%sysname%/MCP-GPIO/A6,[plugin#mcpgpio#pinstate#6]
Publish,%sysname%/MCP-GPIO/A7,[plugin#mcpgpio#pinstate#7]
Publish,%sysname%/MCP-GPIO/B0,[plugin#mcpgpio#pinstate#8]
Publish,%sysname%/MCP-GPIO/B1,[plugin#mcpgpio#pinstate#9]
Publish,%sysname%/MCP-GPIO/B2,[plugin#mcpgpio#pinstate#10]
Publish,%sysname%/MCP-GPIO/B3,[plugin#mcpgpio#pinstate#11]
Publish,%sysname%/MCP-GPIO/B4,[plugin#mcpgpio#pinstate#12]
Publish,%sysname%/MCP-GPIO/B5,[plugin#mcpgpio#pinstate#13]
Publish,%sysname%/MCP-GPIO/B6,[plugin#mcpgpio#pinstate#14]
Publish,%sysname%/MCP-GPIO/B7,[plugin#mcpgpio#pinstate#15]
timerSet,1,5
EndOn``

Greetz
Sascha

giig1967g, 2018-10-22:
I will check it

TD-er, 2018-10-22:
Just to be clear, it is just a matter of reading the pin?
Setting it is OK?

giig1967g, 2018-10-22:
@Sasch600xt:
Just to understand better, is the issue related to this command:
`Publish,%sysname%/MCP-GPIO/A0,[plugin#mcpgpio#pinstate#0]`
that returns the state of `mcpgpio 1` instead of `mcpgpio 0`?

Sasch600xt, 2018-10-22:
ok....it is really strange.

with this command
`http://192.168.123.60/control?cmd=MCPGPIO,0,1`

no gpio pin is HIGH, no Feddback of a HIGH value.

with this command
`http://192.168.123.60/control?cmd=MCPGPIO,1,1`

i messure 3,3volt (HIGH) at gpoi0.
Feedback shows gpio1 HIGH.

can someone proof that ?
maybe a have a broken hardware MCP23017
I have another one here, but before i change that on my circuit i would love to hear results from someone else with same hardware.

Greetz
Sascha

giig1967g, 2018-10-22:
Hi 
according to this:
`https://www.letscontrolit.com/wiki/index.php/MCP23017`, port range starts from 1 not from 0:

```
A0: | A1: | A2: | Address: | Port range:
-- | -- | -- | -- | --
0 | 0 | 0 | 0x20 | 1 - 16
1 | 0 | 0 | 0x21 | 17 - 32
0 | 1 | 0 | 0x22 | 33 - 48
1 | 1 | 0 | 0x23 | 49 - 64
0 | 0 | 1 | 0x24 | 65 - 80
1 | 0 | 1 | 0x25 | 81 - 96
0 | 1 | 1 | 0x26 | 97 - 112
1 | 1 | 1 | 0x27 | 113 - 128
```

So it seems to me that executing http://192.168.123.60/control?cmd=MCPGPIO,1,1 will set to HIGH the first PIN (that could be 0)

Sasch600xt, 2018-10-22:
ok...this was the problem !

I changed the rules and started counting from 1-16 now.
on my hardwaremodule the gpios strting from 0.
This is confusing, but now i got it to work.

Thank you for your help! :)

Greetz
Sascha

giig1967g, 2018-10-22:
@Sasch600xt 
glad to be helpful.
please close the issue.

closed 2018-10-22 by Sasch600xt