letscontrolit/ESPEasy #1938: Stability problem with logs

michass, 2018-10-22:
Hi,
When we work with #1934 I found problem.
After apply Syslog for my Synology server (debug) and weblog (debug) in same time, unit goes to restart loop after few seconds. I sthere posibility to get two way logs, or maybe add condition to break syslog when weblog is opened?

TD-er, 2018-10-22:
Can you show the memory usage when you enable syslog?
What version/build are you running?

michass, 2018-10-22:
I will tray, when go back home, I'm affraid, because when I enabe this two, i can loose control and only reflash can bring unit to life. I have  @uzi18 custom flash,, i think 21.10.2018 dev 4096

TD-er, 2018-10-22:
I will add some option to check if it is entering some bootloop (e.g. several reboots with less than N seconds uptime)
Do you have any idea how quick it will reboot?

michass, 2018-10-22:
I have domoticz based script to restore relay state after reboot with log uptime with 1 min step. I don't know exactly, but in 90% under 1min, but sometimes few:
`

2018-10-21 20:55:59 | 2
-- | --
2018-10-21 20:54:59 | 1
2018-10-21 20:54:06 | 0
2018-10-21 20:53:35 | 1
2018-10-21 20:52:44 | 0
2018-10-21 20:52:19 | 0
2018-10-21 20:50:49 | 0
2018-10-21 20:49:25 | 0
2018-10-21 20:47:52 | 0
2018-10-21 20:46:58 | 0
2018-10-21 20:46:25 | 0
2018-10-21 20:45:50 | 0
2018-10-21 20:44:46 | 0
2018-10-21 20:44:09 | 0
2018-10-21 20:43:28 | 0
2018-10-21 20:42:04 | 7
2018-10-21 20:41:03 | 6
2018-10-21 20:40:04 | 5
2018-10-21 20:39:04 | 4
2018-10-21 20:38:04 | 3
2018-10-21 20:37:03 | 2
2018-10-21 20:36:03 | 1
2018-10-21 20:35:10 | 0
2018-10-21 20:32:56 | 0
2018-10-21 20:32:08 | 0
2018-10-21 20:31:28 | 0
2018-10-21 20:30:26 | 0
2018-10-21 20:29:03 | 0
2018-10-21 20:28:17 | 0
2018-10-21 20:27:17 | 0
2018-10-21 20:26:38 | 16
2018-10-21 20:25:38 | 15
2018-10-21 20:24:38 | 14
2018-10-21 20:23:38 | 13
2018-10-21 20:22:38 | 12
2018-10-21 20:21:38 | 11
2018-10-21 20:20:38 | 10
2018-10-21 20:19:38 | 9
2018-10-21 20:18:38 | 8
2018-10-21 20:17:38 | 7
2018-10-21 20:16:38 | 6
2018-10-21 20:15:38 | 5
2018-10-21 20:14:38 | 4
2018-10-21 20:13:38 | 3
2018-10-21 20:12:38 | 2
2018-10-21 20:11:38 | 1
2018-10-21 20:10:46 | 0
2018-10-21 20:10:10 | 0
2018-10-21 20:09:46 | 4
2018-10-21 20:08:46 | 3
2018-10-21 20:07:46 | 2
2018-10-21 20:06:46 | 1
2018-10-21 20:05:55 | 0
2018-10-21 20:05:07 | 6
2018-10-21 20:04:07 | 5
2018-10-21 20:03:07 | 4
2018-10-21 20:02:07 | 3
2018-10-21 20:01:07 | 2
2018-10-21 20:00:07 | 1
2018-10-21 19:59:14 | 0
2018-10-21 19:58:15 | 3
2018-10-21 19:57:15 | 2
2018-10-21 19:56:15 | 1
2018-10-21 19:55:24 | 0

2018-10-21 19:44:19 | 2
-- | --
2018-10-21 19:43:19 | 1
2018-10-21 19:42:26 | 0
2018-10-21 19:41:32 | 3
2018-10-21 19:40:32 | 2
2018-10-21 19:39:32 | 1
2018-10-21 19:38:38 | 0
2018-10-21 19:37:56 | 74

2018-10-21 18:25:56 | 2
-- | --
2018-10-21 18:24:56 | 1
2018-10-21 18:24:05 | 0
2018-10-21 18:23:03 | 3
2018-10-21 18:22:03 | 2
2018-10-21 18:21:03 | 1
2018-10-21 18:20:13 | 0
2018-10-21 18:19:05 | 3
2018-10-21 18:18:05 | 2
2018-10-21 18:17:05 | 1
2018-10-21 18:16:13 | 0
2018-10-21 18:12:13 | 34

`

michass, 2018-10-24:
After suggestion of @uzi18 to find the reason of reboots. Syslog was turned on.

[image]
[image]

Fortunately I did tufn it off after long time of reboots.

TD-er, 2018-10-24:
Those free memory values are too low to work stable.
Do you really need the 'dev' build? Or is "normal" good enough?

michass, 2018-10-24:
But what with CSE7766 in normal?

TD-er, 2018-10-24:
It is apparently a "testing" plugin. Dev even has more plugin compared to 'test', so you may want to use the 'test' builds
By the way, a lot has been changed to that plugin the last few days, which has not yet been merged. I think it will be merged later today, so then it will be included in the build tomorrow.
I'll advice to use that one.

michass, 2018-10-24:
Thanks, 
then I will try next test version.

TD-er, 2018-10-24:
Not sure yet, but I guess somewhere between 6 and 10k free would be nice.

michass, 2018-10-24:
[image]

This is regular state without crash loop and without any logs enabled

TD-er, 2018-10-24:
You need CSE7766 
Do you want to run it on a Sonoff POW r2?

michass, 2018-10-24:
Yes, why?

TD-er, 2018-10-24:
There is a Sonoff POW R2 build, which has a lot less plugins.
I just installed it on my R2 I had here and I also noticed it is using 4MB flash.
So I am now changing the build definition for it to allow OTA updates of it. (1MB flash cannot be used at the moment for OTA)
So you can also try to flash the Sonoff POW R2 build of tomorrow.

michass, 2018-10-24:
Wery nice!
Thanks!

michass, 2018-10-24:
[image]
Can it be the reason?

TD-er, 2018-10-24:
Those 74 plugins use a lot of memory.

Unit: | 20
-- | --
Local Time: | 2018-10-24 20:36:28
Uptime: | 0 days 3 hours 16 minutes
Load: | 10.60% (LC=7149)
Free Mem: | 17120 (12656 - sendContentBlocking)
Free Stack: | 3552 (1488 - LoadTaskSettings)
-- | --
Build:⋄ | 20102 - Mega
Libraries:⋄ | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2
GIT version:⋄ |  
Plugins:⋄ | 6 [Normal]

michass, 2018-10-24:
And Md5 check fail.

TD-er, 2018-10-24:
MD5 check fail is normal when you compile it yourself.

Which reminds me, I have to remove that check if the DummyPlaceHolder.... string is still not removed.
This will only confuse people.

closed 2019-10-11 by Grovkillen