letscontrolit/ESPEasy #1940: PCA9685/PCF8547 add two checkboxes for the mode2 settings INVRT (Bit 4) and OUTDRV (Bit 2)

micropet, 2018-10-22:
The PCA9685 board is the only way to run multiple Leddrivers with ESPEasy.

Unfortunately the possibility for mode2 settings INVRT (bit 4) and OUTDRV via a checkbox is still missing.

On 21 Dec 2017 ShardanX wrote the following:

The PCA9685 is an external PWM chip.
It can be set to different modes for direct LED use or for use with MOSFET or MOSFET/IGBT driver chip.
These modes are not accessible from the task setup in ESPEasy.

As LED stripes and such are daily used items to day this should be changed.

In the plugin init there is this line:
Plugin_022_writeRegister(PCA9685_MODE2, (byte)0x10); // set to output
It sets the output to inverted and open drain configurtation.

I suggest to use two checkboxes for the mode2 settings INVRT (Bit 4) and OUTDRV (Bit 2).
In the init create a variable mode2set for examle.
If OUTDRV is checked do a mode2set=mode2set + 4 (Sets bit 2 to 1)
If INVRT is checked do a mode2set=mode2set + 16 (Sets bit 4 to 1)
Then write the mode2set to the MODE2 register of the PCA9685

This makes the use of FET's and driver chip possible. They need non inverted output, direct driving of a MOSFET needs totem pole config of the outputs, some drivers might need that too. As tehere are driver chips inverting and non-inverting, this setting should be separate from the OUTDRV setting.

The INVRT should be added to the PCF8547 too if possible as the outputs are always inverted with the current settings.

giig1967g, 2018-10-22:
@micropet: PCF8574 already has the inverted logic setting (I did it :))

micropet, 2018-10-22:
Aha.

Can you do it synonymous for the PCA9685?
That's the most important to me.

giig1967g, 2018-10-22:
Hi,
As I don't have a 9685 I can develop it, but you will have to test it before merging.

micropet, 2018-10-22:
Yes this is not a problem. :)

giig1967g, 2018-10-22:
ok. let me finish doubleclick and longpress for PCF and MCP and then I will look into it.

micropet, 2018-10-22:
OK, i will wait.

TD-er, 2018-10-22:
Maybe we should also have a discussion about what to do with internal/external GPIO handling.
Now we're implementing stuff several times and that's asking for bugs.

We should have a generic way of GPIO handling and I guess the comment I made earlier today about the Switch plugin also applies here.
For GPIO handling, we have:
- input
- state
- output

svmac, 2018-10-23:
Micropet looks at PR #1944. It allows you to define MODE2 register.

micropet, 2018-10-23:
Wau, thank you. It compiles fine. Will check it later this day.

TD-er, 2018-10-23:
@micropet If that PR works out great, please add a comment to that PR too, then we can merge it.

P.S. I love it to see so much activity in bug hunting and progress in testing. :)

micropet, 2018-10-23:
Thank you. I will test today.

uzi18, 2018-10-23:
1,4096?

micropet, 2018-10-23:
I had not rebooted.

If I set MODE2 to 6 and send the following:
mosquitto_pub -t ESP-218 / cmd -m 'pcapwm, 1.4095'
is the led on

mosquitto_pub -t ESP-218 / cmd -m 'pcapwm, 1,0' turns it off.

Much better.

micropet, 2018-10-23:
1,4096 is not accepted

micropet, 2018-10-23:
Well, I could not figure out the meaning of the possible settings.

For the application with a Ledtreiber but it is enough.

MODE 2 to 6 inverts the default setting and I can dim my light from 0 to 4095 - perfect.

But maybe someone can explain the remaining MODE 2 settings.

uzi18, 2018-10-23:
some. lines of info log please

micropet, 2018-10-23:
What do you expect to see in the logfile?

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er