letscontrolit/ESPEasy #1945: Correction of CSE7766 values

michass, 2018-10-23:
Hi, in device tab, when I try to calibrate values by formula %value% +-/*, it's not working, every value have the same value as before.

TD-er, 2018-10-23:
Calibration is a basic part of that plugin (or at least it should be), maybe we should add basic calibration settings as part of the plugin.
Computing formula takes a lot more CPU power.

Apart from that, the formula should work, so that's also a proper bug I guess.

TD-er, 2018-10-23:
Maybe related issue: https://github.com/letscontrolit/ESPEasy/issues/1876

clumsy-stefan, 2018-10-23:
Calibration Values can be put in as part of the definition. they are static I guess. If not set (or set to 0) it takes the ones stored in chip as all the chips seem to be calibrated individually before they leave the factory.

I never played with formula's though and if that would need to be done within the plugin, I can tell you it's not! (I wrote it)...

michass, 2018-10-23:
You mean, that My CSE7766 is well calibrated?
I have checked with multimeter witk calibration cert and i see difference. For what purpose we have formula fields in Espeasy, when we don't need them?

clumsy-stefan, 2018-10-23:
calibration: 
as far as I understand the CSE documentation the chips get initially calibrated. try to set all three calibration values to 0 in the plugin wait some cycles and check again! Remeber as well the CSE6677 as your multimeter have (quite some) variance with each measurement, except you have real expensive equipment...

Formulas:
I think formulas are used for other things, like if you have Wh and want to send kWh to the controller server. or want to mix variables or similar... (please correct me someone if I do get that wrong).

of course you could alsp use formulas for calibration, but as @TD-er commented it's probably not the most optimal/performant way to do it.

TD-er, 2018-10-23:
Formulas are great for conversion and also calibration options if the plugin doesn't support internal calibration.
I don't think the modules themselves are calibrated per piece. Maybe per batch to allow for some different components.
I will also install it on my POW nodes and maybe we should add some proper calibration interface to give the actual value and then let the computation of calibration values be done in the plugin.

Still if there is a formula field, it should work and if it doesn't, it is a bug.

michass, 2018-10-23:
I saw something simillar in tasmota

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er