letscontrolit/ESPEasy #1947: Reset Pin

flexiti, 2018-10-23:
What is the idea of this pin? Only return to factory settings?
If it is also reset it is not better to connect the switch to the reset pin because it will always work?
(software reset will not always work)

I do not know if I understand the idea correctly 

[image]

v-a-d-e-r, 2018-10-23:
E.g. the Sonoff nodes have no accessible RESET-pin. So the software reset is a chance to exit a boot loop.

uzi18, 2018-10-23:
what mean "press about 10s for factory reset"? so short to vcc or gnd?
what about 5s - enable AP. mode, 
10s - disable also all controllers, task and rules, but not delete
15s - wipe all data even, core wifi stuff

uzi18, 2018-10-23:
please remember - in the past it was possible to reset settings with short TX-RX

v-a-d-e-r, 2018-10-23:
Normally short to GND.

TD-er, 2018-10-23:
> please remember - in the past it was possible to reset settings with short TX-RX

That should still be possible.
I saw that piece of code and left it alone.

uzi18, 2018-10-23:
just to consider if we really need another pin... to reset

closed 2018-10-24 by flexiti