letscontrolit/ESPEasy #1948: wifiSSID and wifiKEY are not getting stored

chunter1, 2018-10-23:
I just flashed a Wemos D1 mini with the blank_4MB image and afterwards with the latest mega version.
Over the serial console i configured IP, wifiSSID and wifiKEY and saved it using SAVE.
However when rebooting, the modul refuses to connect ot the AP.
Verifying the settings, i always get the following wrong response:
wifiSSID = "ssid"
wifiKEY = "wpakey"
Am i doing something wrong or is there a bug?

Grovkillen, 2018-10-23:
Could you please try my flasher?

chunter1, 2018-10-23:
Do you mean the included "FlashESP8266.exe"?
I did flash both, the BLANK and the unchanged, locally compiled NORMAL image.

Grovkillen, 2018-10-23:
https://github.com/Grovkillen/ESP_Easy_Flasher

Grovkillen, 2018-10-23:
But, did you compile yourself you say?

chunter1, 2018-10-23:
ESP_Easy_Flasher is new to me.
Yes, i did compile by myself (no single line changed).

Grovkillen, 2018-10-23:
But please try an official bin first before opening a issue.

chunter1, 2018-10-23:
Ok, i will try and report in a view hours

Grovkillen, 2018-10-25:
Close this for now. Open if you still are struggling.

closed 2018-10-25 by Grovkillen