letscontrolit/ESPEasy #1949: Help me with Sonoff 3 way. Fake?!

Cino2424, 2018-10-23:
Got bought an 3 way touch. Think its called LX-WIFI-03O
[image]
[image]

Is it possible to flash? Nothing found on the net for such a board.

Grovkillen, 2018-10-23:
Be inspired by this unit, you can do the same to flash any ESP82XX unit. https://www.letscontrolit.com/wiki/index.php/Sonoff_DIY_1ch

And then, take this to the forum.

Grovkillen, 2018-10-23:
And another one https://www.letscontrolit.com/wiki/index.php/Sonoff_DIY_4ch

closed 2018-10-23 by Grovkillen

Cino2424, 2018-10-23:
I need Pin for TX RX GPIO0

The pins on the ESP are to small for me to soldier. Thought somebody has flashed else. Couse the is a pad in right corner down.

Grovkillen, 2018-10-23:
Practice makes perfect :)