letscontrolit/ESPEasy #1951: BlynkGet 2,1,get/V2

tyomikh, 2018-10-23:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

tyomikh, 2018-10-23:
ESP_Easy_mega-20181017_test_ESP8266_1024.bin on D1/1MB
same with ESP_Easy_mega-20181017_test_ESP8266_4096.bin on NodeMCUv3/4MB

To reproduce: 
- create a project in Blynk cloud,
- add Button with a virtual PIN,
- create a Rule in ESPEasy to check for PIN state BlynkGet 2,1,get/V2 and write it contents to a dummy device
- ESPEasy log says state was read (either =>0 or =>1) and then infinite Timeout every 1 sec

I uploaded some screenshots into issue #838

tyomikh, 2018-10-29:
Strange. Same Wemos D1, same version mega-20181017, same Blynk project, almost same ESPEasy Rules, different router with redirect to proxy (I tried to sniff and analyze the traffic) - BlynkGet 2,1,get/V2 works as expected

tyomikh, 2018-10-29:
More strange things. Back on home network:
1) update/V0?value=30.0 -> Updated OK
2) get/V2 -> Infinite retries and timeouts
TCPDUMP analysis
1) blynk-cloud.com resolved to 139.59.206.133 and used. success
2) suddenly 45.55.96.146 used for blynk-cloud.com and returned 400 Bad Request, Line-based text data: Invalid token
Nslookup for blynk-cloud.com: both my router and upstream DNS servers and Google and Yandex DNS always return 139.59.206.133 
for i in `dig blynk-cloud.com any | grep NS.*ns- | sort | uniq | awk '{print $NF}'`; do nslookup blynk-cloud.com $i; done
also return 139.59.206.133

Googled the latest error: update is required for Arduino libs (blynk).
https://community.blynk.cc/t/solved-error-cannot-connect-my-esp-to-wifi/11410/6
referenced by
https://community.blynk.cc/t/solved-wifi-connect-issue-to-blynk-server-on-wemos-d1-mini/11436
and
https://community.blynk.cc/t/solved-blynk-runs-to-wrong-server/11426/2

Can anyone check what Blynk library version was used to build ESP_Easy_mega-20181017_test_ESP8266_1024.bin ??

OK, I changed Blynk-HTTP Controller from hostname to IP=139.59.206.133
Another TCPDUMP for BlynkGet 2,1,get/V2: Blynk server replies with 200 OK and return the value of V2. BUT ESPEasy log still invaded by "Timeout while reading input data!", new one every 1 sec, regardless of if AgnoreACK or CheckACK is chosen in controller settings. I'm lost...

TD-er, 2018-10-29:
What IP settings do you use in ESPeasy? (not the Blynk controller)

uzi18, 2018-10-30:
It looks like problems with Your router/internet connection.
We not use any of Blynk libs here, just Blynk REST api.

uzi18, 2018-10-30:
@TD-er but loop is not good thing here, need to catch it and exit.

tyomikh, 2018-10-30:
Hello,

IP settings are simple: ESPEasy gets address via DHCP then router does NAT.

Today network setup:
esp --> via wifi --> mikrotik1 --> NAT --> ethernet --> mikrotik2 --> NAT --> via wifi --> iphone modem --> internet

tcpdump on mikrotik1:
http get/v2 esp -> 139.59.206.133
http reply 200 OK, array of values returned

esp log:
Timeout while reading input data!
    repeated every 1 sec

may be double/tripple NAT spoils everything while transparent proxy works. I don't know

TD-er, 2018-10-30:
I think I have an idea what's happening here.
ESPeasy is trying to send data using a random port and connect to the specified port.
How is your port forwarding configured in the NAT on both sides?
Do you filter on source port, or destination port?
It should be on destination port.

tyomikh, 2018-10-30:
NAT is classic for Mikrotik:
/ip firewall nat add chain=srcnat action=masquerade out-interface=bridge log=no log-prefix=""
nothing is filtered, no static port-forwarding in use

TD-er, 2018-10-30:
2 times NAT in the same network is often causing issues.
Why do you need this extra NAT layer?

tyomikh, 2018-10-30:
I checked commits to C012 and rolled back to mega-20180826 where Blynk controller has no queues and ACK options. It simply works. But Timer processing is not very good, TimerSet,3,0 doesn't stop it.

uzi18, 2018-10-30:
thx for this info

uzi18, 2018-10-30:
@tyomikh please give me "debug more" log.

tyomikh, 2018-10-30:
I checked how rules works on 20180826 and I don't like it (compared with 20181017): timer is not stopped, nested IF processing is bizarre, GPIO not always set as instructed by a rule command.

Flashed 20181017 again and captured Debug dev log:
41538: Read settings: TaskSettings index: 0
41543: Read settings: TaskSettings index: 1
41547: DEBUG DEV: Parsed String=' if 0'
41556: Read settings: TaskSettings index: 0
41561: Read settings: TaskSettings index: 1
41566: DEBUG DEV: Parsed String=' if 0'
41576: Read settings: TaskSettings index: 0
41581: Read settings: TaskSettings index: 1
41588: DEBUG DEV: Parsed String=' TaskValueSet 2,2,0'
41603: sendcontent free: 10784 chunk size:400
41826: HTTP : Success
41827: content-type: application/json;charset=utf-8^
41827: HTTP : Success
41828: access-control-allow-origin: *^
41828: HTTP : Success
41829: content-length: 5^
41829: HTTP : Success
41829: ^
41829: HTTP : Success
41851: sendcontent free: 9264 chunk size:400
41858: sendcontent free: 9480 chunk size:347
41860: sendcontent free: 9480 chunk size:0
41990: sendcontent free: 9480 chunk size:274
41992: sendcontent free: 7880 chunk size:0
42829: Timeout while reading input data!
42830: ['1']
42830: HTTP : Success
42830: Blynk get - get/V2 => 1
43041: sendcontent free: 9264 chunk size:400
43046: sendcontent free: 9480 chunk size:121
43050: sendcontent free: 9480 chunk size:0
43830: Timeout while reading input data!
43830:
43831: Blynk get - get/V2 => 1
43932: sendcontent free: 9264 chunk size:400
43936: sendcontent free: 7880 chunk size:139
43938: sendcontent free: 7880 chunk size:0
44831: Timeout while reading input data!
44832:
44832: Blynk get - get/V2 => 1

current rule:

on Rules#Timer=1 do
  BlynkGet 2,1,get/V2
  if [dummy#switch]<>[dummy#switchPrev]
    if [dummy#switch]=1
      event,ON
      GPIO,5,1
    endif
    if [dummy#switch]=0
      GPIO,5,0
      event,OFF
    endif
    TaskValueSet 2,2,[dummy#switch] // renew switchPrev
    TimerSet,2,5   // update uptime in Blynk cloud in a while
    TimerSet,3,10  // update timestamp in Blynk cloud in a while
  else
    event,NoChange
  endif
  TimerSet,1,30
endon

on Rules#Timer=2 do
  BlynkGet update/V1?value=%uptime%
  TimerSet,2,0 // no need for more updates
endon

on Rules#Timer=3 do
  BlynkGet update/V5?value=%sysyear%-%sysmonth%-%sysday%%20%systm_hm%%0a
  TimerSet,3,0 // no need for more updates
endon

on System#Boot do
  TimerSet,1,40
endon

uzi18, 2018-10-30:
ok now I see what's wrong, it is easy to fix, are You able to recompile espeasy on Your own?

tyomikh, 2018-10-30:
yes, I can try to recompile. need to remember/google how to filter plugins to fit into 1MB

uzi18, 2018-10-30:
Just build one from *_1024 images.
at the end of file _C012.ino:
```
sprintf_P(log, PSTR("Blynk get - %s => %s"),command.c_str(), value_char   );
        addLog(LOG_LEVEL_DEBUG, log);
break; // add this line in this place
      }
      delay(0);
```

TD-er, 2018-10-30:
@uzi18 Did you already test it?
If that fixes it, shouldn't it 'break' on any place where `success = true;` is set?
So the while loop could then be changed from:
`while (client_available(client)) {`
to:
`while (client_available(client) && !success) {`

Or am I missing something?

uzi18, 2018-10-30:
@TD-er No I can't today. 

It is true if You add also here:
https://github.com/letscontrolit/ESPEasy/blob/007ef78f2bc48a6604a5e91a15eb499e8e972656/src/_C012.ino#L122
if (!data) success = true;

Because this function is used to set and fetch data.

TD-er, 2018-10-30:
@uzi18 A few lines above, in the same while loop, there is also a success = true; statement.
So that's why I suggested to check for success in the while invariant statement.
So then you also have a quick exit on success when not waiting for data.

uzi18, 2018-10-30:
@TD-er both changes are needed - PR ready :)
and "break" was only for test if it works correctly

@tyomikh You got big lag with data - 1,3s. for all data 5-6 lines - it is very bad internet connection

TD-er, 2018-10-30:
So should be fixed by #1984

closed 2018-10-30 by TD-er

uzi18, 2018-10-30:
@TD-er we need to add timeout there ... to get out of this loop, what do You think 1,5s?

uzi18, 2018-10-30:
https://github.com/letscontrolit/ESPEasy/blob/c67becac0c4c663b2c0a3e1a213d07ae6c8ef3b8/src/_C012.ino#L136 we are looking for " char but ... in log above there is `['1']`

uzi18, 2018-10-30:
#1985

TD-er, 2018-10-30:
> [ESPEasy/src/_C012.ino](https://github.com/letscontrolit/ESPEasy/blob/c67becac0c4c663b2c0a3e1a213d07ae6c8ef3b8/src/_C012.ino#L136)
> 
> Line 136 in [c67beca](/letscontrolit/ESPEasy/commit/c67becac0c4c663b2c0a3e1a213d07ae6c8ef3b8)
> 
>  byte pos = strValue.indexOf('"',2); 
> we are looking for " char but ... in log above there is `['1']`

Where do you see this '1' ?

uzi18, 2018-10-30:
@TD-er in debug-dev/more log from @tyomikh

TD-er, 2018-10-30:
@uzi18 Ah I see, it is indeed a mismatch in quotes.
That will never match.

Maybe we should match the line with a very simple parser instead of using the string functions.
But a simple work-around can be to:

```c++
        byte pos = strValue.indexOf('"',2);
        if (pos == -1)
                pos = strValue.indexOf('\'',2);
```

uzi18, 2018-10-30:
we can also check 2nd char and search for it

uzi18, 2018-10-31:
https://blynkapi.docs.apiary.io/#reference/0/get-pin-value/get-pin-value

docs states with ["1"], but we receive ['1']

tyomikh, 2018-10-31:
Hi. I installed 20181031. No loop now.
Generic http controller to blynk-cloud.com/***/update?v0=26 works as before - sends several identical HTTP packets in parallel (BTW why?).
Blynkget 2,1,get/V2 doesn't write dummy dev variable, it is not changed

513627: Command: blynkget
513628: Read settings: ControllerSettings index: 1
513638: HTTP : C012 connecting to 139.59.206.133:80
513725: GET /*********************/get/V2 HTTP/1.1^^ Host: 139.59.206.133 ^^ Connection: close^^^^
513727: HTTP : closing connection (012)
513845: Command: event
513846: EVENT: NoChange
513903: EVENT: NoChange Processing time:57 milliSeconds
513909: Command: timerset

The rule didn't change:
on Rules#Timer=1 do
BlynkGet 2,1,get/V2
if [dummy#switch]<>[dummy#switchPrev]
if [dummy#switch]=1
event,ON
GPIO,5,1
endif
if [dummy#switch]=0
GPIO,5,0
event,OFF
endif
TaskValueSet 2,2,[dummy#switch] // renew switchPrev
TimerSet,2,5 // update uptime in Blynk cloud in a while
TimerSet,3,10 // update timestamp in Blynk cloud in a while
else
event,NoChange
endif
TimerSet,1,30
endon

802425: DHT : Temperature: 26.00
802425: DHT : Humidity: 18.00
802428: EVENT: DHT11#Temperature=26.00
802480: EVENT: DHT11#Temperature=26.00 Processing time:51 milliSeconds
802481: EVENT: DHT11#Humidity=18.00
802527: EVENT: DHT11#Humidity=18.00 Processing time:46 milliSeconds
802560: HTTP : C008 connecting to 139.59.206.133:80
802718: GET /******************/update/V0?value=26.0 HTTP/1.1^^Host: 139.59.206.133^^User-Agent: ESP Easy/20102/Oct 31 20
802864: HTTP : C008 Success! HTTP/1.1 200 OK^
802865: HTTP : C008 closing connection
802975: HTTP : C008 connecting to 139.59.206.133:80
803070: GET /******************/update/V0?value=26.0 HTTP/1.1^^Host: 139.59.206.133^^User-Agent: ESP Easy/20102/Oct 31 20
803151: HTTP : C008 Success! HTTP/1.1 200 OK^
803151: HTTP : C008 closing connection

I have max retries = 3 and max queue depth = 1. generic http sends 2 packets. is it by design or for reliability?

uzi18, 2018-10-31:
why You are useing generic http controller?

tyomikh, 2018-10-31:
I'm using generic HTTP to send updates to blynk-cloud. I started working with blynk on main branch, then switched to C012 on test branch because I wanted to read from blynk-cloud

tyomikh, 2018-10-31:
No it doesn't

3095297: HTTP : C012 connecting to 139.59.206.133:80
3095409: GET /***/update/V1?value=30.0 HTTP/1.1^^ Host: 139.59.206.133 ^^ Connection: close^^^^
3095411: HTTP : closing connection (012)
3095825: Read settings: ControllerSettings index: 1
3095834: Read settings: ControllerSettings index: 1
3095842: HTTP : C012 connecting to 139.59.206.133:80
3095953: GET /***/update/V2?value=35 HTTP/1.1^^ Host: 139.59.206.133 ^^ Connection: close^^^^
3095955: HTTP : closing connection (012)

[image]

same with IDX=0

TD-er, 2018-10-31:
I don't know what you try to say with the last screenshot + remark about IDX=0.
That index can be used as an extra parameter for a controller.
For example in Domoticz, it is used to identify the sensor data.

uzi18, 2018-10-31:
I need to check this on my own Blynk.

tyomikh, 2018-10-31:
My local Blynk server:
***/update/V0 works via C012
Blynkget 2,1,get/V2 doesn't work as expected: json returned but not put into dummy dev variable

uzi18, 2018-10-31:
Works for me on local server, update value via controller and also blynkget command sets value of dummy.
Did not tested with blynk-cloud.com server.

uzi18, 2018-10-31:
@TD-er @tyomikh I think it is problem with command queue, timers also does not work realible.

tyomikh, 2018-10-31:
I can confirm check ACK works both on local server and blynk-cloud, both update and get commands though it takes time and variable assigned from the 2nd attempt.
Thank you very much!!

uzi18, 2018-10-31:
ok so we can close, if You interested in issues with timers etc. feel free to fill new issue.

tyomikh, 2018-10-31:
Thank you. Please close.

PR #1984 (2018-10-30): BlynkGet fix loop #1951
this commit fix #1951
commit 0e581f02 (2018-10-30): BlynkGet fix loop #1951

this commit fix #1951