letscontrolit/ESPEasy #1959: Delete button at Notifications

SargonofAssyria, 2018-10-25:
At Controllers and Devices there is a 'Delete' button in the web interface.
That button is missing at Notifications.
So Notifcations can be disabled but not deleted.

TD-er, 2018-10-25:
You can still edit the notification setting and set the combobox on the top back to "none"

I agree it is not the same routine as the rest, but still you can "delete" notifications.

SargonofAssyria, 2018-10-26:
Tnx. That worked.
So very minor thing. If there is time left, create the delete notifications button, then the webui is consistant

TD-er, 2018-10-26:
Yep we will make the frontend more consistent.
We're already working on a new frontend, so I will not change it in this one, but we will for sure look into making things more consistent.

Since there will be a new frontend, I will close this issue for now.

closed 2018-10-26 by TD-er