letscontrolit/ESPEasy #1960: Sonoff 4ch Pro R2 DHT22 no rating after reboot

norimikan, 2018-10-25:
I'm problem in Sonoff 4ch Pro R2 with DHT22.

Use Hardware : Sonoff 4ch Pro R2
Use ESPEasy : ESP_Easy_mega-20181025_normal_ESP8266_1024.bin

Problem : After reboot from WebUI tool > Reboot or auto reboot(hugup?),
                 DHT22 show Nan Temp and Hum.

Clean install : yes (blank_1MB.bin)

Use controller : Domoticz MQTT

Try install other version : ESP_Easy_mega-20180910_normal_ESP8266_1024.bin
                                          same with ESP_Easy_mega-20181025_normal_ESP8266_1024.bin
                                          problem.

GPIO connect DHT22 : GPIO2(D4)

Unplug power off and re-plugin power, DHT22 is show nomal.

Screen shot "Device show nomal"
[image]

Screen shot "Device after reboot"
[image]

sysinfo : 
Build:⋄
20102  - Mega
Libraries:⋄
ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2
GIT version:⋄
mega-20181025
Plugins:⋄
46 [Normal]
Build time:⋄
Oct 25 2018 02:16:27
Binary filename:⋄
ESP_Easy_mega-20181025_normal_ESP8266_1024.bin

Log : 
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: Webserver: start
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: MQTT : Intentional reconnect
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: Read settings: ControllerSettings index: 0
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: MQTT : Connected to broker with client ID: ESPClient_DC:4F:22:C6:BE:28
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: Subscribed to: domoticz/out
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: EVENT: MQTT#Connected
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Oct 25 21:04:03 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: EVENT: MQTT#Connected Processing time:93 milliSeconds
**Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: DHT  : Temperature: 29.70
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: DHT  : Humidity: 56.40**
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: EVENT: Outside-temp#Temperature=29.70
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: EVENT: Outside-temp#Temperature=29.70 Processing time:83 milliSeconds
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: EVENT: Outside-temp#Humidity=56.40
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: EVENT: Outside-temp#Humidity=56.40 Processing time:79 milliSeconds
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: Read settings: ControllerSettings index: 0
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy:  Domoticz: Sensortype: 2 idx: 14 values: 29.70;56.40;1
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: MQTT : {"idx":14,"RSSI":8,"nvalue":0,"svalue":"29.70;56.40;1"}
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: NTP  : NTP host 0.asia.pool.ntp.org (210.173.160.87) queried
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: NTP  : NTP replied: 100 mSec
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 420 min
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: EVENT: Time#Initialized
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Oct 25 21:04:04 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
[... 46 lines cut ...]
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 12736 chunk size:400
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 14520 chunk size:400
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 14520 chunk size:400
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 14520 chunk size:400
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 14480 chunk size:400
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 14496 chunk size:400

**Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy:      : **Rebooting**...**
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 15064 chunk size:141
Oct 25 21:04:28 Sonoff 4ch Pro R2 EspEasy: sendcontent free: 15048 chunk size:0
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: Webserver: start
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: MQTT : Intentional reconnect
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: Read settings: ControllerSettings index: 0
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: MQTT : Connected to broker with client ID: ESPClient_DC:4F:22:C6:BE:28
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: Subscribed to: domoticz/out
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: EVENT: MQTT#Connected
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: EVENT: MQTT#Connected Processing time:90 milliSeconds
**Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: DHT  : No Reading**
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: NTP  : NTP host 0.asia.pool.ntp.org (103.47.76.177) queried
Oct 25 21:04:34 Sonoff 4ch Pro R2 EspEasy: NTP  : NTP replied: 40 mSec

Best Regards.

TD-er, 2018-10-25:
Nothing to do with this issue, but can you also post the flash info of this 4ch pro? (just to check if it has 1 MB or 4 MB flash)

This issue about the DHT22 does sound a bit familiar.
I will try to find the related issue (if there is one)

norimikan, 2018-10-26:
Dear TD-er

Flash info is this one?
[image]

Best Regards.

TD-er, 2018-10-26:
Just as verification, the unit uses 1MB flash.
Thanks for that.

But what I don't understand is that it states "Flash IDE size" of 8 MB.
Is this a nightly build we made?
The settings for creating the image are wrong.
I don't yet understand if that will make it the cause of the problems you're facing, but it is something we have to look into.
The "Normal" build should be using `-Tesp8266.flash.1m128.ld` as partition layout.

norimikan, 2018-10-27:
I'm misstake in install command.
write 8MB in esptool.
change to 1MB is this file.
[image]

But this DHT22 problem is same.

Best Regards.

TD-er, 2018-10-27:
Ah, that makes sense :)
It also makes sense it has nothing to do with the displayed flash IDE size.

I can have a look at the code ofcourse, but I am not sure I have the DHT22 here to test.
So it may take a few iterations to get this fixed.

norimikan, 2018-11-01:
I'm check sonoff 4ch pro R2 ESP board.
Use ESP8285.

I try install ESP_Easy_mega-20181027_normal_ESP8285_1024.bin.
But this problem is same.

for information.

TD-er, 2018-11-19:
> I'm trying Use GPIO3 and after reboot work fine.
> 
> And use GPIO1 and GPIO3 to BME280(I2C), after reboot work fine too.
> 
> Maybe problem in GPIO2?

Yep it is indeed a problem.
See also https://github.com/letscontrolit/ESPEasy/issues/2062#issuecomment-439676249

TD-er, 2018-11-19:
If I'm not mistaken, this is no longer an issue, but needs some proper documentation.
See the linked issue in my last post.
So I will now close this one.

closed 2018-11-19 by TD-er