letscontrolit/ESPEasy #1961: Feature: Add Wifi light sleep

s0170071, 2018-10-25:
It seems that the light sleep function finally works.
https://github.com/esp8266/Arduino/pull/5210
It should be added to ESPEasy

ascillato, 2018-10-25:
Yes, it works great and it drops the power consumption a lot just using sleep to 1 ms for each main loop. Also you can touch the esp8266 and feel that it is cooler.

TD-er, 2018-10-25:
When will this be included in the PlatformIO builds?
Or should I use the 'stage' version for this?

I prefer to use a single core version to make sure errors are reproducible as possible.

ascillato, 2018-10-25:
You need to use the STAGE version for that. Also take into account that Platformio will download just once the Arduino core. Then, for new updates to the STAGE you need to overwrite the files in platformio with the new STAGE version (C:\Users\Usuario\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910)
In platformio.ini you need to use platform = https://github.com/platformio/platform-espressif8266.git#feature/stage

You can add to the scheduler a delay(sleep), so you can configure at runtime how much time will be for your sleep.

If using 1, you didn't notice any lag.
If you use 50, you will notice some lags with pushbuttons
if you use 200, you will notice also some reconnections to mqtt. But wifi will work all the time. no disconnections.

ascillato, 2018-10-25:
Yes, exactly. 

Delay(0) is like yield(). Works fine.

Sleep feature is meant to be used when your device is idle or when you know that nothing will happen or when you need to lower the power consumption.

For example, one user did a rule to set sleep to 0 when his ws2812 led strip was on, and another rule to set sleep to 10 when the led strip was off. So, for your eyes, the led was working without any lag, but when it is off, the power consumption was very low.

And yes, of course, it is a great idea to make sleep to be automatic by the scheduler. But also, for some cases, might be needed to be manually set by the user.

s0170071, 2018-10-28:
@ascillato I tried the current core and found that the ESP gets a fluctuating power consumption.
Using 
```
WiFi.setSleepMode(WIFI_LIGHT_SLEEP);
while(1) delay(10000);

```
it consumes mostly 5mA, sometimes 20mA, peaks to 66mA. On average14mA.

I followed the track down and tried the nonos SDK example: 
```
void fpm_wakup_cb_func1(void)
{
   wifi_fpm_close();     
   wifi_set_opmode(STATION_MODE);       
   wifi_station_connect();    
}
void loop() {
    wifi_station_disconnect();
   wifi_set_opmode(NULL_MODE);       
   wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);  
   wifi_fpm_open();               
   wifi_fpm_set_wakeup_cb(fpm_wakup_cb_func1); // Set wakeup callback
   wifi_fpm_do_sleep(0xFFFFFFF); 
delay(1000);
}

```
That one does actually sleep at 4mA (of which 3 may be the remainder of my circuitry) but only if I sleep 0xFFFFFFF us.
I found this bug is described too, but I thought it was removed by the latest core PR. 
Am I missing something ? Do I need to manually update some SDK files ?
Can you maybe provide a very simple sleep example ?

s0170071, 2018-10-28:
I pretty much spend the weekend playing with sleep modes. 
With the NonOS SDK functions, the different Arduino core versions, automatic sleep and so on. 

**Long story short: forget it.** 

Some devices go to auto modem sleep if you delay(). Some just don't, especially if you have been experimenting with wifi_fpm_open() calls before. There is some persistent aspect to it. 
(Auto) light sleep does not with one exception: You have to use the infinite version, wake it by GPIO and then it either crashes or does not enter sleep ever again.
I just wonder why many users have been cheering here esp8266/Arduino#5210 - yet there is not a single proof of concept piece of code around.

Closing since this not any any where near stable...

closed 2018-10-28 by s0170071

ascillato, 2018-10-28:
Hi,

Tasmota uses sleep with no issues with core 2.3.0 and with core STAGE.

The code for using it is just delay(sleep) where sleep is from 0 to 200 msec.

The code to activate sleep feature for delay command is:

```
/* Excerpt from the esp8266 non os sdk api reference (v2.2.1):
 * Sets sleep type for power saving. Set WIFI_NONE_SLEEP to disable power saving.
 * - Default mode: WIFI_MODEM_SLEEP.
 * - In order to lower the power comsumption, ESP8266 changes the TCP timer
 *   tick from 250ms to 3s in WIFI_LIGHT_SLEEP mode, which leads to increased timeout for
 *   TCP timer. Therefore, the WIFI_MODEM_SLEEP or deep-sleep mode should be used
 *   where there is a requirement for the accurancy of the TCP timer.
 *
 * Sleep is disabled in core 2.4.1 and 2.4.2 as there are bugs in their SDKs
 * See https://github.com/arendst/Sonoff-Tasmota/issues/2559
 */

// Sleep explanation: https://github.com/esp8266/Arduino/blob/3f0c601cfe81439ce17e9bd5d28994a7ed144482/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L255
#if defined(ARDUINO_ESP8266_RELEASE_2_4_1) || defined(ARDUINO_ESP8266_RELEASE_2_4_2)
#else  // Enabled in 2.3.0, 2.4.0 and stage
  if (sleep) {
    WiFi.setSleepMode(WIFI_LIGHT_SLEEP);  // Allow light sleep during idle times
  } else {
    WiFi.setSleepMode(WIFI_MODEM_SLEEP);  // Disable sleep (Esp8288/Arduino core and sdk default)
  }
#endif
```

s0170071, 2018-10-28:
@ascillato Tried that auto sleep, but that's just giving an average of 14 to 20mA just delay()ing. Should be 1mA though.

ascillato, 2018-10-28:
The Wifi light sleep will make the cpu to sleep but will awake it depending on the DTIM configuration to answer the wifi broadcast. If you don't specify the DTIM, will use 3 DTIM as default. DTIM is configured in your router. How many DTIMs to use are configured in your WiFi.setSleepMode(WIFI_LIGHT_SLEEP, DTIM); 
DTIM can be from 1 to 10 (wifi standard). more than that is set to 10 by the Arduino core.

Sleep explanation: https://github.com/esp8266/Arduino/blob/3f0c601cfe81439ce17e9bd5d28994a7ed144482/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L255

So, will not be a deep sleep. It will be sleeping and awaking to mantain the wifi connection and to receive requests.