letscontrolit/ESPEasy #1962: Feature request:  support for Gosund SP1

micropet, 2018-10-26:
Is a support of Gosund SP1 planned for ESPEasy.

It is a switchable socket with 16A switching capacity.
It has a voltage and power measurement and costs about 12 EUR in Germany.

[https://www.amazon.de/Intelligente-Stromverbrauch-Fernsteurung-Schaltbare-kompatibel/dp/B0777BWS1P/ref=sr_1_1?ie=UTF8&qid=1540524622&sr=8-1&keywords=Intelligente+Steckdose+2er+Pack%2C16A+3680W%2C+Stromverbrauch+messen+Timer](url)

Infos:
[https://www.bastelbunker.de/gosund-sp1-mit-tasmota/](url)

Other Names:
Gosund SP1
Blitzwolf BW-SHP2
HomeCube
Wuudi
CNXUS
TECKIN
VANZAVANZU
Ablue
Yosoo
Texsens
Expower
Potok
Geekbes
Swiftrans
TFlag
BricoPlus

Grovkillen, 2018-10-26:
Looks like a good product. We will look into it, need the hardware though.

TD-er, 2018-10-26:
The Blitzwolff shp version is already on my desk, waiting to be implemented

micropet, 2018-10-26:
Yes, good news!

micropet, 2018-10-27:
So, my 2 Gosund SP1 are arriving today.
I have a 5 pin female connector installed.
Now I do not need to screw on the case for flashing.

The space for the connector is ideal.

[image]

[image]

TD-er, 2018-10-27:
Probably not for you, but just as a general warning when someone else finds this issue + photo.
Do not flash/connect the node to a PC and leave it plugged in the mains.

Also I am not sure about the safety aspect when you unplug the node and press with your fingers on the pins.
You may get a distance less than 3 mm from your skin to the exposed pins and I'm not sure these are isolated from ground.

micropet, 2018-10-27:
I think it's okay.
All my S20 have the connector.
You can quickly flash a new version without opening the case.

And of course the node is disconnected from the mains voltage when it is flashed.

s0170071, 2018-10-29:
Here's an image of  ESP module the pinout: 
https://github.com/arendst/Sonoff-Tasmota/issues/2223

ingoiot, 2019-08-01:
is there a working image for the Gosund or compatible plugs now?

TD-er, 2019-08-02:
There is an 1M build with the same plugins as the Sonoff POW.
But I guess it is too big to perform an OTA, which is rather a pity with these modules since they are so hard to upgrade via serial.

ingoiot, 2019-08-03:
i could flash the 1M version, but unfortunately it shows no values.

TD-er, 2019-08-03:
Do you know the chip and GPIO pins used?

ingoiot, 2019-08-03:
the plug works with tasmota Gosund SP1 template.
in hope this pic is enough info:

[image]

TD-er, 2019-08-03:
Not really, it looks like 2 chips are used?

ingoiot, 2019-08-03:
i can only find a chip labled with bl0937.

i think they might just not have changed the pin names from the old sensor?

[image]

uzi18, 2019-08-03:
it looks like old sonoff pow with power to frequency converter, so we need just to know pins and multiplers in equations

ingoiot, 2019-08-03:
found this in Tasmotas sonoff_template.h
maybe that helps - not shure where the calculating is done..
```
  { "Gosund SP1 v23",  // https://www.amazon.de/gp/product/B0777BWS1P
     0,
     GPIO_LEDLNK_INV,  // GPIO01 Serial RXD and LED1 (blue) inv - Link status
     0,
     GPIO_KEY1,        // GPIO03 Serial TXD and Button
     GPIO_HJL_CF,      // GPIO04 BL0937 or HJL-01 CF power
     GPIO_NRG_CF1,     // GPIO05 BL0937 or HJL-01 CF1 current / voltage
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
     GPIO_LED1_INV,    // GPIO13 LED2 (red) inv - Power status
     GPIO_REL1,        // GPIO14 Relay (0 = Off, 1 = On)
     0, 0, 0
  },
```

uzi18, 2019-08-03:
that's described in datasheet

TD-er, 2019-08-04:
In the old POW we read some default multipliers from the module which can later be corrected in a calibration.
[image]

These are the settings used on my POW r1 so they should at least give some voltage value around 230V (if you live in 230V area)

ingoiot, 2019-08-04:
i just flashed tasmota to check the hardware -seems to work.
flashed back ESP_Easy_mega-20190803_hard_other_POW_ESP8285_1M.bin

no matte what preset i use, there are no values.

 with the Tekin us or Blitzwolf preset i get some weird values,
P076: Read values - V=0 - A=0.00 - W=7123 - Pf%=100

with the others it looks like:
P076: Read values - V=0 - A=0.00 - W=0 - Pf%=0

and the switch doesnt work on any.

TD-er, 2019-08-04:
Can you make a screenshot of the used settings in ESPeasy?

ingoiot, 2019-08-04:
only the device tab?
suddenly after a reboot i had some (wrong) values in Voltage.
i changed the voltage multiplier and it was back to zero.
even if i changed everything back.

[image]

> 453873: P076: Read values - V=316 - A=0.00 - W=0 - Pf%=0
456792: SaveToFile: free stack after: 2752
456792: P076: Saved Calibration from Config Page
456792: P076: Multipliers Reassigned
456792: P076: PIN Settings  curr_read: 0 cf_edge: 2 cf1_edge: 3
456795: SaveToFile: free stack: 3200
457110: FILE : Saved config.dat
457111: SaveToFile: free stack after: 3200
457118: SaveToFile: free stack: 3136
457880: FILE : Saved config.dat
457880: SaveToFile: free stack after: 3136
457893: P076: Init object done
457894: P076: Init Basic Resistor Values done
457899: P076: Saved Calibration after INIT
457899: P076: Applied Calibration after INIT
459281: P076: Read values - V=0 - A=0.00 - W=0 - Pf%=0
461379: P076: Read values - V=0 - A=0.00 - W=0 - Pf%=0

>

TD-er, 2019-08-04:
You have not set the GPIO pins like in the Tasmota config you quoted.

```
GPIO_HJL_CF,      // GPIO04 BL0937 or HJL-01 CF power
GPIO_NRG_CF1,     // GPIO05 BL0937 or HJL-01 CF1 current / voltage
GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
```

TD-er, 2019-08-04:
After you change the device combo box, you need to save the settings and look at the page again.
It will then base the GPIO pins on a predefined config.
Otherwise you must use the 'custom' config and set the GPIO pins manual

ingoiot, 2019-08-04:
to me i t looks like the same

[image]

only in tasmota ist called SEL_INV,in espeasy only SEL
dont know if thats different.

TD-er, 2019-08-04:
SEL_INV suggests the signal may be inverted?
Normally the I2C is also located on GPIO-4 and -5.
Make sure those pins are re-assigned to some other pins.

ingoiot, 2019-08-04:
now im getting something.
i disabled i2c in the hardware menu, i think its not needed?

unfortunately the values are all over the place and not consistent off by a certain value.

> 65090: P076: Read values - V=318 - A=0.00 - W=0 - Pf%=0
65093: EVENT: 937#Voltage=318.00
65102: EVENT: 937#Current=0.00
65110: EVENT: 937#Power=0.00
65123: EVENT: 937#PowerFactor=0.00

> P076: Read values - V=5675 - A=0.00 - W=0 - Pf%=0
11216: EVENT: 937#Voltage=5675.00
11228: EVENT: 937#Current=0.00
11237: EVENT: 937#Power=0.00
11245: EVENT: 937#PowerFactor=0.00

i added  a rule to switch on the realis on boot, 
`on System#Boot do
GPIO,14,1
timerSet,1,3
endon `
(copied from the website) - it turns on but not off after 30 sec. as i expected...

anyway, it seems like only voltage is measured.
it looks like this must be altered with the CF1 input by the software.
maybe thats why the value are so different.

> P076: Read values - V=318 - A=0.00 - W=2425 - Pf%=100
19983: EVENT: 937#Voltage=318.00
19992: EVENT: 937#Current=0.00
20000: EVENT: 937#Power=2425.00
20008: EVENT: 937#PowerFactor=100.00

TD-er, 2019-08-04:
Yep, looks like the signal for the SEL line should be used inverted.
That's a bit more than just a very quick fix, so I will not alter it right now.

uzi18, 2019-08-04:
it looks like hlw8012 lib does not support BL0937 chip correctly (inverted sel)

TD-er, 2019-08-04:
I'm not sure if inverted signal is by design of that chip.
And it is just an assumption based on the description of the pin.
Knowing Theo, he wouldn't just add that for fun or it must be to start messing with us here at ESPeasy ;)

uzi18, 2019-08-04:
@TD-er maybe it is hacked somwhere else ...
my wrong it must be used like this:
x.begin(CF_PIN, CF1_PIN, SEL_PIN, CURRENT_MODE, true);
where:
CURRENT_MODE is LOW for HLW8012,HJL-01
and HIGH for BL0937
should be easy fix

uzi18, 2019-08-05:
@TD-er  BL0937 has different Vref, so multiplers must be different here

ingoiot, 2019-08-05:
doesnt really work

> 33457: P076: Read values - V=60 - A=0.00 - W=2334 - Pf%=100
33460: EVENT: 937#Voltage=60.00
33470: EVENT: 937#Current=0.00
33479: EVENT: 937#Power=2334.00
33487: EVENT: 937#PowerFactor=100.00
36562: P076: Read values - V=60 - A=0.00 - W=1161 - Pf%=100
36565: EVENT: 937#Voltage=60.00
36575: EVENT: 937#Current=0.00
36584: EVENT: 937#Power=1161.00
36592: EVENT: 937#PowerFactor=100.00
39762: P076: Read values - V=29 - A=0.00 - W=1155 - Pf%=100
39765: EVENT: 937#Voltage=29.00
39775: EVENT: 937#Current=0.00
39783: EVENT: 937#Power=1155.00
39791: EVENT: 937#PowerFactor=100.00

when i double the load it doubles the power  - so far so good.

but still no current

uzi18, 2019-08-05:
@ingoiot  please give us screenshot of entire task configuration, thanks

ingoiot, 2019-08-05:
first i choose the gosund preset, than i change to custom and alter current to high.

[image]

TD-er, 2019-08-05:
Have you tried changing the "Current (A) reading" ?

uzi18, 2019-08-05:
@ingoiot ok - try like this:
Device = custom
Current Reading = LOW
CF and CF1 edge = FALLING
and paste here some logs - what kind of load you connected to plug (how many watts)?

I will try to recalculate Multipliers, but it is not strightforward

@TD-er more digging more questions ... 
- HLW gives 50% PWM - but we use freq
- BL gives 38uS signal - but we use only freq (change interupt is not ok here)
- Multipliers from HLW are not correct for BL.
- We change mode in plugin but in "interrupt mode" it is changed automaticly after measure, it is only needed on pooling mode.
- pulse_timeout is also not good for BL

uzi18, 2019-08-05:
@TD-er Current (A) reading LOW should be correct

ingoiot, 2019-08-05:
Its an
[image]
 infrared heater

> 
gosund
295v
0,00A
1152,00W
100PF

Energymeter
228,6V
1,7A
391W
98PF

gosund
293V
0,00A
2312W
100pf

Energymeter
227,8V
3,46A
786W
99pf

157482: P076: Read values - V=295 - A=0.00 - W=1152 - Pf%=100
157485: EVENT: 937#Voltage=295.00
157498: EVENT: 937#Current=0.00
157506: EVENT: 937#Power=1152.00
157514: EVENT: 937#PowerFactor=100.00
160587: P076: Read values - V=297 - A=0.00 - W=1154 - Pf%=100
160590: EVENT: 937#Voltage=297.00
160602: EVENT: 937#Current=0.00
160610: EVENT: 937#Power=1154.00
160618: EVENT: 937#PowerFactor=100.00
163782: P076: Read values - V=251 - A=0.00 - W=2361 - Pf%=100
163785: EVENT: 937#Voltage=251.00
163798: EVENT: 937#Current=0.00
163806: EVENT: 937#Power=2361.00
163815: EVENT: 937#PowerFactor=100.00
166887: P076: Read values - V=293 - A=0.00 - W=2360 - Pf%=100

TD-er, 2019-08-05:
As long as the Amp reading is 0, I don't think we can compute proper values

uzi18, 2019-08-05:
please paste also sysinfo, are you able to test custom image?
have bought one of these gosund sp1, will. be at the end of week in. my hands =)

ingoiot, 2019-08-05:
> 

Build:⋄
20103  - Mega
System Libraries:⋄
ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20190803
Plugins:⋄
7 [Normal] [Sonoff POW R1/R2]
Build Time:⋄
Aug  3 2019 02:44:22
Binary Filename:⋄
ESP_Easy_mega-20190803_hard_other_POW_ESP8285_1M.bin

> are you able to test custom image

i thought i already am^^

i have solderd on wires to flash

uzi18, 2019-08-05:
try multipliers:
current 25740
voltage 313400
power 3414290

uzi18, 2019-08-06:
@ingoiot any update?

ingoiot, 2019-08-06:
just back from work.
looks promising, but power still missing

the voltage jumps around qiet a bit

> 

02440: EVENT: 937#Voltage=228.00
102454: EVENT: 937#Current=0.00
102462: EVENT: 937#Power=387.00
102471: EVENT: 937#PowerFactor=100.00
105632: P076: Read values - V=214 - A=0.00 - W=386 - Pf%=100
105636: EVENT: 937#Voltage=214.00
105649: EVENT: 937#Current=0.00
105657: EVENT: 937#Power=386.00
105665: EVENT: 937#PowerFactor=100.00
108833: P076: Read values - V=228 - A=0.00 - W=388 - Pf%=100
108838: EVENT: 937#Voltage=228.00
108868: EVENT: 937#Current=0.00
108877: EVENT: 937#Power=388.00
108885: EVENT: 937#PowerFactor=100.00
111964: P076: Read values - V=225 - A=0.00 - W=388 - Pf%=100
111967: EVENT: 937#Voltage=225.00
111982: EVENT: 937#Current=0.00
111990: EVENT: 937#Power=388.00
111998: EVENT: 937#PowerFactor=100.00

uzi18, 2019-08-06:
@ingoiot ok but this is another issue that need fix

uzi18, 2019-08-06:
@micropet @Grovkillen @s0170071 please test #2534 fix?

uzi18, 2019-08-08:
@TD-er have received my gosund sp1 so now need to reflash it with espeasy somehow, maybe I will try to upgrade esp module to esp8266 with 4MB flash from nodemcu.
After that will try to do some cleanup and maybe some improvements here :)

as #2534 merged, we have working hardware

little to do:
- cleanup
- stbilize reading / simple filtering
- add predefinied multipliers
- add suggestion/warning about calibration
- something more?

TD-er, 2019-08-08:
Documentation :)

coelner, 2020-04-06:
Hello, documentation sounds perfect. It is not simple to follow the different settings over the different pcb versions. Which pcb version do you support? I'm searching for V3.3

dersparkiie, 2020-05-20:
I would like to use an SP111 with ESPEasy.
Which build should I try? These didn't work:
ESP_Easy_mega-20200426_hard_other_POW_ESP8285_1M
ESP_Easy_mega-20200426_hard_SONOFF_POW_4M1M

TD-er, 2020-05-21:
@uzi18 Maybe you can help out @dersparkiie with this one?

dersparkiie, 2020-06-16:
No version possible?

dersparkiie, 2020-12-06:
Nothing new?

Sminter4, 2022-01-24:
Hello everyone, I would like to run espeasy on the Gosund SP1 instead of Tasmota.
Unfortunately, the ...ESP8285_1M... does not include the "Energy (AC) - HLW8012/BL0937" plugin.
Can someone recommend me the appropriate BIN?

uzi18, 2022-01-24:
hard_other_POW_ESP8285_1M should be ok for SP1 and SP111
[ESP_Easy_mega_20220124_hard_other_POW_ESP8285_1M.bin.zip](https://github.com/letscontrolit/ESPEasy/files/7925552/ESP_Easy_mega_20220124_hard_other_POW_ESP8285_1M.bin.zip)

uzi18, 2022-01-24:
@dersparkiie @Sminter4  hard_other_POW_ESP8285_1M should be ok for SP1 and SP111
[ESP_Easy_mega_20220124_hard_other_POW_ESP8285_1M.bin.zip](https://github.com/letscontrolit/ESPEasy/files/7925552/ESP_Easy_mega_20220124_hard_other_POW_ESP8285_1M.bin.zip)
unzip it before flash :)

Sminter4, 2022-01-24:
Thank you for your quick response. I tried your file. However, only the P076 plugin is included here. I need the plugin P075 for the chip "BL0937" for the SP1. Is there a bin based on ESP8285_1M that includes plugin P075?

tonhuisman, 2022-01-24:
You can use an ESP8266 bin file on the ESP8285, they are compatible.

Plugin P076 supports both the HLW8012 and the BL0937.
Plugin P075 is a Nextion display, I guess you are pointing at Plugin P077 for CSE7766, that should also be included in that 'POW' build.

tonhuisman, 2022-01-24:
There you go, a POW build that should have P076 included (can't check atm, not at my normal workstation)

[ESP_Easy_mega_20220124_hard_other_POW_ESP8285_1M.zip](https://github.com/letscontrolit/ESPEasy/files/7926783/ESP_Easy_mega_20220124_hard_other_POW_ESP8285_1M.zip)

tonhuisman, 2022-01-25:
I didn't re-read this thread, but I think to remember that one of the issues is the unstable readouts, but I don't know how to fix that.

[3 more replies]

PR #3929 (2022-03-31): [P076] Build: Restore plugin in POW builds
Resolves #1962

TODO:
- [x] Testing (waiting for [confirmation](https://github.com/letscontrolit/ESPEasy/issues/1962#issuecomment-1021384140) that the issue is solved)
- [x] Check if more (other) work is desired, ~like a predefined config option~
  - ~Add averaging of measurements?~ (Separate, generic, project)
closed 2022-03-31 by TD-er