letscontrolit/ESPEasy #1963: Feature request... DS18B20 plugin enhancement. Alarm feature.

Budman1758, 2018-10-26:
I was just looking thru the datasheet for these little sensors and found out they have 2 user configurable alarm functions. From what I gather it will trigger an alarm output whenever the set temp is reached.  I was thinking that would be a pretty handy thing to have available.  I realize that the same thing can be done with rules but if it were available in the plugin it would simplify things quite a bit. 

I kind of envision an event triggered when the alarm is activated.  
I know there is a lot of other work that's more important going on now so this can go on the back burner. Just throwing this out there for folks to dwell on.

Thoughts?

Grovkillen, 2018-10-26:
we want to add more event based rules handling and thus it would make sense of making use of such an feature within the plugin.

TD-er, 2018-10-26:
There are more sensors capable of sending an alarm when reaching some threshold.
So an "alarm" event would make sense.

tonhuisman, 2023-09-03:
This issue is no longer relevant, and can be closed.

TD-er, 2023-09-04:
Not 100% about closing this one...
The "alarm" feature itself hasn't been added.
We do have lots of event handling and for plugins that do read way more often than the task interval, we also record the max/min values, which can be queried in rules.  (part of the "stats")
Those can be queried on a loop, so when something happens, you can act on it.
It isn't as responsive as firing an alarm event, but can be checked at sub-second intervals.

We don't read the DS18b20 sensors more often then the "Interval", which would have been a requirement to read the alarm state anyway.

So closing this issue as it is possible to act on extreme values.

closed 2023-09-04 by TD-er