letscontrolit/ESPEasy #1964: Q: Ambient light & proximity sensor?

Tom-Bom-badil, 2018-10-26:
Quick question, as I didn't find anything suitable while browsing the playground: Did anyone already implement a plugin for a combined ambient light & proximity sensor, like e.g. CJMCU-9900 or VCNL4010?

One of my next projects will be to build an ESPEasy device that scrolls °C/%rH/Pa/Datetime on a display (so far, so easy - have 5 of them up and running in my house). But it should dim down the display at night time (the darker, the less brightness); and switch back to full brightness for ~30secs if there is motion in short distance (e.g. waving with hand in front of device).

Any hints, or do I need to start from scratch? Thanks!

/tom

Tom-Bom-badil, 2018-10-26:
**_Veeerrryyyy_** cool Hans, thanks for sharing. This will make the project mucheasier - much appreciated! :)
/tom

Tom-Bom-badil, 2018-11-10:
I can confirm the behaviour of 'P064 - Gesture APDS9960' as mentioned by Hans above. Ordered 4 different sensors from 4 different makers, none of them is working properly.

All is good as long as the sensor is just connected to I²C. As soon as the sensor is enabled, the system stalls (tested on Sonoff 8285, NodeMCU und ESP32). One NodeMCU couldn't be recovered at all due to boot loop and needed to be restored. Most other devices were ok after reboot, but showing ridiculous values without any proximity/light/gesture detection. After a while, systems stalled again.

Therefore, APDS9960 should be considered broken.

giddyhup, 2018-11-14:
I can confirm the erratic behavior of the APDS9960 sensor. I could use one relatively reliably with a Raspberry Pi but it is hit or miss with an ESP8266.

TD-er, 2019-05-27:
Not sure why this issue is closed, since it apparently is still broken.

tonhuisman, 2022-05-07:
This is most likely fixed by #3037, and again improved by #3974, so I'll add this as an issue that it will resolve.

PR #3974 (2022-09-18): [P064] Unblock while reading gestures when object is very close
As [reported in the forum](https://www.letscontrolit.com/forum/viewtopic.php?t=9046)
Resolves #1964

Prevent the APDS-9960 gesture sensor from blocking ESPEasy when an object is (quite) close.
Changed status from DEVELOPMENT to TESTING.

TODO:
- [ ] Confirmation this improves/fixes the blocking
  - [x] Local testing proved stable for months now
- [x] Remove [TESTING] tag before merge
closed 2022-09-18 by TD-er