letscontrolit/ESPEasy #1969: Error in Arduino compile of esp-easy-mega 20181026

ozett, 2018-10-27:
is this an already known error?

D:\daten\olaf.arduino\arduino-1.6.11mega20181002\sketches\ESPEasy\_P013_HCSR04.ino: In function 'float Plugin_013_read(unsigned int)':

_P013_HCSR04:298: error: 'convert_cm_F' is not a member of 'NewPingESP8266'

     return NewPingESP8266::convert_cm_F(echoTime);

            ^

_P013_HCSR04:300: error: 'convert_in_F' is not a member of 'NewPingESP8266'

     return NewPingESP8266::convert_in_F(echoTime);

            ^

D:\daten\olaf.arduino\arduino-1.6.11mega20181002\sketches\ESPEasy\_P013_HCSR04.ino: In function 'String Plugin_013_getErrorStatusString(unsigned int)':

Image:

[image]

TD-er, 2018-10-27:
That is a local change in the newping library.
You must copy the library directory of the GitHub repo to your Arduino lib directory and not use the Arduino library installer.

There are more libraries with specific changes for espeasy.
These may be able to compile when using the official versions of the libraries, but will not work as stable.

For example the pubsublib for mqtt, has some patches which still have to be added to the official library

ozett, 2018-10-27:
the i2cdevlib was not found,
i moved it under windows arduino-ide to the user-lib-path.

that worked.

than i had the same issue with another lib.
i did the same, the error repeated with other libs.

i moved all libraries from the arduino-ide-lib-path  to the user-lib-path .
error resolved, everything compiled.

i am not experienced enough,
but you must have changed something which distrubed finding the libs under the arduino-ide-lib-path

can i do something to investigate this?

ozett, 2018-10-27:
only some warings remained:

[image]

TD-er, 2018-10-27:
I think we have to make a proper manual for compiling using Arduino IDE.
These questions pop up every now and then, caused by the rather big differences between PlatformIO and Arduino IDE.

Also we should make pull requests to the library owners when we really fix some issues.
For some libraries pull requests have been made, but even then it may take a long time (if ever) until these fixes will be merged.
Another cause of these build differences is about library versions.
With Arduino IDE it is very easy to install a newer version without noticing it.
In the PlatformIO environment, the distributed version is the preferred one.
This makes PlatformIO more consistent between builds, since you have no guarantee the used libraries are the same in the Arduino IDE while PlatformIO does force the same version as decided by the files in the repository.

ozett, 2018-10-27:
i dont think that this error is caused my too little documentation for compiling.

i normally stick with my windows-folder in wich is the arduino-ide executables in version 1.6.11
i dont update anything.

in the library subfolder of my arduino-ide-folder i put only (and only) the libs from the source of espeasy
from here: https://github.com/letscontrolit/ESPEasy/releases. i clean up the library-folder an put there in the libs which come strict with the source, so that i have **no conflicting versions**

as i compiled the old espeasy sources, there had not been such issues for me.
in the versions of espeasy-mega of the last month the issue arose.

i know there is change in the source, but there must be a clear way to code the right path to the libraries in the subfolder of the arduino-ide for the libs.

mybe something is broken in the syntax or the path-convention for the arduino-ide-libs, as far as i found in **ESPEasy-Globals.h** some includes of libs with <> or with "" an a google comment on this. Maybe mixing the syntax breaks the path to the libs for the ide?

i try to investigate this, as i am not that much experienced as a code. but i can check different this from google hints with libraries and report here later

ozett, 2018-10-27:
:see_no_evil:
:see_no_evil:
:see_no_evil:

MY Apollogies, i am sorry.

all your hints are right.
(and it dont need no more documentation, only more care on my side)

i triple-checked all the libs, folders an path and made a new clean single-folder for the new source,
checked that in the lib-subfolder are the libraries that came with the source -- :boom: and checked again that in the user-subfolder there a no other libs. 👍 

that was it. now it compiled flawless

under windows, you can copy the ide-folder and have there a unique subfolder for unique libs,
but all this separate ide-folder share the same user-folder, which as an additional folder for libs.
and the user-subfolder is shared across all separate-ide folders with there lib-subfolders.
thats the place where my mix-up of versions had happened.

i am sorry to bother you with my mistage, appologies,
but keep on coding espeasy
🥇

ozett, 2018-10-27:
just one thing about the versioning:
i use mega-20181027 to compile, but this information gets lost as i compiled it myself.
is it possible to keep it in? after some days i want to lookup which sourcebase i used, but as i compiled it myself on the info-page in the webgui the 

GIT version:⋄ | (custom)
Binary filename:⋄ | Self built!

has not the base-infomation i wanted it to have.

Grovkillen, 2018-10-27:
We inject that information into the binary file (BIN) _after_ the firmware is compiled. We do this using script, and that is why we have a set length for the BIN file name (to inject the characters at a specific binary position). You will have to set up your own environment to do this yourself OR you add your own variable that replace the "Self built!" upon compile time. Anyway you do it, you have to maintain that yourself, we will not take care of that.

ozett, 2018-10-27:
i will try myself for my single builds,
but where is your script to patch?, whats the name?

maybe some hint before i search my disk for this "scipt".. .

anyway, thanks for fast and quick reply,
👍

Grovkillen, 2018-10-27:
https://github.com/letscontrolit/ESPEasy/blob/mega/crc2.py

That is the actual script and it's triggered by this script:

https://github.com/letscontrolit/ESPEasy/blob/mega/before_deploy

ozett, 2018-10-27:
why not using Custom-sample.sh in the source?

```
/*
	To modify the stock configuration without changing the EspEasy.ino file :

	1) rename this file to "Custom.h" (It is ignored by Git)
	2) define your own settings below 
    3) define USE_CUSTOM_H as a build flags. ie : export PLATFORMIO_BUILD_FLAGS="'-DUSE_CUSTOM_H'" 	
*/
---
#ifdef BUILD_GIT
#undef BUILD_GIT
#endif

#define BUILD_GIT "MyBuildNum"

```

it looks like you made it for my purpose, didnt you?
😸

ozett, 2018-12-21:
just an late addition:
for compiling with a clean arduino-ide you only need a clean subfolder "portable".
than you have a clean IDE with only needed libs.

i found this here in the wiki, clear state:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#install-arduino-ide

tonhuisman, 2023-09-03:
Compiling the ESPEasy code using Arduino IDE is not endorsed, better use VSCode/PlatformIO as [documented](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html).
This issue can be closed.

closed 2023-09-03 by TD-er