letscontrolit/ESPEasy #1970: Fix: HTTP Header broken in 'Generic HTTP Advanced' controller

PRO2XY, 2018-10-27:
When using 'Generic HTTP Advanced' controller, if headers are specified in the HTTP Header textbox, an extra CRLF is introduced in the payload. This leads to a broken HTTP header, like so:
(Notice the extra line between `Connection: close` and `Accept: */*`)
```
POST /api/v1/PtbZQgxsg5X7R5V18xVh/telemetry HTTP/1.1
Host: 192.168.8.101
User-Agent: ESP Easy/20102/Oct 25 2018 02:19:42
Connection: close

Accept: */*
Content-Type:application/json
Content-Length: 8

{B:4.78}
```

**Fix:**
This issue can be addressed by removing the preceding two characters from the payload when appending the custom headers, in [_C011.ino (line 168)](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_C011.ino#L168).
Change the `if` block to:
```
if (strlen(customConfig.HttpHeader) > 0) {
    payload.remove(payload.length()-2);
    payload += customConfig.HttpHeader;
}
```

TD-er, 2018-10-27:
I guess these CRLF are entered somewhere in the text, so we should remove any trailing \n or \r and not just remove the last 2 characters.
If entered on a unix machine it will probably be only one character.

PRO2XY, 2018-10-29:
The payload returned by `create_http_request_auth(...)` on [Line 164](https://github.com/letscontrolit/ESPEasy/blob/ad3514e765af0e3807dfa74ffc047dde48aa260f/src/_C011.ino#L164) ends with `\r\n\r\n` which is the way it should be in case no extra headers are to be added (mark start of body). 

I would suggest to either :
1. make `create_http_request_auth(...)`  return payload ending with only one `\r\n`, and add an extra one after the entire header is built ([line 171](https://github.com/letscontrolit/ESPEasy/blob/ad3514e765af0e3807dfa74ffc047dde48aa260f/src/_C011.ino#L171)). But that might require editing other parts of where `create_http_request_auth(...)` is used.
2. Let `create_http_request_auth(...)` be as is, and remove one set of `\r\n` from the end of the payload, since we already know it ends with `\r\n\r\n`. This way the rest of the code is not affected.

TD-er, 2018-11-11:
Is this still an issue?
@uzi18 fixed a number of other HTTP header issues last week.
Perhaps this is also fixed now?

uzi18, 2018-11-11:
@RogerHartmann please test actual behaviour.

uzi18, 2018-11-13:
ok it is not fixed

TD-er, 2018-11-30:
Should be fixed.
Please close when verified to be working.

popaserge, 2019-03-11:
@TD-er 
It seems that the issue was not really fixed (see also issue 2227)
I always saw the same message in the log output: HTTP : C011 Error: HTTP/1.1 411 Length Required
So I learned how to fork mega on github, how to work with platformio, what where the rules using \r\n in http requests and using Wireshark to capture the data sent by EspEasy, I think I found a solution as my server now agrees to take the data and EspEasy receives a 200 OK.
The "Content-Length:  " field in line 183 shall be a part of the Header and separated from the previous field only by a single \r\n. If the previous header in line 174 brings a \r\n, then line 182 adds again a new \r\n and the total header is broken. It is what could be seen with Wireshark.
So duplicating line 170-172 after line 175 and suppressing line 182 gives a single \r\n in the result.
Previous
[image]
New
[image]
Result in wireshark
[image]
Result in log output
[image]
What to do now? Modify mega?Could any body else test if no problem with the modification?

TD-er, 2019-03-12:
I will make a new PR for this and also make a test build for people to test.

TD-er, 2019-03-12:
[Test build with suggested fix](https://www.dropbox.com/s/lr83pw9y0hrz5go/ESPEasy_mega-20190311-2-PR_2389.zip?dl=0)

popaserge, 2019-03-12:
@TD-er : does not work
My Esp8266 was running all the night without problem (see curves from a cloud server).
Changing nothing in rules, controllers or devices I flashed with your new ESP_Easy_mega-20190311-2-PR_2389_test_core_242_ESP8266_4M.bin and lost the success around 15h30 with the "411 Length Required" message.
Changed the server to be able to analyse with wireshark, did it, flashed again with my try and recovered the success.
Comment : 
From what I understood of my reading on the web, each field of a Header has to be on a new line and Body has to be separated from the header by an empty line
-in your source of c011.ino, you did not delete line 179 as I did: is that the difference?
-does the function removeExtraNewLine(payload); remove only the last \r\n or all of them except one?

Hereafter the screenshots
Wireshark new TD-er build : 3 times \r\n after "Connection : close"
[image]
Wireshark popaserge reflashed : only 1 \r\n after "Connection : close"
[image]
Log with popaserge reflashed
[image]
Curves this night
[image]

TD-er, 2019-03-12:
Can you then post the part of the code you tested with as plain text here?
Comparing the differences from a screenshot is very hard.

popaserge, 2019-03-12:
Oh yes I posted pictures because I could not find a way to show the code with the lines numbers.
Hereafter the code starting from line 164 copied from Notepad++:

 if (strlen(customConfig.HttpHeader) > 0) {
    if (payload.endsWith("\r\n\r\n")) {
    //Remove extra newline, see https://github.com/letscontrolit/ESPEasy/issues/1970
      payload.remove(payload.length()-2);
  }
      payload += customConfig.HttpHeader;
  }
      if (payload.endsWith("\r\n\r\n")) {
    //Remove again extra newline, see https://github.com/letscontrolit/ESPEasy/issues/1970
      payload.remove(payload.length()-2);
  }
  ReplaceTokenByValue(payload, event);

  if (strlen(customConfig.HttpBody) > 0)
  {
    String body = String(customConfig.HttpBody);
    ReplaceTokenByValue(body, event);
    //payload += "\r\n";
    payload += F("Content-Length: ");
    payload += String(body.length());
    payload += "\r\n\r\n";
    payload += body;
  }
  payload += "\r\n";

popaserge, 2019-03-13:
@TD-er 
Found the involved function in StringConverter.ino
/******************************************\
   Typical string replace functions.
  \*****************************************/
void removeExtraNewLine(String& line) {
  if (line.endsWith("\r\n\r\n")) {
    line.remove(line.length()-2);
  }
}
I was surprised as the lines inside the function are the same as in my try.
As I am not a developper I tried to understand the String& statement.

I found : https://www.quora.com/What-does-const-string-in-C++-mean

But I found a lot of discussion about passing a variable by reference and saw that lot of people says this is not a C statement but a C++ one. As I know that Arduino file .ino are in C, I wonder if that could be the reason of not working that I saw with Wireshark (3 times \r\n).

Any opinion or other track.

TD-er, 2019-03-13:
That parameter must be passed by reference, since it will be changed in the function and thus after calling that function its content has changed.
And we are not limited to C, but are really in a C++ environment.

I added that function in this PR, so it was intentional.
I didn't have time today to look into it, maybe later this evening.

TD-er, 2019-03-14:
Ah, now I see it, you did also comment out this line:
```c++
    ReplaceTokenByValue(body, event);
    //payload += "\r\n";
```

TD-er, 2019-03-14:
Can you try [this test build](https://www.dropbox.com/s/a2k1ohkxv8wd274/ESPEasy_mega-20190311-3-PR_2389.zip?dl=0) ?

popaserge, 2019-03-14:
@TD-er 
ESP_Easy_mega-20190311-3-PR_2389_dev_ESP8266_4M.bin works well on 2 servers which receive the data.
I do not understand why logout does no more show the answer of the servers but it must be ok 200.
To be sure I looked with Wireshark: the number of \r\n is now good 1 before ContentLength and 2 after (see picture).
By the way the controller can be used without login and password and for me with a token in URI field which will be tested by the server (see issue #2227).
Maybe other could test now if no problem with their application.
Wireshark result
[image]

TD-er, 2019-03-14:
One of the settings of the controller is to wait for a response. It could be the connection was already closed before a response was given, so that's why you don't see it in Wireshark?

popaserge, 2019-03-14:
The server allready sent its answer and it can be seen line 22 on top of the Wireshark previous screenshot.
Just I can see that the log of ESP is not so fast as before and never the answer nor the information about the connection is given in the log like on a previous picture like this one.
[image]
Probably some modifications as the test I made was done from a release on February 22.

TD-er, 2019-03-14:
One of the recent changes was to remove the debug logs (to save space/memory)
This log line is part of this debug define.

I will remove that part, since this is actually very practical info and also not really saving a lot of space in the build.

Unr3al2, 2019-04-29:
Hey,
From the version: https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20190315 he does not send me headlines.
The mega-20190311 version is OK
Can you tell what has been changed?
[image]

Unr3al2, 2019-04-29:
I added enter and it's OK.

TD-er, 2019-04-29:
So we can consider this fixed I guess?

closed 2019-04-29 by TD-er