letscontrolit/ESPEasy #1971: Plugin-playground: _P124 double

ozett, 2018-10-27:
i wanted to try the neopixelfx _P124
but in the playground there a two pluins with no. 124

maybe this has to be sorted out?

another question:
how do i compile, activate compilation of this plugin?
the docu is sparse, so i looked up ESPeasy_globals.h an set it this way:

```
//build all the normal stable plugins (on by default)
//#define PLUGIN_BUILD_NORMAL

//build all plugins that are in test stadium
#define PLUGIN_BUILD_TESTING
```

is this the right way?
how i control wich plugin is compiled, and which number is not?
only with categroy "test" or "normal" ?

ozett, 2018-10-27:
i found the listing of "stable/testing/development" here in define_pluginsets.h
and i saw that you know, that P124 is double.

hard to find how to compile a single plugin-nr....

```
// EXPERIMENTAL (playground) #######################
#ifdef PLUGIN_SET_EXPERIMENTAL
    #define USES_P046   // VentusW266
    #define USES_P050   // TCS34725 RGB Color Sensor with IR filter and White LED
    #define USES_P064   // APDS9960 Gesture
    #define USES_P077	// CSE7766   Was P134 on Playground

    // [copied from Playground as of 6 March 2018]
    // It needs some cleanup as some are ALSO in the main repo,
    // thus they should have been removed from the Playground repo
    #define USES_P100	// SRF01
	   #define USES_P101	// NeoClock       (MERGED?)
	#define USES_P102	// Nodo
	#define USES_P103	// Event
	#define USES_P104	// SRF02
	#define USES_P105	// RGBW
	#define USES_P106	// IRTX
	#define USES_P107	// Email_Demo
	#define USES_P108	// WOL
	#define USES_P109	// RESOL_DeltaSol_Pro
	   #define USES_P110	// P1WifiGateway      (MERGED?)
	#define USES_P111	// RF
	   //#define USES_P111	// SenseAir     (MERGED?)
	#define USES_P112	// Power
	//#define USES_P112	// RFTX
	#define USES_P113	// SI1145
	#define USES_P114	// DSM501
	#define USES_P115	// HeatpumpIR
	#define USES_P116	// ID12
	#define USES_P117	// LW12FC
	//#define USES_P117	// Neopixels
	//#define USES_P117	// Nextion
	#define USES_P118	// CCS811
	#define USES_P119	// BME680
	#define USES_P120	// Thermocouple
	#define USES_P121	// Candle
	   #define USES_P122	// NeoPixel       (MERGED?)
	      #define USES_P123	// NeoPixel_Clock  (MERGED?)
	#define USES_P124	// NeoPixelBusFX
	//#define USES_P124	// Ventus_W266_RFM69
```

tonhuisman, 2022-05-06:
Has been moved to the mega repository via PR #3899

PR #3899 (2022-06-15): [P128] Migrate NeoPixelBusFX from playground / djcysmic/NeopixelBusFX
Resolves #3895 
Resolves #1971

- Add plugin NeoPixelBusFX, migrated from ESPEasyPluginPlayground and djcysmic/NeopixelBusFX
- Enable support for ESP32/ESP32s2

Available in (new) `NEOPIXEL` builds.

Known issues:
- ESP8266: H/W is fixed using GPIO-2 (Library only supports this GPIO)
- ~ESP32: H/W is fixed on HSPI~ (Fixed: Switched to Rmt method, that allows GPIO selection)
- Only supports GRB-mode, other modes require re-compilation
- Only supports NeoPixel LED-strips, not DotStar LED-strips.

TODO:
- [x] Select build set t [... cut]
closed 2022-05-06 by TD-er