letscontrolit/ESPEasy #1973: Some DS18B20 are not detected [_P004_Dallas]

stephjj, 2018-10-28:
I had a couple of DS18B20 (the waterproof version). They are not detected by ESPMega.
If I use a TO-92 version on the OneWire bus, the firmware works as expected.
If i try with TO-92 and waterproof versions, none are detected.
I have tested the release Mega-20181028, Mega-20180625, Mega-20180111 and R120 with no luck.

With these Arduino libraries, all the captors are correctly detected on the same hardware.
https://github.com/milesburton/Arduino-Temperature-Control-Library
http://www.pjrc.com/teensy/td_libs_OneWire.html

The hardware is a Wemos D1 mini

uzi18, 2018-10-28:
@stephjj please give us identification numbers of working and not working thermometers.

TD-er, 2018-10-28:
These sensors need 1 (and only 1) pull-up resistor in the 1-wire bus.
Does the waterproof version have such a resistor?
Can you add one  to test? 
4k7 between 3.3V and data line.

stephjj, 2018-10-28:
TO-92 : 282D9685050000A4
Waterproof : 28A10A43980102D5, 28A90B43980102B9

@uzi18 They are NOT detected. So no dropdown list
@TD-er : Exactly the same hardware for esapmega and arduino. One bus line with the correct pullup. 

If I put only the TO-92, it is correctly detected
If only the waterproof, no detection.
If both (three sensors on the wire) no detection.

uzi18, 2018-10-28:
@stephjj if it is possible remove all dallas tasks and add one, how many did it found in Device Adress selector?

stephjj, 2018-10-29:
None if the three sensors are connected.
But if I try with only one waterproof and one TO-92, the two sensors are detected
It's the same behavior  for each waterproof sensors.
So, 2 waterproof doesn't works. 1 waterproof + 1 TO-92 works. 1 waterproof doesn't works.

uzi18, 2018-10-29:
these waterproof have got 2 or 3 cables?

stephjj, 2018-10-29:
Hey, I just noticed the 20181029 release.
It's working !
One, two or three sensors are correctly detected.

Thanks, for the very quick fix.

stephjj, 2018-10-29:
And yes, the waterproof sensor got the cables. All connected (no parasitic)

uzi18, 2018-10-29:
ok so please close isse, thx for tests

closed 2018-10-29 by stephjj