letscontrolit/ESPEasy #1976: Firmware load does not work

DittelHome, 2018-10-29:
In newer releases, i am not able to load firmware via gui on my ESP8266.
If a disconnect all sensor's from my ESP it works.
Is this known ?

Grovkillen, 2018-10-29:
You need to give us more info.. from what version did you upload from?

DittelHome, 2018-10-29:
Ok sorry.
I have tried from ESP_Easy_mega-20181029_normal_ESP8266_4096.bin to
ESP_Easy_mega-20181023_normal_ESP8266_4096.bin and vice versa.
Sometimes it works, most time not.
If it does not work, i have to power off/on the ESP8266. After then the old version is running.

This devices are in use:
Communication - Serial Server
Display - OLED SSD1306
Generic - MQTT Import | MQTT_IMPORT
Switch input - Switch
Generic - Dummy Device

In addition rules are activ.

RSSI is 55 dbi

regards
Klaus

TD-er, 2018-10-29:
It is something I have also seen, but on a non-reproducible manner.
I was thinking about temporary disable all plugins when starting an OTA upgrade.
Only thing is the OTA code does not have a wrapper, so it may be a bit harder to see when OTA is starting.

uzi18, 2018-10-29:
@DittelHome plugin nr 23 or 36?

DittelHome, 2018-10-30:
@uzi18 Plugin 23

TD-er, 2018-10-30:
I had something similar this morning.
I tried to perform an OTA on the Sonoff POW r2 and I first had to disable the energy monitoring sensor to be able to flash it.

Grovkillen, 2018-10-30:
I suspect that we need to have the firmware go into some sort of OTA mode which disable pretty much everything making sure that it's not halting the update.

DittelHome, 2018-10-31:
My feeling, since Core 2.4.2 the problem is fixed ...

clumsy-stefan, 2018-10-31:
just had a similar issue. when the node is quite busy (whyever, I tested a new ESP Nonos core, cpu load was near 100%), the upload of the firmware via OTA can take up to 10min! now I assume if you do that via GUI the browser will at some point in time (much less than these 10min) run into a timeout and stop sending data.
I pushed the firmware via commandline (wget) and specified a large enough timeout, so that it actually can cope with the slow nodes....
I also had the least problems with 2.4.2 high bandwith FW..... so again some esp cores seem to have networking issues which pushes the nodes to a high CPU load and make them not responsive anymore...

clumsy-stefan, 2018-11-03:
one other thing related to firmware upgrades: when using the nodes with deepsleep at least  deepsleep should be disabled when doing a OTA upgrade.

I think it should, so it would be possible to detect a wakeup from deepsleep by the controller/server and initiate a OTA upgrade before it goes into deepsleep again. should be enabled again after the upgrade.

any views comments on that idea?

clumsy-stefan, 2018-11-03:
just had a look at the code, it seems except a `delay(0)` only the OTA update is handled. so basically no other tasks or deepsleep could run... if I do get it right...

probably @TD-er or @Grovkillen can confirm/deny?

TD-er, 2018-11-03:
You will have to call deepSleep to get to sleep. This will be called by our own code from the `loop()`, it is not a background task.
Only waking up can be done automatically.

So if the OTA update is indeed the only thing being served, the OTA procedure is the only thing running and thus the current call of the loop() will take as long as the OTA takes.
Only thing I am not really sure about, whether it may occur that the node is put to sleep between storing the new sketch and the reboot.

clumsy-stefan, 2018-11-03:
I use the standard deepsleep function in the config menu.

tried it with updating a unit that awakes every 10 min. for 30sec. I just used a ping-loop on the server to find out when it's awake and immediatly called a curl/wget with the update after the first succesfull ping. did the update smoothly rebooted and went into the deepsleep/wakeup loop again...

so I think this works quite nicely only now... juest becaus @Grovkillen mentioned:
> I suspect that we need to have the firmware go into some sort of OTA mode which disable pretty much everything making sure that it's not halting the update.

looking at the code probably that's already working like this?

However I can't judge what background tasks are performed when calling `delay(=)`... probably there is stil lan issue..

berbergh, 2018-11-25:
I had a similar problem. Update almost always fails when generic MQTT import and MQTT are enabled.
You can see what's going on with the serial monitor. The update dots appear very very slowly and grind to a halt when both MQTT host and import are enabled. Disabling one of them, already improves speed.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er