letscontrolit/ESPEasy #1977: LET and TASKVALUESET: calculate does not parseString before process

uzi18, 2018-10-29:
```

>let,1,100
Ok
3600610 : Command: let
>taskvalueset,1,1,10
Ok
3637409 : Command: taskvalueset
>taskvalueset,1,1
10.00
Ok
3647609 : Command: taskvalueset
>taskvalueset,1,1,%v1%
Ok
3664309 : Command: taskvalueset
>taskvalueset,1,1
0.00
Ok
3672309 : Command: taskvalueset

```

taskvalueset,1,1 must show 100.0

giig1967g, 2018-10-29:
I will look at it.
Just today I was looking at anticipating the call to ParseSystemVariable in the Rules execution flow.

uzi18, 2018-10-29:
thx @giig1967g 
@TD-er I'm just curious why Calculate does not parse everytime

giig1967g, 2018-10-29:
I just tested it and it works for me.
```
on step1 do
let,1,123
endon

on step2 do
taskvalueset,1,1,%v1%
endon
```

If you call step1 and then step2 it works and the taskvalue at 1,1 is correctly set to 123.
You cannot put the two statements in the same rule though, as it's a known bug.

uzi18, 2018-10-29:
I call it from serial ... maybe this is root case

giig1967g, 2018-10-29:
> 
> 
> thx @giig1967g
> @TD-er I'm just curious why Calculate does not parse everytime

Calculate is called by the original design only in certain conditions or you can force it if you preceed the formula with '='.
let,1,=[task#value]*2+%eventvalue%

giig1967g, 2018-10-29:
> 
> 
> I call it from serial ... maybe this is root case

ok. I will check this case. Maybe serial commands have different path during rules execution.
I am currently making a map of the rules execution path because it's really very complex.
Haven't yet started using serial commands.

giig1967g, 2018-10-29:
Anyway, %v1% is evaluated in the "ParseSystemVariables()" function.
Not in the Calculate function.

uzi18, 2018-10-29:
have testing variables1-16 and found such issue

uzi18, 2018-10-29:
Maybe ParseSystemVariables is needed here, I'm not sure thats why added this issue

giig1967g, 2018-10-29:
just for testing does the same occurs using other system variables, or just with %v1%-%v16%?
For example using %ip4% or %wi_ch%

TD-er, 2018-10-29:
Please also take into account the commands are still run from the scheduler.
So for now, take into account the commands may be run out of order.

I really must look into the reason why it is now reportedly crashing when commands are run immediately.

giig1967g, 2018-10-30:
I checked the code: indeed the serial and web commands do not have any string parsing, nor conversion nor format.
Will fix the code in the next few days.

closed 2019-10-10 by Grovkillen