letscontrolit/ESPEasy #1995: Hangs when access web ui from different ip range

wolverinevn, 2018-11-01:
### Checklist
- [ x] The title of this issue is "to the point" and descriptive.
- [ x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [ x] This issue describes how to reproduce it.
- [ x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
2
Local Time:	2018-11-01 20:48:32
Uptime:	0 days 6 hours 22 minutes
Load:	6.50% (LC=12487)
Free Mem:	9040 (2696 - sendContentBlocking)
Free Stack:	3552 (1616 - ExecuteCommand)
Boot:	Cold boot (0)
Reset Reason:	External System
Network ❔
Wifi:	802.11N (RSSI -61 dB)
IP config:	DHCP
IP / subnet:	192.168.31.106 / 255.255.255.0
GW:	192.168.31.1
Client IP:	192.168.31.40
DNS:	192.168.31.1 / 8.8.8.8
Allowed IP Range:	All Allowed
STA MAC:	84:F3:EB:Xxxxxx
AP MAC:	86:F3:EB:Xxxxx
SSID:	(F0:B4:29:Xxxxxxx)
Channel:	11
Connected:	44 m 36 s
Last Disconnect Reason:	(201) No AP found //Right after issue, it was “beacon timeout
Number reconnects:	2
Firmware
Build:⋄	20102 - Mega
Libraries:⋄	ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2
GIT version:⋄	mega-20181026
Plugins:⋄	46 [Normal]
Build Md5:	40a5583b53e6eb408ed35a943564
Md5 check:	passed.
Build time:⋄	Oct 26 2018 02:20:30
Binary filename:⋄	ESP_Easy_mega-20181026_normal_ESP8266_4096.bin
System Status
Syslog Log Level:	None
Serial Log Level:	None
Web Log Level:	None
ESP board
ESP Chip ID:	5928724 (0x5A7714)
ESP Chip Freq:	80 MHz
ESP Board Name:	PLATFORMIO_ESP12E
Storage
Flash Chip ID:	Vendor: 0x68 Device: 0x4016
Flash Chip Real Size:	4096 kB
Flash IDE Size:	4096 kB
Flash IDE speed:	40 MHz
Flash IDE mode:	DIO
Flash Writes:	7 daily / 7 boot
Sketch Size:	747 kB (2324 kB free)
SPIFFS Size:	934 kB (860 kB free)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
ESP_Easy_mega-20181026_normal_ESP8266_4096
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x ] Tested previous/other build (mention which one already tested)
- [x ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I have 2 Nodemcu 12E running same Espeasy version at same location (network). Both use Openhab mqtt connect to broker in another location (mosquitto). Both are working normally (not very stable but acceptable). I enable allow access from all ips 255.255.255.255.
On my router (which 2 units connects to), I have a Open VPN server installed which I used to connect to access devices remotely. VPN server was configured: TUN, NAT to LAN, client ip: 10.8.0.0. Lan ip: 192.168.31.1.

Everytime I tried to connect to my 2 Espeasy units via VPN, both 2 units were hangs. Web ui keep loading or quickly finish load without showing anything. If I disconnect VPN and access Espeasy from LAN right after, web ui keeps unreachable for 4 or 5 minutes. After that, one operated normally, one was rebooted (which has more tasks and rules). When they were frozen, no mqtt sent to broker. I am still able to access all other devices (main routers, camera, etc.) via same VPN comnection and session. 

I also have another Nodemcu with Espeasy released on 03rd of October with same mqtt broker connect, after another network with same VPN server setup. This one is good with web access via VPN.

### Expected behavior

I should access web ui them same as from lan.

### Actual behavior

Unit hangs.

### Steps to reproduce

1. Enable access from all and mqtt controller
2. Try open web ui from other ip range, better via VPN

Yes.

### System configuration

Hardware:
Only DHT-22.

ESP Easy version: mega-20181026 (normal,4096)

ESP Easy settings/screenshots: 
Openhab Mqtt, min send 150ms, queue 3, retries 1, delete oldest, timeout 200.

### Rules or log data

```

```
Sorry for very long post. I would like to describe as much details as I can.
You team did a great job until now. My first Espeasy version might be 2.4.0.

giig1967g, 2018-11-01:
please see #1952

TD-er, 2018-11-01:
It looks indeed related (maybe even duplicate????) of #1952 
We have to take a look at what may have changed since the last version that worked.

Do you have any idea on the 'ping time' via the VPN to the node?

giig1967g, 2018-11-01:
@wolverinevn 
please try version 20181101. In my configuration it works again.

wolverinevn, 2018-11-01:
> Do you have any idea on the 'ping time' via the VPN to the node?
Ping time: 
64 bytes from 192.168.31.106: icmp_seq=8 ttl=127 time=60.383 ms
64 bytes from 192.168.31.106: icmp_seq=9 ttl=127 time=48.283 ms
After 10 seconds try to connect to web ui:
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1

And Safari when access from LAN right after:
server unexpectly dropped connection, sometime occurs when server is busy.

I'm trying mega20181101 and report back.

wolverinevn, 2018-11-01:
As far as I know, web ui access via VPN is back on mega20181101. So the issue is not create by unstable wifi as described in #1987.

closed 2018-11-04 by wolverinevn