letscontrolit/ESPEasy #1999: New event MQTT#PublishComplete

Grovkillen, 2018-11-03:
Given the new event based approach for rules we cannot be sure that this will work:

```
On MQTT#Connected Do
 Publish,%sysname%/Temperature,[BME#temp]
 DeepSleep,600
EndOn
```
Currently this means that the Publish and DeepSleep will be triggered "at the same time". And thus the deepsleep will make the publish not take place. To overcome this we would use a timer:
```
On MQTT#Connected Do
 Publish,%sysname%/Temperature,[BME#temp]
 TimerSet,1,5
EndOn

On Rules#Timer=1 Do
  DeepSleep,600
EndOn
```

But given that we want to save as much time as possible (save battery) we don't guess the time it takes for it to be done publishing. So my idea is that after each publish we get an event `MQTT#PublishComplete` or even better, after each publish the event would add an integer, i.e.:

```
On MQTT#Connected Do
 Publish,%sysname%/Temperature,[BME#temp]
 Publish,%sysname%/Humidity,[BME#rh]
EndOn

On MQTT#PublishComplete=2 Do
  DeepSleep,600
EndOn
```

uzi18, 2018-11-03:
just to be sure all task has got updated data we need to introduce task_x_valid bool and when all sent alredy, esp can go sleep.

uzi18, 2018-11-03:
@TD-er it is not bad idea with async command execution, but some of them must be marked to be executed in sync, to many problems with that and no backwards compatibility.

Grovkillen, 2018-11-03:
@uzi18 I agree that we need to get the backwards compatibility a bit better but the future is event based and sequences can be done using timers. But maybe an event called "sync1"..."sync4" where the commands are triggered one line at the time.

```
On System#Boot Do
 Sequential,1
EndOn

On Sequential=1 Do
 GPIO,14,0
 Pulse,14,1,200
 Pulse,14,0,3000
 Pulse,14,1,200
EndOn
```

M0ebiu5, 2018-11-03:
What should the benefit of executing rule commands asynchronously be?
I can only see a ton potential problems :)

Grovkillen, 2018-11-03:
If you want to (as an example) want to pulse 4 IOs at the same time?

M0ebiu5, 2018-11-03:
then i would provide an advanced pulse command, where i could mask the ports i want to manipulate at once...
There is no real "same time", even if the commands are async - just an undefined order

Grovkillen, 2018-11-03:
I know it's not "same time", but my point is that we'd like both worlds but in sequence is not always preferred, and doing neopixel flows could only be made one pixel at the time as an example. So an closing code chunk (in any form), is preferred in my opinion, for letting the parser know that these are blocking sequential commands.

M0ebiu5, 2018-11-03:
Doing more pixels at once should be done with maybe a range command and not some weird async construct in the rules - it's neither intuitive nor user friendly.
Just speed up the rule execution time (eg only parse rules at startup to create an object - operator tree) and it will also look like "same time" :)

Grovkillen, 2018-11-03:
Okay, lets stay on topic :) We're discussing the `MQTT#PublishCompleted` :)

M0ebiu5, 2018-11-03:
Sure, so i just wanted to tell you, that this whole async thing inside the rules is crap :)

uzi18, 2018-11-03:
@M0ebiu5 no, It just can't be done with all commands.

TD-er, 2018-11-03:
@M0ebiu5 
I agree the rules must be executed in a very predictive order.
The async idea of running network related commands is about not blocking for a long time.

So if we want to be able to do publish commands async, we simply need an event to trigger when it is finished.
But most important is to be predictive and the current state with commands running from the scheduler is not as predictive as it should be when executing rules.
So that really has to change, but I really have no clue why things are crashing when I try to run commands from withing rules directly instead of running from the scheduler.

TD-er, 2018-11-03:
@M0ebiu5 That's correct, but at the moment we don't allow for temporary stop rule execution and continue later on.
And I will wait implementing that, since that will open a new can of worms.
The whole idea is to have short (smaller) steps to do when executing rules.
At this moment it is possible to have one on... section in the rules which is calling numerous commands. Each of them may take 100's of msec to complete and thus keep the system blocked.

That's why I initially moved the commands to the scheduler, but I never thought of the side effects we now see when running rules.
So for now, we should add them back out of the scheduler and later think of a way to make them non blocking anymore.
Your suggestion is one of the possible options.

Grovkillen, 2019-10-10:
Is this still something we need?

TD-er, 2019-10-10:
Yep, but not now.
I really would like to have an "async" version of the commands that are now blocking.
Like SendToHTTP, Publish, etc.
But that also would demand a "callback" function, which would result in an event in our code.

One other thing that would really be great is to have a cue of events. Now we can call an event from within rules, but that can result in a very long rule execution.
So if you fire some async command, then it may result in a new event which is then just appended to the event cue.
Only problem is how to distinguish "derived" events from "external triggered" events and should we even care about that? (a chain of "derived" events may block external events)