letscontrolit/ESPEasy #2000: Plugin OLED SSD1306 causes problems

DittelHome, 2018-11-03:
I have a ESP8266 with this firmware:
ESP_Easy_mega-20181031_normal_ESP8266_4096.bin
This devices are defined:
- Communication - Serial Server
- Display - OLED SSD1306
- Generic - MQTT Import
- Switch input - Switch
- Generic - Dummy Device

My Serial Server device gets dropped out at least every 24 hours.
If i disable the OLED SSD1306 device the Serial Server works great for a long time, at least 4 days.
If i enable the SSD1306 device the Serial Server isn't reliable.

I have already tried to set the interval paramter of the SSD1306 to at least 10 seconds, no success.

regards
Klaus

TD-er, 2018-11-03:
Which OLED plugin do you use?
"OLED" or "Framed OLED" ?

DittelHome, 2018-11-03:
Ok, i have activated the serial log (log server).
I will report, we have to wait ...

TD-er, 2018-11-03:
Those timing dumps appear every 30 seconds in the log.
The first dump may contain stats for more than 30 seconds, so if you can gather 2 of those dumps that would be great.

DittelHome, 2018-11-04:
This is the first dump after start:
```
Nov  3 23:50:56 HMUART3 EspEasy: WD   : Uptime 1 ConnectFailures 0 FreeMem 21520
Nov  3 23:50:57 HMUART3 EspEasy: LoopStats: shortestLoop: 81 longestLoop: 1295262 avgLoopDuration: 118.02 loopCounterMax: 370370 loopCounterLast: 241728 countFindPluginId: 0
Nov  3 23:50:57 HMUART3 EspEasy: PluginStats P_1_Switch input - Switch ONCE_A_SECOND        Count: 30 Avg/min/max 12.50/11/31 usec
Nov  3 23:50:57 HMUART3 EspEasy: PluginStats P_1_Switch input - Switch TEN_PER_SECOND       Count: 276 Avg/min/max 42.92/28/68 usec
Nov  3 23:50:57 HMUART3 EspEasy: PluginStats P_1_Switch input - Switch FIFTY_PER_SECOND     Count: 1373 Avg/min/max 21.74/15/31 usec
Nov  3 23:50:57 HMUART3 EspEasy: PluginStats P_19_Communication - Serial Server ONCE_A_SECOND        Count: 30 Avg/min/max 18.20/14/55 usec
Nov  3 23:50:57 HMUART3 EspEasy: PluginStats P_19_Communication - Serial Server TEN_PER_SECOND       Count: 276 Avg/min/max 53.37/35/230 usec
Nov  3 23:50:57 HMUART3 EspEasy: PluginStats P_19_Communication - Serial Server SERIAL_IN            Count: 3 Avg/min/max 2553.67/2099/2792 usec
Nov  3 23:50:57 HMUART3 EspEasy: PluginStats P_19_Communication - Serial Server FIFTY_PER_SECOND     Count: 1373 Avg/min/max 14.34/6/39 usec

Second Dump:

Nov  3 23:51:27 HMUART3 EspEasy: LoopStats: shortestLoop: 81 longestLoop: 3406864 avgLoopDuration: 132.48 loopCounterMax: 370370 loopCounterLast: 221386 countFindPluginId: 0
Nov  3 23:51:27 HMUART3 EspEasy: PluginStats P_1_Switch input - Switch ONCE_A_SECOND        Count: 27 Avg/min/max 13.44/11/28 usec
Nov  3 23:51:27 HMUART3 EspEasy: PluginStats P_1_Switch input - Switch TEN_PER_SECOND       Count: 261 Avg/min/max 41.79/32/64 usec
Nov  3 23:51:27 HMUART3 EspEasy: PluginStats P_1_Switch input - Switch EVENT_OUT            Count: 1 Avg/min/max 31.00/31/31 usec
Nov  3 23:51:27 HMUART3 EspEasy: PluginStats P_1_Switch input - Switch FIFTY_PER_SECOND     Count: 1278 Avg/min/max 22.28/15/31 usec
Nov  3 23:51:27 HMUART3 EspEasy: PluginStats P_19_Communication - Serial Server ONCE_A_SECOND        Count: 27 Avg/min/max 21.78/15/62 usec
Nov  3 23:51:27 HMUART3 EspEasy: PluginStats P_19_Communication - Serial Server TEN_PER_SECOND       Count: 261 Avg/min/max 67.23/35/2336 usec
Nov  3 23:51:27 HMUART3 EspEasy: PluginStats P_19_Communication - Serial Server EVENT_OUT            Count: 1 Avg/min/max 66.00/66/66 usec
```

I hope this is the information you requested ?

regards 
Klaus

TD-er, 2018-11-04:
I was hoping to see the plugin stats for the OLED plugin.
I will move these dumps from the log to a JSON page. Then we can query them whenever needed, since they provide great detail on timing statistics for just about anything.
But the information is just too bulky to process in logs.

fluppie, 2018-11-06:
Mine freezes also after a few days with the Framed OLED plugin. It looses the wifi connection. The IP adress on the screen says 0.0.0.0 where normally it is 192.168.1.35

I use the official build: Binary filename: ESP_Easy_mega-20181027_normal_ESP8266_4096.bin

With this ruleset:
```
on PIR1#Switch=1 do
   OLEDFRAMEDCMD,on
endon

on PIR1#Switch=0 do
   OLEDFRAMEDCMD,off
endon
```
I used to have the MQTT import as well, but removed that one as I thought it was the reason for the lockups, but that's not the case.

[image]

Grovkillen, 2018-11-06:
Could you try this rule instead?

on PIR1#Switch do
 if [PIR1#Switch]=1
   OLEDFRAMEDCMD,on
 else
   OLEDFRAMEDCMD,off
 endif
endon

TD-er, 2018-11-06:
I moved the timing statistics from being dumped on the log, to a new page (link in the Tools tab)
Every time the page is displayed, it will reset the stats, so you may also run it for a longer time to get an idea of the effect of some changes you may make between views of that page.

kischde, 2018-11-16:
Same here. I´d tried it with the second rule you mentioned. After a while the device is not reachable via network any more.
Workaround:
I send the switch-state via MQTT to the server, and send back  OLEDFRAMEDCMD,off / on to the display esp.
The device stays now in the network, but after around 2 days I have to reboot (via webinterface), because it does not send any MQTT messages any more

kischde, 2018-11-16:
Yes, OLED frame is set to scroll, cause I need the big font...

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
P_1_Switch input - Switch | ONCE_A_SECOND | 3108 | 1.00 | 0.007 | 0.019 | 0.041
P_1_Switch input - Switch | TEN_PER_SECOND | 21674 | 6.97 | 0.036 | 0.086 | 25.175
P_1_Switch input - Switch | WRITE | 40 | 0.01 | 0.561 | 0.585 | 0.633
P_1_Switch input - Switch | EVENT_OUT | 1611 | 0.52 | 0.015 | 0.025 | 0.055
P_1_Switch input - Switch | FIFTY_PER_SECOND | 105177 | 33.82 | 0.015 | 0.019 | 0.050
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | READ | 16 | 0.01 | 22.533 | 22.754 | 23.018
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | ONCE_A_SECOND | 3108 | 1.00 | 0.006 | 0.011 | 0.089
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | TEN_PER_SECOND | 21674 | 6.97 | 0.006 | 0.010 | 0.061
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | WRITE | 40 | 0.01 | 0.042 | 0.050 | 0.155
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | EVENT_OUT | 1611 | 0.52 | 0.018 | 0.026 | 0.094
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | FIFTY_PER_SECOND | 105177 | 33.82 | 0.002 | 0.007 | 0.039
P_35_Communication - IR Transmit | ONCE_A_SECOND | 3108 | 1.00 | 0.006 | 0.016 | 0.033
P_35_Communication - IR Transmit | TEN_PER_SECOND | 21674 | 6.97 | 0.006 | 0.008 | 0.029
P_35_Communication - IR Transmit | WRITE | 40 | 0.01 | 0.178 | 0.188 | 0.230
P_35_Communication - IR Transmit | EVENT_OUT | 1611 | 0.52 | 0.010 | 0.022 | 0.060
P_35_Communication - IR Transmit | FIFTY_PER_SECOND | 105177 | 33.82 | 0.006 | 0.013 | 0.040
P_36_Display - OLED SSD1306/SH1106 Framed | READ | 1555 | 0.50 | 573.543 | 644.088 | 705.964
P_36_Display - OLED SSD1306/SH1106 Framed | ONCE_A_SECOND | 3108 | 1.00 | 0.023 | 0.037 | 1.179
P_36_Display - OLED SSD1306/SH1106 Framed | TEN_PER_SECOND | 21674 | 6.97 | 0.008 | 0.019 | 0.066
P_36_Display - OLED SSD1306/SH1106 Framed | WRITE | 40 | 0.01 | 0.447 | 0.506 | 0.536
P_36_Display - OLED SSD1306/SH1106 Framed | EVENT_OUT | 1611 | 0.52 | 0.018 | 0.021 | 0.043
P_36_Display - OLED SSD1306/SH1106 Framed | FIFTY_PER_SECOND | 105177 | 33.82 | 0.011 | 0.015 | 0.048
P_37_Generic - MQTT Import | ONCE_A_SECOND | 3108 | 1.00 | 0.025 | 2.410 | 1330.258
P_37_Generic - MQTT Import | TEN_PER_SECOND | 21674 | 6.97 | 0.018 | 0.125 | 7.223
P_37_Generic - MQTT Import | EVENT_OUT | 1611 | 0.52 | 0.026 | 0.030 | 0.052
P_37_Generic - MQTT Import | FIFTY_PER_SECOND | 105177 | 33.82 | 0.003 | 0.009 | 0.043
Load File |   | 13835 | 4.45 | 0.854 | 1.227 | 25.796
Loop |   | 14159168 | 4553.05 | 0.105 | 0.184 | 1430.717
Plugin call 50 p/s |   | 105177 | 33.82 | 1.657 | 2.338 | 4.280
Plugin call 10 p/s |   | 21674 | 6.97 | 1.740 | 2.460 | 27.629
Plugin call 10 p/s U |   | 21674 | 6.97 | 0.499 | 0.722 | 1.184
Plugin call  1 p/s |   | 3108 | 1.00 | 2.891 | 5.550 | 1333.328
SensorSendTask() |   | 1571 | 0.51 | 48.873 | 644.793 | 712.597
sendData() |   | 1611 | 0.52 | 3.274 | 5.119 | 26.303
Compute formula |   | 1571 | 0.51 | 0.018 | 0.034 | 1.723
setNewTimerAt() |   | 145782 | 46.88 | 0.075 | 0.199 | 0.307
Delay queue MQTT |   | 70 | 0.02 | 1.133 | 1.675 | 1.932

TD-er, 2018-11-16:
Take these lines:

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
P_36_Display - OLED SSD1306/SH1106 Framed | READ | 1555 | 0.50 | **573.543** | **644.088** | **705.964**
Loop |   | 14159168 | 4553.05 | 0.105 | 0.184 | **1430.717**

The OLED is scrolling every 2 seconds and it takes on average 644 msec.
This means it is **blocking all other tasks** for over 25% of the time.
This will cause a lot of issues.
So please set the scroll to "instant", which will take about 90 - 120 msec.
Also try to set the update frequency a bit lower, to 5 seconds.

kischde, 2018-11-16:
done...

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
P_1_Switch input - Switch | ONCE_A_SECOND | 173 | 0.99 | 0.007 | 0.019 | 0.041
P_1_Switch input - Switch | TEN_PER_SECOND | 1710 | 9.80 | 0.036 | 0.095 | 23.921
P_1_Switch input - Switch | WRITE | 4 | 0.02 | 0.577 | 0.596 | 0.616
P_1_Switch input - Switch | EVENT_OUT | 41 | 0.24 | 0.015 | 0.023 | 0.055
P_1_Switch input - Switch | FIFTY_PER_SECOND | 8352 | 47.89 | 0.015 | 0.019 | 0.050
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | READ | 2 | 0.01 | 23.093 | 23.131 | 23.170
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | ONCE_A_SECOND | 173 | 0.99 | 0.006 | 0.013 | 0.101
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | TEN_PER_SECOND | 1710 | 9.80 | 0.006 | 0.010 | 0.061
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | WRITE | 4 | 0.02 | 0.043 | 0.072 | 0.149
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | EVENT_OUT | 41 | 0.24 | 0.018 | 0.027 | 0.079
P_5_Environment - DHT11/12/22  SONOFF2301/7021 | FIFTY_PER_SECOND | 8352 | 47.89 | 0.002 | 0.006 | 0.052
P_35_Communication - IR Transmit | ONCE_A_SECOND | 173 | 0.99 | 0.007 | 0.016 | 0.054
P_35_Communication - IR Transmit | TEN_PER_SECOND | 1710 | 9.80 | 0.006 | 0.008 | 0.024
P_35_Communication - IR Transmit | WRITE | 4 | 0.02 | 0.187 | 0.194 | 0.212
P_35_Communication - IR Transmit | EVENT_OUT | 41 | 0.24 | 0.011 | 0.022 | 0.059
P_35_Communication - IR Transmit | FIFTY_PER_SECOND | 8352 | 47.89 | 0.006 | 0.012 | 0.040
P_36_Display - OLED SSD1306/SH1106 Framed | READ | 35 | 0.20 | 85.806 | 102.192 | 124.754
P_36_Display - OLED SSD1306/SH1106 Framed | ONCE_A_SECOND | 173 | 0.99 | 0.019 | 0.043 | 1.191
P_36_Display - OLED SSD1306/SH1106 Framed | TEN_PER_SECOND | 1710 | 9.80 | 0.008 | 0.018 | 0.066
P_36_Display - OLED SSD1306/SH1106 Framed | WRITE | 4 | 0.02 | 0.510 | 0.518 | 0.525
P_36_Display - OLED SSD1306/SH1106 Framed | EVENT_OUT | 41 | 0.24 | 0.018 | 0.020 | 0.044
P_36_Display - OLED SSD1306/SH1106 Framed | FIFTY_PER_SECOND | 8352 | 47.89 | 0.011 | 0.015 | 0.047
P_37_Generic - MQTT Import | ONCE_A_SECOND | 173 | 0.99 | 0.034 | 8.329 | 461.885
P_37_Generic - MQTT Import | TEN_PER_SECOND | 1710 | 9.80 | 0.004 | 0.109 | 5.875
P_37_Generic - MQTT Import | EVENT_OUT | 41 | 0.24 | 0.027 | 0.030 | 0.053
P_37_Generic - MQTT Import | FIFTY_PER_SECOND | 8352 | 47.89 | 0.003 | 0.009 | 0.031
Load File |   | 352 | 2.02 | 0.856 | 1.410 | 25.369
Loop |   | 1129445 | 6475.99 | 0.109 | 0.146 | 479.398
Plugin call 50 p/s |   | 8352 | 47.89 | 1.573 | 2.332 | 4.730
Plugin call 10 p/s |   | 1710 | 9.80 | 1.639 | 2.463 | 26.309
Plugin call 10 p/s U |   | 1710 | 9.80 | 0.451 | 0.676 | 1.182
Plugin call  1 p/s |   | 173 | 0.99 | 2.924 | 11.966 | 477.912
SensorSendTask() |   | 37 | 0.21 | 49.379 | 106.660 | 135.600
sendData() |   | 41 | 0.24 | 3.148 | 6.903 | 26.715
Compute formula |   | 37 | 0.21 | 0.018 | 0.097 | 1.662
setNewTimerAt() |   | 11167 | 64.03 | 0.075 | 0.197 | 0.279
Delay queue MQTT |   | 6 | 0.03 | 0.691 | 1.523 | 1.880

TD-er, 2018-11-16:
Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
P_37_Generic - MQTT Import | ONCE_A_SECOND | 173 | 0.99 | 0.034 | 8.329 | **461.885**

That one also takes quite some time. (the extreme value)
Maybe also have a look after a while to see if the average value increases.
If average stays below 10 msec, then there is not really a problem I guess.

kischde, 2018-11-16:
This is now less than 1

P_37_Generic - MQTT Import | ONCE_A_SECOND | 5487 | 1.00 | 0.029 | 0.058 | 0.097
-- | -- | -- | -- | -- | -- | --

TD-er, 2018-11-19:
I just noticed the OLED display plugin (the one mentioned in the 1st post) is reading all lines from flash over and over again.
So it is reading 512 bytes from flash every call to the plugin.

I will add some timing stats to the reading process of these CustomTaskSettings, just to get an idea of how long it takes.

kischde, 2018-11-19:
Mine is working since we did the changes (3 days ago) without issues = MQTT still sending / receiving  OLEDFRAMEDCMD,off / on

The flash reading was also the problem, so the devs/you guys stopped the frame plugin displaying received text messages (which would be very nice to get back)

Thx for your help!

kischde, 2018-11-19:
Framed Plugin:

in V. 2 (mega) http and mqtt string upload to the 12 lines in framed oled has been removed by the dev's for reaons. use v. 148 or below.

http://www.letscontrolit.com/forum/viewtopic.php?f=4&t=1149&sid=c8e5d4a616a05cc317f951284adeac5e&start=100

kischde, 2018-11-28:
Today the ESP was again not reachable via network. I had to power off/on.
Yesterday I saw on the info page, that it was rebooting via watchdog probably once a day.
I updated now to the latest release 2018-11-28

TD-er, 2018-11-28:
> I updated now to the latest release 2018-11-28

Please let me know if that also fixes the watchdog issues. (or at least make it much less reproducible ;) )

kischde, 2018-11-30:
> > I updated now to the latest release 2018-11-28
> 
> Please let me know if that also fixes the watchdog issues. (or at least make it much less reproducible ;) )

For now it's running since 48 hours without watchdog reboot. However there was one 30 min after power up, but none anymore. Before after 20hours latest..

closed 2018-12-02 by DittelHome