letscontrolit/ESPEasy #2001: Feature request: WiFi mesh plugin

s0170071, 2018-11-04:
I would love a plugin which enables some simple wifi mesh capability. Just for the occasional IoT traffic. I have ESP nodes all over the house, but some of them sit in a very weakly covered area.
One or two repeater nodes would do the trick. There are some similar projects around, they just need to be "pluginified". 
 
[image]
image source: https://github.com/martin-ger/esp_wifi_repeater

ps: if the crowd starts cheering now, I may be tempted to do that myself...

giig1967g, 2018-11-04:
count on my cheer!!!

TD-er, 2018-11-04:
I would also love to see it :)
It is already present in the core libraries, but we may need to think about how to configure it.
This isn't something like a plugin, nor a controller, so it will be at the heart of the configuration.

Things to consider:
- configuration
- what traffic is allowed and what not
- security
- ease of setup
- power consumption
- is it a standard, or Espressif proprietary? (ESP WiFi mesh)
- compatible between ESP8266/ESP32
- also allow the ESP32 BlueTooth mesh capability? (released test code about a week ago)
- What wifi mesh strategy is present, or is it just a "self learning one setup runs everywhere" kind of thing?
- Bandwidth control/monitoring?
- gather information on connection routes?

That's just a few things I can think of right now.
But I like the idea and it was already on our to-do list :) Only not on the "right now" list.

s0170071, 2018-11-06:
What do you think about [ESPnow](https://www.espressif.com/en/products/software/esp-now/overview) ? Its a proprietary protocol, even below UDP. There is  a [discussion thread](https://github.com/esp8266/Arduino/issues/3408)   where a user claims to wake a node from deep sleep, boot, read a bmp280, TX the data and go back to sleep within 40ms.

TD-er, 2018-11-06:
ESPnow is also very nice, but serves a completely different purpose.

- You have to know MAC address of the ones you want to peer with
- Just a single peer-to-peer connection
- limited set of approved MAC addresses per node
- Limited message size

But it is indeed great for dumping sensor data from a battery powered one to one net powered.
Reading a BMP280 in 40 msec is a bit optimistic (will result in big error), but indeed it is possible to connect and send a message in no-time. So the WiFi radio has to be on for only a very short time.

s0170071, 2018-11-06:
It would serve my purpose, which is to cope with weak wifi coverage in my house. The nodes without direct wifi access could send their data to a node within wifi range. Maybe that solution does not interfere so much with all the wifi code we already run.

TD-er, 2018-11-06:
ESPnow cannot be used to route traffic. ESPnow is only between nodes.
So we could add it to the ESPeasy p2p layer, but not for letting other controllers communicate to an otherwise unreachable computer in the network.
That's what mesh is for.

s0170071, 2022-12-28:
Hi all, its been a while. I have done some recent digging and it appears like the Espressif folks have shared some of our concerns. 

Now, ESPNow can 
- access WIFI and ESPnow at the same time (if they use the same channeI) so that a node can act as a gateway to the local wifi.
- do [mesh](https://randomnerdtutorials.com/esp-mesh-esp32-esp8266-painlessmesh/) networking
- have a l[ong range (LR) mode ](https://www.hackster.io/news/long-range-wifi-for-the-esp32-9429ab89f450)with [3 times ](https://www.youtube.com/watch?v=oz0a7Ur7nko)the range of a normal wifi (ESP32 only for now)
- the ESPNow mode does not need to connect/negotiate wifi parameters and is therefore perfect for sending quickly after waking up from deep sleep (battery saving)
- and there are even examples for [auto peering](https://randomnerdtutorials.com/esp-now-auto-pairing-esp32-esp8266/) where a broadcast to all nodes collects the mac addresses from the slaves

From the top of my head it looks more like the slaves need a controller plugin "ESPNow" and the Master/Gateway part has to be  more tightly integrated into the wifi code so it does not interfere with it.