letscontrolit/ESPEasy #2002: Plantower PMS3003 support?

franki29, 2018-11-04:
Dear all,
i tried to connect a PMS3003 but i do not get any values. No matter using Software Serial or Hardware Seriel . I also tried it with Tasmota Firmware and get a error message Checksum error. Any ideas what i can do? 
I tested the device with https://github.com/fu-hsi/PMS-Tool and i can read the values from the device.
So the device itselve seems to be OK.

Br. Frank

micropet, 2018-11-04:
For me, the PMS works at least a few minutes.
Then no values come out.

It works perfectly with the samples from the Arduino Lib.

uzi18, 2018-11-04:
@micropet @TD-er maybe we can try with my checksumming?

micropet, 2018-11-05:
What do you mean?
I like to try something out.

uzi18, 2018-11-05:
what kind of image are you using with PMS?

tonhuisman, 2021-07-17:
I have created a PR to add support for this sensor to P053, if any of the users/owners of a PMS3003 (or PMS2003 that seems to use the same data format), would be so kind to test and report their findings here, that would be very much appreciated. A download for the desired build type can be provided here, if needed.

tonhuisman, 2021-07-25:
@franki29 @micropet Are you still available? And able & willing to test the updated plugin? I added support for the PMS2003 and PMW3003 sensors, but without the hardware available I can't test them myself, so a little help would be appreciated very much.
As said, the desired .bin I can provide here if you can't build, or download from the Actions build, yourself, just mention what ESP model it should run on, or the desired build type (the plugin is included in many of the builds).

franki29, 2021-07-26:
Hello, yes still here, currently I am using Tasmota, but If you could provide a ESP8266 bin file, I can test it.
Br. Frank

tonhuisman, 2021-07-26:
> but If you could provide a ESP8266 bin file, I can test it.

What size of flash do you have on your ESP (and assuming it is for ESP8266)?

franki29, 2021-07-26:
ESP8266 with 1024 kB

tonhuisman, 2021-07-26:
Hm, in these 1M builds I've deliberately disabled these new features, due to the added code-size, and the desire to allow these builds to do OTA updates (though not all of these 1M builds actually allow OTA). I'll have to build a special 1M build for you that does include this feature (or only exclude these features for OTA builds, that might be better), but I'm not at my personal laptop ATM, so it will take until tonight (CET) before I can supply that.

franki29, 2021-07-26:
No problem. As I have to change the firmware, I have to send it over USB , so no problem with OTA.

franki29, 2021-07-27:
Hi I test it with Software Serial, but no values are read out of the device.

franki29, 2021-07-27:
If I choose the wrong device, like PMS5005, I get the error I had before.

franki29, 2021-07-27:
Debug shows : 
: Debug  : PMSx003 : count/0.1L : 0.3um=662, 0.5um=2, 1.0um=38656, 2.5um=552, 5.0um=65535, 10um=65535␍␊
[16:00:16:149] 358733 : Debug  : PMSx003 : pm1.0=11, pm2.5=15, pm10=16, pm1.0a=11, pm2.5a=15, pm10a=16␍␊

But web view shows 0 for all values

tonhuisman, 2021-07-27:
Oh, OK, I'll have a look at the code, later today, maybe I've done something wrong there for this sensor model.

tonhuisman, 2021-07-27:
Do you also have a line from the log that shows the `pm` values? It should be just above the line with these `count` values, and these `pm` values are the ones put into the Values fields.

tonhuisman, 2021-07-27:
Fixed the issue mentioned above, for as far as I can see, without the actual hardware available 😊 

New download:
[ESPEasy_normal_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6888284/ESPEasy_normal_ESP8266_1M.zip)

franki29, 2021-07-28:
Hi, now it is working, thank you very much for fixing the issue
Br.Frank

PR #3707 (2021-07-31): [P053] Add support for PMS5003ST, PMS2003 and PMS3003
- Add support for the extra measurement values available in the PMS5003ST
- Add support for the slightly different data format of PMS2003 and PMS3003
- Add option to select the output values (only applicable for PMS5003ST)
- Add option to send event for values not simultaneously available for PMS5003ST

Resolves #1633
Resolves #2002

NB: No sensors available here, so testing should be done by others, please.
NB2: These extensions are disabled on 1M builds to avoid exceeding the max. build size for these modules.

TODO:
- [x] Testing (extern [... cut]
closed 2021-07-28 by franki29