letscontrolit/ESPEasy #2003: SONOFF POW PUYA 4MB

Wiki591, 2018-11-05:
Last week I have received my first SONOFF POW.

Flashing was only possible using the PUYA distribution. Unfortunately the PUYA distribution only exists for 1MB flash, the SONOFF POW I have got it equipped with 4MB flash.

Would it be possible to include a 4MB PUYA compilation in the release bot?

Would be nice concernig the OTA possibilities. I think, in not so far future there would be some more requests for this. I assume, that the standard of the POW was changed by the producer.

Wiki591, 2018-11-05:
Sure, here we go:
[image]
[image]

uzi18, 2018-11-05:
@Grovkillen my patch for auto puya detection and workaround is ready and working, tested by  Arduino guys, in proper issue, flawless victory, but still not merged into core.

Grovkillen, 2018-11-05:
@uzi18 thank you for that! No shame on you. I really looking forward ditching that special puya build.

uzi18, 2018-11-05:
with this patch we don't even need special build

TD-er, 2018-11-05:
@uzi18 We should indeed include it in the nightly build script to patch for all builds.
Just to save me the search for your patch, can you give a link?

uzi18, 2018-11-05:
I think so

Wiki591, 2018-11-05:
> 
> 
> is it possible. to. get foto of flash chip, it doesn't look like puya

I tried. No, my camera isn't able to take a readable picture closeup of the chip. Had a closer look to the chip (at first I had to find a strong lens). You're right, its marked with "HK 25032 6S36244". But indeed, flashing was only possible using the ESP8266PUYA_1024 build. Before I've tried 20181101_hard_SONOFF_POW and 20181101_hard_SONOFF_POW_R2_4M, both did not work, ending up in a boot loop.

To avoid any unnecessary efforts I will try normal_ESP8266_4096 later today, I will report.

Wiki591, 2018-11-05:
Eeeehm, I'm if I prodiced any inconvenience with my thread.

I've tried now flashing ESP_Easy_mega-20181101_normal_ESP8266_4096 - and it worked as expected. Then I've flashed ESP_Easy_mega-20181101_hard_SONOFF_POW_R2_4M to double-check my problems - and it works, too.

Summarized, what did I do (using builds 20181101): 
- Flashing of SONOFF POW 1024 - did'nt work, boot loop "Incorrect PID or version"
- Flashing of SONOFF POW R2 4M - didn't work, boot loop "Incorrect PID or version"
- Flashing PUYA 1024 - worked out of the box
- Flashing ESPEasy normal 4096 - worked out of the box
- Flashing SONOFF POW R2 4M - works now out of the box

Could anyone explain to me what happened?

TD-er, 2018-11-05:
What POW version do you have?
The R2 version is communicating with the serial port, so it is possible the chip is interacting with the flashing of the node.
After the first flash, you can perform OTA updates, which don't have these issues.

Grovkillen, 2018-11-05:
It's not an exact science. Sometimes it happens

Wiki591, 2018-11-05:
On the board there is a printed version# V2.0. But it was sold as lean SONOFF POW.

[quote]
After the first flash, you can perform OTA updates
[/quote]
Yes, I know. But what is confusing me: After first flash of which device ever ESPEasy is running through the formatting procedure for spiffs. Thats standard as you know. After flashing SONOFF POW R2 4M for the first time using the serial montoring I could see that the device was running through this procedure correctly, reporting the formatted blocks. But nevertheless ended up in the boot loop. And the puya build worked instantly.

2bontb, 2018-11-07:
I have a number of these POW 4M. I did add this to my platformio.ini to build it myself.

; SONOFF : 4096k version ----------------------------
[env:SONOFF_POW_4096]
platform                  = ${core_2_4_1.platform}
lib_deps                  = ${common.lib_deps}
lib_ignore                = ${common.lib_ignore}
lib_ldf_mode              = ${common.lib_ldf_mode}
lib_archive               = ${common.lib_archive}
framework                 = ${common.framework}
board                     = ${common.board}
upload_speed              = ${common.upload_speed}
monitor_speed             = ${common.monitor_speed}
board_build.f_cpu         = ${esp8266_4M.board_build.f_cpu}
board_build.flash_mode    = ${esp8266_4M.board_build.flash_mode}
build_unflags             = ${esp8266_4M.build_unflags}
build_flags               = ${esp8266_4M.build_flags} -D PLUGIN_SET_SONOFF_POW

TD-er, 2018-11-11:
Do one of the more recent builds work on this node?

Wiki591, 2018-11-11:
Well, it took plenty of time, here my results:

To reproduce the problem I've had with the build 20181101 I blanked the device completely (blank_4MB.bin). Afterwards I flashed the ESP_Easy_mega-20181101_hard_SONOFF_POW_R2_4M.bin. Same result as before, ended up in the well known bootl loop:

```
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 1
RESET: formatting...
RESET: formatting done...

Exception (28):
epc1=0x4022a1a7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000005 depc=0x00000000

ctx: cont
sp: 3ffffcb0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffe50:  00000000 00000001 00000000 4022a530
3ffffe60:  3fff1f28 0000018e 0000018e 4010020c
3ffffe70:  00000000 00000000 3ffffeb0 401006dc
3ffffe80:  00000000 00000000 00000000 402452c0
3ffffe90:  00000000 00000000 00000000 402452dc
3ffffea0:  00000000 00000000 00000000 4022a180
3ffffeb0:  00000000 00000000 00000000 00000000
3ffffec0:  00000000 00000001 3fffff20 00000000
3ffffed0:  00000000 00000000 3fffff20 3fffff20
3ffffee0:  00000000 00000001 00000000 4022a570
3ffffef0:  ffffffff 40288ee0 3fffff20 40245710
3fffff00:  00000000 3fff120c 00000001 00000000
3fffff10:  7fffffff 3fff120c 00000001 4022a5be
3fffff20:  3fff30e4 0000000f 00000007 00000000
3fffff30:  7fffffff 3fff120c 00000001 4022a916
3fffff40:  00000000 00000000 00000000 3fff44b4
3fffff50:  0000001f 0000000a 3ffe8780 3fff1e34
3fffff60:  0000003c 3fff160c 402463c4 3fffefb0
3fffff70:  4020311e 000003e8 000003e8 3fff1dcc
3fffff80:  3fff0efa 3fff1d08 3fff120c 40234fd5
3fffff90:  00000000 00000000 00000000 40247bc0
3fffffa0:  3fff38fc 0000006f 0000002f feefeffe
3fffffb0:  3fffdad0 00000000 3fff1dc2 4024644c
3fffffc0:  feefeffe feefeffe 3ffe8780 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
   ¦U
     PID:0
Version:0
INIT : Incorrect PID or version!
RESET: Resetting factory defaults...
```

I ran through the procedure I did before to get the device running. Means, directly afterwards I have flashed the PUYA version of 20181101, then the SONOFF POW 4M of 20181101. It worked as before, device up and running.

I blanked the device again and flashed directly the SONOFF POW 4M of 20181111. Result as before, boot loop:

```
RESET: formatting done...

Exception (28):
epc1=0x4022ae37 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000005 depc=0x00000000

ctx: cont
sp: 3ffffca0 end: 3fffffd0 offset: 01a0
[... 114 lines cut ...]
ESP board
ESP Chip ID:	8387766 (0x7FFCB6)
ESP Chip Freq:	80 MHz
ESP Board Name:	PLATFORMIO_ESP12E
Storage
Flash Chip ID:	Vendor: 0xE0 Device: 0x4016
Flash Chip Real Size:	4096 kB
Flash IDE Size:	4096 kB
Flash IDE speed:	40 MHz
Flash IDE mode:	DIO
Flash Writes:	0 daily / 0 boot
Sketch Size:	614 kB (2456 kB free)
SPIFFS Size:	934 kB (860 kB free)

```

I blanked the device again and flashed directly ESPEasy normal 20181111. It worked right out of the box, flashing the SONOFF POW 4M afterwrads worked, too.

????weird???

And now I wanted to know it exactly. I blanked the device once more again and flashed directly the ESP_Easy_mega-201811**01**_normal_ESP8266_4096.bin. And guess, what? The device is up and running.

Conclusion: there is no PUYA problem at all but a problem with the build for the SONOFF POW 4MB. And with the build of 20181101 as well as 20181111.

So I am sitting here awaiting the next testing tasks to get.......

TD-er, 2018-11-11:
What POW version do you have?

- hard_SONOFF_POW has partition layout for 1 MB flash
- hard_SONOFF_POW_R2_4M  has partition layout for 4 MB flash

Also the last one has some specific setup which will be activated directly after first boot, which is specific (!) for the POW R2, so it cannot be mixed with the POW (without "r2")
If you mix images for different partition layout, without performing a full clean flash, you are bound to have crashes.

Also I noticed the notification LED can also trigger crashes when set to the wrong pin (and sometimes regardless of pin)

So please tell me which POW version you have.

Wiki591, 2018-11-11:
Ofcourse I am aware of the different partition layouts. I don't know how to differ between POW and POW R2. The package says only "POW". On the board is printed V2.0 - these facts I have posted already in my first post opening this issue. So I assume its the POW R2. Definitely equipped with 4MB as shown in the sysinfo. Please just have a deep look at the informations I have collected and posted during really several hours of blanking and flashing and read them carefully - it will take you less time than my flashing, copying and posting. Thats what I have done these efforts for, hoping that someone might read and analyze the data.

I did not configure anything, I reported the states after flashing and directly after first reboot.

Wiki591, 2018-11-11:
Additionally: I posted already that the bin for POW without R2 - 1MB flash size - didn't work ether.

TD-er, 2018-11-27:
I just flashed the (new) `hard_SONOFF_POW_4M` build on my POW (not r2) and it is working fine.
So could you please try the build which is going to run tonight?

Nice new feature: 
via  Tools => Factory Reset 
You will get a new page with a combobox.
This allows you to set the button/relay/rules and status LED for the POW/POWr2. (allowed presets are enabled based on detected flash size)

After the new Factory reset you only have to select the right AC Energy sensor in the plugins and you're good to go.

2bontb, 2018-11-28:
You wrote: allowed presets are enabled based on detected flash size
How is the flas size used to determine enabled rules. Mind that there are POW R1 versions with 1M and 4M (I have only 4M versions), and I assume that POW R2 is only in 4M version.

TD-er, 2018-11-28:
I have never seen 1M POW versions.
So I just set it to only allow POW/POW-r2 for 4MB devices.
Do the 1M POW versions have the same chips as the 4M POW versions?