letscontrolit/ESPEasy #2007: Feature request: Disable SleepMode via MQTT

silmaril42, 2018-11-06:
I'd love to have an option to have some special MQTT topic that does the same as pulling GPIO-16 to GND.

Setting this value on the MQTT broker should make the ESP stay awake after it's next wakeup so it can be configured via HTTP. 

This way we could get rid of any need to physically get our hands on the device to change the configuration or make an OTA update, which could make things much easier (depending on where your devices are doing their work).

TD-er, 2018-11-06:
Changing configuration may also demand some level of security I think.

Grovkillen, 2018-11-06:
Can't we use `nosleep` ?

silmaril42, 2018-11-06:
> Changing configuration may also demand some level of security I think.

Good Point!
The data type of the MQTT topic could be a string that functions as some kind of password.
In the configuration you could enter a compare string and only if the topic value is exactly the same, it stops the device from going to sleep.

This way it's up to the users to decide what level of security they want.
One would just use "1" as a trigger, while someone else would generate 40 random characters for each device separately and store them securely in his password safe.

dkotte, 2018-12-29:
Would be great to allow changing the sleep settings (Sleep awake time & Sleep time) from within a rule.
Possible these settings are made within a device "sleepdevice", i can change sleep timers, disable sleep with "Sleep awake time=0" while changing this device-settings. (if no device exist, use system settings)
and with rulesengine we could add security to these settings ...
```
on MQTT-IN#mysleep do
    if [MQTT-IN#mysleep]=47110815
       sleepdevice,0,10   // deepSleep off
    endif
    if [MQTT-IN#mysleep]=47110816
       sleepdevice,120,10   // deepSleep on
   endif
endif
```

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er