letscontrolit/ESPEasy #2008: Cant connect to wifi

ymoona, 2018-11-06:
### Checklist
- [x ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request
I cannot connect to wifi

### Expected behavior
I can connect to wifi

### Actual behavior
Output of the terminal:
```
INIT : Booting version: mega-20181106 (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
109 : INIT : Warm boot #10 - Restart Reason: External System
111 : FS   : Mounting...
136 : FS   : Mount successful, used 75802 bytes of 957314
469 : CRC  : program checksum       ...OK
480 : CRC  : SecuritySettings CRC   ...OK
588 : INIT : Free RAM:29000
589 : INIT : I2C
589 : INIT : SPI not enabled
602 : INFO : Plugins: 46 [Normal] (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
604 : WIFI : Set WiFi to STA
636 : WIFI : Connecting Hans attempt #0
1971 : WD   : Uptime 0 ConnectFailures 0 FreeMem 27320
3496 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
4416 : WIFI : Connecting Hans attempt #1
>wifiscan
WIFI : SSID Scan start
WIFI : 8 networks found
WIFI : 1: Ziggo626194E 54:67:51:E4:53:51 Ch:1 (-83dBm) WPA2/PSK
WIFI : 2: FamilievanderPloeg F8:D1:11:4D:C2:44 Ch:1 (-82dBm) WPA/WPA2/PSK
WIFI : 3: HansBoven 38:2C:4A:69:14:28 Ch:5 (-74dBm) WPA2/PSK
WIFI : 4: UPC245144906 F8:04:2E:1C:3D:48 Ch:1 (-83dBm) WPA2/PSK
WIFI : 5: DIRECT-VM-BRAVIA 12:A0:96:F4:FC:0F Ch:10 (-67dBm) WPA2/PSK
WIFI : 6: Hans 00:11:32:50:3C:B7 Ch:10 (-74dBm) WPA2/PSK
WIFI : 7: UPC1840166 C4:27:95:86:EB:C2 Ch:11 (-68dBm) WPA/WPA2/PSK
WIFI : 8: Ziggo C6:27:95:86:EB:C4 Ch:11 (-69dBm)

Ok
```
I tried to get it connected to my ap, but it states it cant find my ap. But with a wifiscan it can find my AP (Hans)

### Steps to reproduce
Install the firmware and configure wifi

It happens all the time

### System configuration

Hardware:
NodeMCU from wemos (4mbit)

ESP Easy version: mega-20181106

ESP Easy settings/screenshots: 

### Rules or log data

```

```

ymoona, 2018-11-07:
Nope. And I'm not sure about casing on the ap name. But I matched that as well. 

I upgraded this node MCU from an older version, could it be that it is stuck on some old config? I already erased the flash and did a factory reset. Any thing else I can do?

Grovkillen, 2018-11-07:
You did flash it with a 4MB blank bin file?

Grovkillen, 2018-11-07:
Other ESP Easy units, do they connect?

ymoona, 2018-11-07:
Yes, but they are way older version.

Grovkillen, 2018-11-07:
Any you haven't tried another unit with the a new version? Please do!

svmac, 2018-11-07:
I have same issue. I've been trying to solve it for several days. It only happens with Wemos NodeMCU modules:
[image]

Log:
312 : EVENT: System#Wake
333 : WIFI : Set WiFi to STA
365 : WIFI : Connecting lada attempt #0
478 : EVENT: System#Boot
1832 : WD   : Uptime 0 ConnectFailures 0 FreeMem 26224

11276 : WIFI : Connecting lada attempt #1

26789 : EVENT: WiFi#Disconnected
26804 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 15 s
27276 : WIFI : Connecting lada1 attempt #2
30133 : EVENT: WiFi#Disconnected
30147 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2856 ms
30276 : WIFI : Connecting lada1 attempt #3
31832 : WD   : Uptime 1 ConnectFailures 0 FreeMem 26192
33133 : EVENT: WiFi#Disconnected
33148 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2857 ms
33276 : WIFI : Connecting lada attempt #4
36133 : EVENT: WiFi#Disconnected
36147 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2856 ms
36276 : WIFI : Connecting lada attempt #5
41027 : EVENT: WiFi#Disconnected
41044 : WIFI : Disconnected! Reason: '(4) Assoc expire' Connected for 4751 ms
41276 : WIFI : Set WiFi to AP+STA
42197 : WIFI : AP Mode ssid will be NodeMCU7_47 with address 192.168.4.1
42197 : WIFI : Connecting lada1 attempt #6
45055 : EVENT: WiFi#Disconnected
45075 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2856 ms
45277 : WIFI : AP Mode ssid will be NodeMCU7_47 with address 192.168.4.1
45277 : WIFI : Connecting lada1 attempt #7
48134 : EVENT: WiFi#Disconnected
48150 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2857 ms
48277 : WIFI : AP Mode ssid will be NodeMCU7_47 with address 192.168.4.1
48277 : WIFI : Connecting lada attempt #8

Wemos D1 mini and clone D1 mini modules work well.

Grovkillen, 2018-11-07:
I got that one pictured (at my office) I might be able to test it myself on Friday when I'm back.

TD-er, 2018-11-07:
Can you give the information from the sysinfo page?
Just to check if there is some difference in the nodes.

TD-er, 2018-11-07:
Oh and could you maybe try to set the wifi credentials using the (latest) version of @Grovkillen flash tool (included in the zip) and als try to set the IP settings to static. (in a second attempt)

svmac, 2018-11-07:
I've just upgrade to mega-20181107 (file ESP_Easy_mega-20181107_normal_ESP8266_4096.bin) and set the wifi credentials using the (latest) version of @Grovkillen flash tool (included in the zip) and now it's working.

Sysinfo page:

Unit: | 47
-- | --
Local Time: | 2018-11-07 16:15:12
Uptime: | 0 days 0 hours 5 minutes
Load: | 3.30% (LC=12452)
Free Mem: | 17320 (14984 - sendContentBlocking)
Free Stack: | 3584 (1728 - ExecuteCommand)
Boot: | Manual reboot (7)
Reset Reason: | External System
Network ❔
Wifi: | 802.11N (RSSI -49 dB)
IP config: | DHCP
IP / subnet: | 192.168.44.192 / 255.255.255.0
GW: | 192.168.44.1
Client IP: | 192.168.44.150
DNS: | 1.0.0.1 / 1.1.1.1
Allowed IP Range: | All Allowed
STA MAC: | 60:01:94:35:C8:BC
AP MAC: | 62:01:94:35:C8:BC
SSID: | lada (06:18:D6:C1:B0:03)
Channel: | 1
Connected: | 5 m 9 s
Last Disconnect Reason: | (1) Unspecified
Number reconnects: | 0
Firmware
Build:⋄ | 20102 - Mega
Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version:⋄ | mega-20181107
Plugins:⋄ | 46 [Normal]
Build Md5: | 9ea46e7428c174166f90298a7ed1a5f2
Md5 check: | passed.
Build time:⋄ | Nov 7 2018 03:33:40
Binary filename:⋄ | ESP_Easy_mega-20181107_normal_ESP8266_4096.bin
System Status
Syslog Log Level: | None
Serial Log Level: | Debug
Web Log Level: | None
ESP board
ESP Chip ID: | 3524796 (0x35C8BC)
ESP Chip Freq: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0xEF Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DIO
Flash Writes: | 2 daily / 2 boot
Sketch Size: | 744 kB (2324 kB free)
SPIFFS Size: | 934 kB (860 kB free)

Grovkillen, 2018-11-07:
So we can close this issue then? :+1:

svmac, 2018-11-07:
Downgrade to mega-20181106 version via OTA ==> WiFi don't work

svmac, 2018-11-07:
Flash ESP_Easy_mega-20181106_normal_ESP8266_4096.bin with ESP Easy Flasher ==> WiFi don't work

Grovkillen, 2018-11-07:
So yesterdays release is misbehaving. What have changed between those two releases? Not much?

Grovkillen, 2018-11-07:
But you changed it to work?

TD-er, 2018-11-07:
> But you changed it to work?

Yesterday I only merged the commits regarding OTA updates
So it should mainly be string related.

I hope I did proper search/replace queries, but maybe something may have changed, regarding wifi setup.

Grovkillen, 2018-11-07:
Today's release seems to work, yesterdays don't. Weird, build related perhaps.

TD-er, 2018-11-07:
Ah, maybe I fixed something too in the process ;)

svmac, 2018-11-07:
ESP_Easy_mega-20181107_normal_ESP8266_4096 flashed via OTA ==> WiFi works OK

svmac, 2018-11-07:
So today's release seems to work, but no. I've compiled from sources in the zip file and WiFi doesn't work again.
I've flashed my compiled version on D1 mini and WiFi works OK.

svmac, 2018-11-07:
No, I haven't. I use PlatformIO with VS code on Windows 10.
OK, I'm going to compile twice and try it. I tell you.

svmac, 2018-11-07:
Sorry, WiFi still doesn't work

Grovkillen, 2018-11-07:
How do you set the settings, serial or through AP?

svmac, 2018-11-07:
With your tool: ESP Easy Flasher

Grovkillen, 2018-11-07:
Ok, then that's not the problem 😎

svmac, 2018-11-07:
I've to disconnect now. I will connect again this night.

redskinhu, 2018-11-07:
Hi

I have couple of these modules and i have a same problem with them. No AP found, Beacon Timeout, Assoc expire errors, but the `scanwifi` serial command is presents my AP on the list.

Usually i update the firmware to the latest on the the modules when the new firmware is released.

If i update, some fw release are working fine some not. If the module can't connect i need to rollback to the previous fw. Sometimes I need to erase the module (blank_4MB.bin). Usually then i upload the 20180818 release which is always working fine. And after i upload the latest OK release.

I'm away from home but at the weekend i can test which fw is ok and which isn't. Maybe it helps to investigate and fix the problem.

Thx
T

TD-er, 2018-11-07:
Can someone make a detailed picture of the ESP chip?
I have had some discussion in the background via email and one reports the module has some "ECO Plug" label on it.
One of the suggestions was to add a number of `delay(1)` calls until the wifi connection is made.

If someone could make a really detailed image of the module (e.g. using a flatbed scanner, rotate module in several orientations compared to the scan direction), then please also make a detailed picture of the 3.3v and GND PCB traces to the ESP module and the voltage regulator.

ymoona, 2018-11-07:
I flashed the firmware of today: mega-20181107 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
First I flashed the blank and then the actual. Wifi works again :-)

I use this module as a temp sensor (DHT22)
The sensor works, but MQTT does not. Also the log page is and stays empty.

uzi18, 2018-11-07:
@ymoona what is your setting for weblog level - None?

ymoona, 2018-11-07:
Btw mqtt does work. Just not without credentials, probably a config issue on my broker. 

Logging is configured for debug but it shows nothing. But everything else works. So I am ok with closing the issue

TD-er, 2018-11-09:
If possible, can you try to power one of the problematic nodes with 5V from the **GND/Vin** pin (not the USB port) ?

TD-er, 2018-11-09:
I only removed the PUYA builds.
If that was the only 1 MB VCC build, then we could add some other for that.

redskinhu, 2018-11-10:
ESP_Easy_mega-20181110_test_ESP8266_4096_VCC.bin

>wifiscan
WIFI : SSID Scan start
WIFI : 4 networks found
WIFI : 1: eir90963965-2.4G 58:D7:59:5B:E0:70 Ch:2 (-82dBm) WPA/WPA2/PSK
WIFI : 2: TNCAPE8A29B C4:EA:1D:C3:BC:7F Ch:8 (-88dBm) WPA/WPA2/PSK
WIFI : 3: VM8BD6864 AC:22:05:79:96:58 Ch:11 (-87dBm) WPA/WPA2/PSK
WIFI : 4: Horizon Wi-Free AE:22:15:79:96:58 Ch:11 (-92dBm) 

Ok
215912 : Command: wifiscan
219603 : WIFI : AP Mode ssid will be Test21_WMD1_21 with address 192.168.4.1
219604 : WIFI : Connecting TNCAPE8A29B attempt #38
221806 : EVENT: WiFi#Disconnected
221811 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2199 ms

TD-er, 2018-11-10:
@redskinhu So it isn't fixed and the same nodes are still problematic and the ones that worked fine still work?

TD-er, 2018-11-11:
OK, that must have been the post which took the most preparation ever!
Between March and somewhere in May we had issues with the nightly builds and those builds resulted in lots of WiFi issues.
One of the most frustrating moments in development, since I was also working on the event based WiFi and the reports were totally not reproducible.

My initial thought on such random looking OK/NG reports is there must be some kind of string/char-array which is just allocated 1 byte too short, or not null-terminated.
It is then up to the compiler and the semi random behavior of the memory allocation to move around blocks, which may yield an extra unused byte after that faulty reserved block.

How long is the SSID of your network and how long your key?
Do they contain spaces or other "strange" characters?

redskinhu, 2018-11-11:
SSID: TNCAPE8A29B
Key: 9 char, 2 capital letters and numbers, nothing strange.

redskinhu, 2018-11-11:
Yes, I got the first modules when the event based WiFi was presented. And i had many "non working" modules...  I thought my new shiny modules are faulty and i spent lots of time to find the solution. And one thing about these modules. They are looks slower than the generic esp07/12 modules (ex: webpage loading). Switching between pages ~3-5s, on ESP07/12 ~1s. Maybe it helps.

TD-er, 2018-11-11:
It is always nice to try, but then you also take away a reproducible node to check if a future fix is working.

TD-er, 2018-11-11:
Maybe you can also see on the (new) timing stats page how long a write action takes.

Remember, the stats are reset when you load the page.
- load timing stats page
- perform some write (e.g. change some setting in a plugin)
- load timing stats page and check the max. time needed to save a file.

uzi18, 2018-11-11:
@redskinhu nice work, maybe this is related to poor antenna connection?
Try host AP with your mobile phone and connect to it.

redskinhu, 2018-11-12:
ESP_Easy_mega-20181112 tested:  NG

TD-er, 2018-11-12:
Still using the `test_ESP8266_4096_VCC.bin` build?

redskinhu, 2018-11-12:
yes

uzi18, 2018-11-12:
@redskinhu ERROR log please

uzi18, 2018-11-12:
@TD-er it is really hard to understand wifi connection/management code.
I could try to move all stuff to one place first (EspEasyWifi.ino) and next we can think how to modify this part of code to be more readable, what do you think?

redskinhu, 2018-11-12:
Hello

Small explanation about this Log:
- Log level: Debug dev
- `ESP_Easy_mega-20181107_test_ESP8266_4096_VCC.bin` was installed. It booting and connecting to the AP and MQTT publishing the IP and MAC .
- After I started the OTA upload to `ESP_Easy_mega-20181112_test_ESP8266_4096_VCC.bin`.
- Module reboots and tries to connect to the AP. After 10 retries I stopped the logging.

```
INIT : Booting version: mega-20181107 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LRuleDebug: 101: publish home/%sysname%/ipaddress,%ip%
WRuleDebug: 101: publish home/%sysname%/macaddress,%mac%
IP: 2.0.3)mode : sta(84:f3:eb:8e:61:8c)
add if0
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: time#initialized do

100 : INIT : Warm boot #3 - Restart Reason: Software/System restart
102 : FS   : Mounting.RuleDebug: 101: publish home/%sysname%/ipaddress,%ip%
..RuleDebug: 101: publish home/%sysname%/macaddress,%mac%

127 : FS   : Mount successful, used 76053 bytes of 957314
502 : CRC  : program checksum       ...OK
513 : CRC  : SecuritySettings CRC   ...OK 
616 : INIT : Free RAM:25384
617 : INIT : I2C
617 : INIT : SPI not enabled
652 : INFO : Plugins: 73 [Normal] [Testing] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
653 : EVENT: System#Wake
697 : WIFI : Set WiFi to STA
731 : WIFI : Connecting TNCAPE8A29B attempt #0
844 : EVENT: System#Boot
2196 : WD   : Uptime 0 ConnectFailures 0 FreeMem 23128
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 20
cnt 

connected with TNCAPE8A29B, channel 8
dhcp client start...
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: time#initialized do
5574 : WIFI : Connected! AP: TNCAPE8A29B (C4:EA:1D:C3:BC:7F) Ch: 8 Duration: 4842 ms
5575 : ERuleDebug: 101: publish home/%sysname%/ipaddress,%ip%
VENRuleDebug: 101: publish home/%sysname%/macaddress,%mac%
T: WiFi#ChangedAccesspoint
ip:192.168.1.247,mask:255.255.255.0,gw:192.168.1.1
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: time#initialized do
5691 : WIFI : DHCP IP: 192.168.1.247 (Test20-WMD1-20) GW: 192.168.1.1 SNRuleDebug: 101: publish home/%sysname%/ipaddress,%ip%
: 255.RuleDebug: 101: publish home/%sysname%/macaddress,%mac%
255.255.0   duration: 117 ms
5703 : EVENT: WiFi#Connected
5722 : Webserver: start
5723 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: time#initialized do
5803 : MQTT : Intentional reconnect
5805 : Read settings: ControllerSettings index: 0
5877 : MQRuleDebug: 101: publish home/%sysname%/ipaddress,%ip%
TT RuleDebug: 101: publish home/%sysname%/macaddress,%mac%
: Connected to broker with client ID: Test20_WMD1_20
5880 : Subscribed to: home/Test20_WMD1/#
5882 : EVENT: MQTT#Connected
[... 140 lines cut ...]
36494 : WIFI : AP Mode ssid will be Test20_WMD1_20 with address 192.168.4.1
36495 : WIFI : Connecting TNCAPE8A29B attempt #8
scandone
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: time#initialized do
39357 : EVENT: WiFi#Disconnected
RuleDebug: 101: publish home/%sysname%/ipaddress,%ip%
RuleDebug: 101: publish home/%sysname%/macaddress,%mac%
39369 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
39494 : WIFI : AP Mode ssid will be Test20_WMD1_20 with address 192.168.4.1
39495 : WIFI : Connecting TNCAPE8A29B attempt #9
scandone
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: time#initialized do
42357 : EVENT: WiFi#Disconnected
RuleDebug: 101: publish home/%sysname%/ipaddress,%ip%
RuleDebug: 101: publish home/%sysname%/macaddress,%mac%
42370 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
43048 : WD   : Uptime 1 ConnectFailures 0 FreeMem 16272
43495 : WIFI : AP Mode ssid will be Test20_WMD1_20 with address 192.168.4.1
43495 : WIFI : Connecting TNCAPE8A29B attempt #10
scandone
```

TD-er, 2018-11-13:
@uzi18 I agree the current wifi code has become too complicated (again)
So I agree we should have a look at it, when I'm back home (just arrived in Stockholm)

uzi18, 2018-11-13:
@redskinhu just ERROR please :)

redskinhu, 2018-11-13:
Same test method like before just the Log level is Error. As requested. :-)

It looks connected but it isn't.

```
>reboot

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
⸮U98 : 

INIT : Booting version: mega-20181107 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
99 : INIT : Warm boot #3 - Restart Reason: Software/System restart
101 : FS   : Mounting...
126 : FS   : Mount successful, used 76053 bytes of 957314
501 : CRC  : program checksum       ...OK
538 : CRC  : SecuritySettings CRC   ...OK 
640 : INIT : Free RAM:25384
641 : INIT : I2C
641 : INIT : SPI not enabled
676 : INFO : Plugins: 73 [Normal] [Testing] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
677 : EVENT: System#Wake
712 : WIFI : Set WiFi to STA
745 : WIFI : Connecting TNCAPE8A29B attempt #0
858 : EVENT: System#Boot
4800 : MQTT : Intentional reconnect

Update: ESP_Easy_mega-20181112_test_ESP8266_4096_VCC.bin
sleep disable
.............pm open,type:0 0
......................................................................................................................................................................................................................................................................................................................................................................................................................Update Success: 857648
Rebooting...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
@cp:0
ld
⸮U

>wifiscan
WIFI : SSID Scan start
WIFI : 3 networks found
WIFI : 1: eir90963965-2.4G 58:D7:59:5B:E0:70 Ch:2 (-74dBm) WPA/WPA2/PSK
WIFI : 2: eir_WiFi 58:D7:59:5B:E0:71 Ch:2 (-73dBm) open
WIFI : 3: TNCAPE8A29B C4:EA:1D:C3:BC:7F Ch:8 (-66dBm) WPA/WPA2/PSK
Ok
>wificonnect
Ok

redskinhu, 2018-11-21:
ESPEasy_mega-20181117 to 21 tested:  NG

TD-er, 2018-11-21:
I was just reading on several of these issues.
It appears Tasmota has some issues too with some modules when connecting to WiFi and their approach to solve it was to do not start reading sensors etc. until wifi connection was made and making lots of delay() calls to lower power consumption of the CPU.
this suggests the power supply of these modules may not be sufficient.

TD-er, 2018-11-21:
@redskinhu Could you test this PR: https://github.com/letscontrolit/ESPEasy/pull/2073

redskinhu, 2018-11-22:
Good Morning

Tested and NG. (the ESPEasy_mega-20181121 is behaving same like this FW)

I got this once:
```
Reply from 192.168.1.185: Destination host unreachable.
Reply from 192.168.1.247: bytes=32 time=2954ms TTL=255
Request timed out.
```
But the answer is most of the time: 

> Destination host unreachable.

See the log:
```
INIT : Booting version:  (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
100 : INIT : Warm boot #8 - Restart Reason: External System
102 : FS   : Mounting...
126 : FS   : Mount successful, used 76053 bytes of 957314
465 : CRC  : program checksum       ...OK
477 : CRC  : SecuritySettings CRC   ...OK 
478 : CRC  : binary has changed since last save of Settings
578 : INIT : Free RAM:24472
579 : INIT : I2C
579 : INIT : SPI not enabled
611 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
613 : EVENT: System#Wake
649 : WIFI : Set WiFi to STA
681 : WIFI : Connecting TNCAPE8A29B attempt #0
796 : EVENT: System#Boot
>debug 4

Ok
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 17
cnt 

connected with TNCAPE8A29B, channel 8
dhcp client start...
5519 : WIFI : Connected! AP: TNCAPE8A29B (C4:EA:1D:C3:BC:7F) Ch: 8 Duration: 4835 ms
5521 : EVENT: WiFi#ChangedAccesspoint

pm open,type:2 0

bcn_timout,ap_probe_send_start
32140 : WD   : Uptime 1 ConnectFailures 0 FreeMem 21312
ip:192.168.1.247,mask:255.255.255.0,gw:192.168.1.1
36085 : WIFI : DHCP IP: 192.168.1.247 (Test20-WMD1-20) GW: 192.168.1.1 SN: 255.255.255.0
36099 : EVENT: WiFi#Connected
36114 : Webserver: start
36114 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
36199 : MQTT : Intentional reconnect
36245 : MQTT : Connected to broker with client ID: ESPClient_84:F3:EB:8E:61:8C
36248 : Subscribed to: home/Test20_WMD1/#
36250 : EVENT: MQTT#Connected
36261 : EVENT: MQTT#Connected Processing time:11 milliSeconds
36585 : NTP  : NTP host 193.1.31.66 (193.1.31.66) queried
36617 : NTP  : NTP replied: delay 30 mSec Accuracy increased by 0.808 seconds
36619 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 60 minSTD time start: 2018-10-28 03:00:00 offset: 0 min
36621 : EVENT: Time#Initialized
36631 : ACT  : Publish home/Test20_WMD1/IPaddress,192.168.1.247
36643 : ACT  : Publish home/Test20_WMD1/MACaddress,84:F3:EB:8E:61:8C
36651 : EVENT: Time#Initialized Processing time:30 milliSeconds
36652 : EVENT: Clock#Time=Thu,09:07
36659 : EVENT: Clock#Time=Thu,09:07 Processing time:7 milliSeconds
36663 : Command: publish
36666 : Command: publish

ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
45099 : EVENT: WiFi#Disconnected
45107 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 39 s
45289 : MQTT : Connection lost, state: Connection lost
45289 : EVENT: MQTT#Disconnected
45587 : WIFI : Connecting TNCAPE8A29B attempt #0
scandone
48449 : EVENT: WiFi#Disconnected
48457 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
48586 : WIFI : Connecting TNCAPE8A29B attempt #1
50587 : EVENT: Clock#Time=Thu,09:08
scandone
51449 : EVENT: WiFi#Disconnected
51456 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2860 ms
51588 : WIFI : Connecting TNCAPE8A29B attempt #2
scandone
54449 : EVENT: WiFi#Disconnected
54457 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
54586 : WIFI : Connecting TNCAPE8A29B attempt #3
scandone
57443 : WIFI : Connecting TNCAPE8A29B attempt #4
57448 : EVENT: WiFi#Disconnected
57455 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2855 ms
scandone
```

redskinhu, 2018-11-22:
These modules are looks too slow to me.
I have a plan, i would like compare the speed of this module and one bare ESP12F.

[https://www.instructables.com/id/ESP8266-NodeMCU-CPU-Speed-Test/](url)

I'll test both modules whitout change the CPU freq.

Or any idea to compare the modules?

TD-er, 2018-11-22:
If the wifi reception is poor, a lot of re-transmits may occur and you would not see it.
Can you measure the 3.3V over the ESP when it is trying to connect to WiFi?
Preferably with a scope.
If that voltage is unstable, it will also affect the RF calibration performed right before connection is made.
This calibration takes a much higher current, so if the traces on the PCB are too thin, or the power supply itself isn't capable of handling this higher current, you may end up with unstable WiFi connection since the calibration is off.
Last few days I've been reading a lot on the WiFi on the ESP8266.

You could also try to add a small capacitor close to the 3.3V of the ESP, just to allow to keep this voltage stable on large variations of the current.
VCC and GND are on opposite sides of the ESP12e module:
[image]
Not sure what would be the best capacity for the capacitor.
Maybe 10 uF? You may also not want to make the capacitor too large, since it will appear as a "short" to the power supply when initially charged. So it may also prevent the ESP from booting when too large.

redskinhu, 2018-11-23:
Hi

I think this problem couldn't related to the voltages, because one fw is working fine and an another isn't on the same module which have same power supply and same location from the wifi AP and the test is happening almost same time.

Anyway the mega-20181122 is tested and OK.

What is the difference between 21 vs 22 fw?

I tried to do the test but it was unsuccessful, see: [https://github.com/letscontrolit/ESPEasy/issues/2076](url)

TD-er, 2018-11-23:
> Anyway the mega-20181122 is tested and OK.
> 
> What is the difference between 21 vs 22 fw?

The only difference is in a plugin.
So it has nothing to do with the wifi part or timing.
I guess if you do a factory reset (or blank flash) and try the 20181122 version again, it may fail.

redskinhu, 2018-11-23:
I think it was a fresh install.
I'll will try it when i get home.
Sadly I need to work... :(

redskinhu, 2018-11-23:
mega-20181122/23 uploaded to blank module: OK

[23 more replies]

closed 2019-10-11 by Grovkillen