letscontrolit/ESPEasy #2011: I cannot running embed_files.sh on git-bash for windows

workgroupengineering, 2018-11-07:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

When running embed_files.sh in git-bash for Windows, the script fails because hexdump command is missing.

### Expected behavior

no error reporting and valid data_h_temp

### Actual behavior

the bash report:  ``bash: hexdump: command not found`` and data_h_temp is empty:
```c++
const char DATA_DATA_H_TEMP_[] PROGMEM = {};
const char DATA_ESPEASY_DEFAULT_CSS[] PROGMEM = {};
const char DATA_ESPEASY_DEFAULT_MIN_CSS[] PROGMEM = {};
const char DATA_FAVICON_ICO_[] PROGMEM = {};
const char DATA_FETCH_AND_PARSE_LOG_JS[] PROGMEM = {};
const char DATA_GITHUB_CLIPBOARD_JS[] PROGMEM = {};
const char DATA_REBOOT_JS[] PROGMEM = {};
const char DATA_TIMED_REFRESH_JS[] PROGMEM = {};
const char DATA_UPDATE_SENSOR_VALUES_DEVICE_PAGE_JS[] PROGMEM = {};
```

### Steps to reproduce

1.  running embed_files.sh on git-bash for windows

### System configuration

Git for Windows 2.17.1
Hardware: **not applicable**

ESP Easy version: **mega-20181106**

TD-er, 2018-11-07:
What version of Bash on Windows do you use?
I installed the Ubuntu 18.04 version from the Windows Store

Since it is based on Ubuntu, you could try to install `bsdmainutils`

workgroupengineering, 2018-11-09:
Hi @TD-er ,

> What version of Bash on Windows do you use?
> I installed the Ubuntu 18.04 version from the Windows Store
> 
> Since it is based on Ubuntu, you could try to install `bsdmainutils`
I have not installed wsl and Ubuntu, I'm using MingW64 bash installed by Git for Windows 2.17.1.
I tried to find hexdump.exe for Windows to insert in C:\Program Files\Git\mingw64\bin, but I'm not successful.

TD-er, 2018-11-10:
I think this script should be replaced by something better.
- Does not append zero termination
- No feedback when minify fails
- sometimes incorrect minified results.

workgroupengineering, 2018-11-27:
I have installed Ubuntu 18.04 and bsdmainutils, but when run throw this errors:

```bash
we@PC2:/mnt/c/github/EspEasy$ ./embed_files
-bash: ./embed_files: No such file or directory
we@PC2:/mnt/c/github/EspEasy$ ./embed_files.sh
./embed_files.sh: line 2: $'\r': command not found
./embed_files.sh: line 11: $'\r': command not found
./embed_files.sh: line 13: $'\r': command not found
rm: cannot remove '/mnt/c/github/EspEasy/data_h_temp'$'\r\r': No such file or directory
./embed_files.sh: line 15: $'\r': command not found
./embed_files.sh: line 16: syntax error near unexpected token `$'{\r''
'/embed_files.sh: line 16: `function minify_html_css {
```

TD-er, 2018-11-27:
Hmm looks like the file contains Windows line endings.
Git should usually take care of that.
Looks like you are using the Windows files.
Simple (temporary) fix is to open the file in Notepad++ and just change the line endings to Unix line endings.

Usually I just perform a git checkout from Linux (on Linux filesystem, not the mount)

closed 2018-11-28 by workgroupengineering