letscontrolit/ESPEasy #2012: Opinion / Feature request: Wireless Network Mode selection b/g/n

Domosapiens, 2018-11-07:
Is it useful to add a selection for the preferred Wireless Network Mode?

From the ESP8266 datasheet:
[image]

One can expect more stable WiFi (stronger signals/longer range) at the cost of trough-put in b/g mode.
I think mode n is far over the top for IoT.

In many applications the ESP is cramped between other components in wall or housing.
Isn't 11Mbps enough at the advantage of much more Rx/Tx budget?

If this is useful, I request a 802.11 b/g/n/auto selection on the Config page.

Remark:
All my ESP nodes are n connected, even at the cost of signal strength.
My dd-wrt router has Wireless Network Mode "Mixed" but no possibilities to force per client the 802.11b/g. Do other routers have this?

TD-er, 2018-11-07:
I can look into the WiFi settings to see if I can set the mode.
But even so if you set the mode to B instead of N, you will run into network instability, due to the fact the access point has to change its mode back and forth between modes.
So it is better to have it connected via the same mode as others do, but only at a lower data rate.
Nodes will connect at different rates, depending on conditions.
When running at lower bandwidth (connection speed), the sensitivity is increased. Or to be more precise, the tolerance for faults due to bad reception will increase.

Having a mixed network (B/G/N) will cause more issues regarding stability (and speed for other nodes), but that also depends on the model/brand of the used access point.

podaen, 2022-10-08:
I discovered that the protocol can only be set to const uint8_t protocol = WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G || WIFI_PROTOCOL_11N; OR const uint8_t protocol = WIFI_PROTOCOL_11B OR const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0; anything else fails.

```
#include "esp_wifi.h"
#include "esp_log.h"

esp_err_t err;

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	err = esp_wifi_init(&cfg);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set wifi power mode!");
		return false;
	}

	err = esp_wifi_set_mode(WIFI_MODE_STA);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set sta!");
		return false;
	}

	//const uint8_t protocol = WIFI_PROTOCOL_11G;
	//const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0;
	const uint8_t protocol = WIFI_PROTOCOL_11B;

	err = esp_wifi_set_protocol(WIFI_IF_STA, protocol);
	if (err != ESP_OK) {
		//258 = ESP_ERR_INVALID_ARG
		Serial.println(err);
		Serial.println("Could not set protocol!");
		return false;
	}

	uint8_t getprotocol;
	//esp_err_t err;
	err = esp_wifi_get_protocol(WIFI_IF_STA, &getprotocol);
	if (err != ESP_OK) {
		Serial.println("Could not get protocol!");
		//log_e("Could not get protocol! %d", err);
		return false;
	}
	if (getprotocol & WIFI_PROTOCOL_11N) {
		Serial.println("WiFi_Protocol_11n");
	}
	if (getprotocol & WIFI_PROTOCOL_11G) {
		Serial.println("WiFi_Protocol_11g");
	}
	if (getprotocol & WIFI_PROTOCOL_11B) {//worst, but dificuld to set else
		Serial.println("WiFi_Protocol_11b");
	}
```

TD-er, 2022-10-08:
Looks like ESP32 code.
For ESP8266 I have been working on these issues last week.  
Connecting using 802.11n only seems to work on specific SDK versions.

I will take this code you posted to see if I can add this to the ESP32 code for ESPEasy too.
I already added code to detect which modes are advertised by the AP, so there is no need to try unsupported modes.

podaen, 2022-10-08:
I still have the old ESP32. Do you have an ESP32-S3 to test it with? I think it might work...

TD-er, 2022-10-08:
ESP32-S3 is not yet supported by ESPEasy.

podaen, 2022-10-08:
You can take my code. Known that the `wifi_config_t `is not in there yet.

TD-er, 2022-10-08:
The `esp_wifi_get_protocol` was already used in ESPEasy:

https://github.com/letscontrolit/ESPEasy/blob/d632f45a41a898b95f4e143a660faf31757438a8/src/src/ESPEasyCore/ESPEasyWifi.cpp#L748-L775

The code for setting the b/g/n config for ESP32 was also already present:
https://github.com/letscontrolit/ESPEasy/blob/d632f45a41a898b95f4e143a660faf31757438a8/src/src/ESPEasyCore/ESPEasyWifi.cpp#L1309-L1326
But commented out as it wasn't working very well, which probably got to do with what you reported that not all combinations are considered valid.

So I will have a look at that part too and also check against the (newly added) info whether an AP advertises some mode so we don't need to attempt to connect to it when not supported.

See also my pending PR regarding fixes for WiFi: https://github.com/letscontrolit/ESPEasy/pull/4285
Would really appreciate it if you could test this PR too.
See here for test builds: https://github.com/letscontrolit/ESPEasy/actions/runs/3211994332

TD-er, 2023-05-11:
This is now working well on the newly supported platforms like ESP32-S2/S3/C3
I also updated the sensitivity values based on the active connection mode and use those for the automatic TX power setting.

closed 2023-05-11 by TD-er