letscontrolit/ESPEasy #2018: HTTP responses have duplicate Content-Types in the header

Drehwaehler, 2018-11-07:
I have tested the Release mega-20181107. Great progress!
I've used the precompiled ESP_Easy_mega-20181107_dev_ESP8285_1024.bin. I also compiled myself with platformio, but with the same result.
I am using "Generic HTTP" as controller.

There are some issues with the HTTP header:
a) The Content-Type is duplicate
b) The Content-Type application/json is not always inserted there it should be.

1) When is send a Command, like <IP-addr>/control?cmd=status,GPIO,12, the response in the body is in json style.
But the header of the response is encoded this way:
HTTP/1.1 200 OK\r\n
    Content-Type: text/html\r\n
    Content-Type: text/html\r\n
    Cache-Control: no-cache\r\n
    Accept-Ranges: none\r\n
    Transfer-Encoding: chunked\r\n
    Connection: close\r\n
    \r\n
In earlier versions (January 2018), there was only one Content-Type: application/json\r\n. This is the expected value.

2) This duplicate Content-Type text/html is also found on nearly all pages of the webserver:
/, /config, /controllers, etc. 

3) There are two exceptions on these pages:
/logjson and /json. On these pages, there is a different duplicate Content-Type:
Hypertext Transfer Protocol
    HTTP/1.1 200 OK\r\n
    Content-Type: text/html\r\n
    Content-Type: application/json\r\n
    Cache-Control: no-cache\r\n
    Access-Control-Allow-Origin: *\r\n
    Accept-Ranges: none\r\n
    Transfer-Encoding: chunked\r\n
    Connection: close\r\n
    \r\n

There should be only  one Content-Type: application/json\r\n.

uzi18, 2018-11-08:
@Drehwaehler please test with PR #2022

Drehwaehler, 2018-11-08:
Tested with the patch PR #2022, but no cange to the behaviour. Still duplicate text/html for the response to commands.
What it looks like to me:
The first line Content-Type text/html is everywhere the same and might come from the central library of the webserver.
The second line Content-Type is application/json for the pages /json and /logjson. So the Webserver.ino code can influence the content of this line.
The response to commands with both times text/html: It looks like the bool json is not passed with the correct value to sendHeaderBlocking. This is the reason for the second Content-Type being text/html.

uzi18, 2018-11-09:
@Drehwaehler  should be resolved by #2022

Drehwaehler, 2018-11-09:
Tested it with the latest patch: 
1) The duplicate Content-Type pronlem is resolved! That is very good.
2) The pages /json and /logjson have the correct Content-Type application/json
But there is one remaining problem:
3) The Content-Type for responses to commands like /control?cmd=status,GPIO,12 is still text/html and not application/json. The bool json in sendHeaderBlocking might be not set correctly in this case?

TD-er, 2018-11-09:
That's possible the json bool has been forgotten in a number of places.
There is also a number of pages which are not using the `TXBuffer` chunked way of communicating.
This might be one of those pages.

uzi18, 2018-11-09:
@Drehwaehler ready to another test?

uzi18, 2018-11-09:
@TD-er it is ok with json bool

Drehwaehler, 2018-11-10:
Sorry for late reply:
All test results are now as expected!
Responds to commands have the correct Content-Type application/json.
All other Webpages tested have also the correct Content-Type:
/json and /logjson have application/json.
Other Webpages have text/html.
I have no remaining issues here.

TD-er, 2018-11-10:
Great!
So we can close this issue ;)

closed 2018-11-10 by TD-er