letscontrolit/ESPEasy #2020: _P124_Ventus_W266_RFM69.ino modify to work with current mega

sisamiwe, 2018-11-07:
@chunter1 

It seem that there was another change on the ESPEasy, so the current _P124_Ventus_W266_RFM69.ino does not work with it.

The RF signal gets received. You can see that in the log. But the devices are not updated anymore.

Could you have a check?
THX

jimmys01, 2018-11-07:
I am super interested in espeasy supporting rfm69

uzi18, 2018-11-07:
@sisamiwe what is the idea behind this plugin?

LeeNX, 2018-11-08:
And the stars align!!  
  
I was thinking about asking about supporting RF modules with ESPEasy, thou in many ways might seem out of scope, then I see this issue.  
  
I have a few Accurite Weather Stations, which I hook up with an Arduino, but need to connect to a server, but thinking ESPEasy and splitting something like this plugin into different support devices?  
  
Not urgent, just putting up my hand, in the mean time, my RFLink http://www.rflink.nl/blog2/ get's me by.

chunter1, 2018-11-08:
> 
> 
> @sisamiwe what is the idea behind this plugin?

I think the idea behind this plugin is clear... receive the Ventus W266 weather station's outdoor unit. ;)
I'll take a look at the code as soon as possible. - maybe tonight

chunter1, 2018-11-08:
I could track the problem down to the missing "PLUGIN_READ" trigger.
However, i don't know why "schedule_task_device_timer(event->TaskIndex, millis() + 100);" does not seem to start a timer to trigger "PLUGIN_READ" anymore?

jimmys01, 2018-11-09:
I have deployed sensors that are already sending measurments for over a year with 3 AA with the combo of arduino and rmf69. I would love to have the esp as a gateway between rfm69 and mqtt. I know that this will be huge for poeple that need to deploy sensors that only ocasionaly sent one or two mesaurments every 5 or 15 minutes and need batery power. Such as the weather stations mentioned above, or greenhouses, or for rooftop boiler tanks etc

chunter1, 2018-11-09:
Updated version is now available on github.
Please test and report issues. :)

jimmys01, 2018-11-09:
My thoughts are that the P124 is specific to the Ventus. Perhaps a 'general' plugin for the RFM69 chip should be made and the specific ones to be implemented as plugins for the RFM69 plugin. Also the RFM69 can act as a 'controller' and send data not only receive.

TD-er, 2018-11-09:
@jimmys01 How do you see this as "generic" plugin?
Using that RFM69 is too specific.
And if it should be a generic interface, it is not really a plugin but more something like an input controller.

sisamiwe, 2018-11-09:
@chunter1 
I've tested the updated plugin with mege-pre-release 2018-11-07 and it works like it should.
Great!
Thanks for the short response

jimmys01, 2018-11-09:
@TD-er If you have a different use of the rfm69 than the venus one that this plug in addresses, you have to make an new plug in altogether. All the code similar that this plug in uses will be dublicated.

TD-er, 2018-11-09:
If there will be more plugins, we can then see what code is the same and what not.
Then we can split the shared code.
Until then we simply don't know yet what code needs to be shared.

uzi18, 2018-11-09:
@TD-er we can split 1wire communication -> iButton plugin and Dallas are almost same or extend Dallas with iButton support.

TD-er, 2018-11-11:
@chunter1 
> Updated version is now available on github.
> Please test and report issues. :)

Maybe you can also make a PR to the playground for this update?

Is there any other reason to keep this issue open?

TD-er, 2018-11-12:
You can also mail me the source, or attach a txt file with it in this issue.

Grovkillen, 2019-10-11:
I will close this, open if its still a valid issue.

closed 2019-10-11 by Grovkillen

TheItschi, 2020-05-16:
It looks like all the new releases have the same issue again:  

> @chunter1
> I've tested the updated plugin with mega-pre-release 2018-11-07 and it works like it should.
> Great!
> Thanks for the short response

Hi @chunter1, I'm struggeling with the same issue trying to use recent releases. Data ist received but  no update of the devices. Would it be possible to do the same fix again? Regards, Itschi