letscontrolit/ESPEasy #2026: Setting default protocol in compile time starts disabled

jimmys01, 2018-11-08:
### Summarize of the problem/feature request

Compiling with the option
`#define DEFAULT_PROTOCOL 3 // Protocol used for controller communications` in globals.h
set to any number but 0 works but the condroler is not started enabled
### Expected behavior
Start the controller enabled or add a compile time option to enable the controller
Also perhaps add compile time defaults for controller password and username?

jimmys01, 2018-11-08:
For testing purposes, you want to reset the device but not to have to configure everything from scratch. Also for completion since most of the parameters are available to be changed in comple time, why leave some of them out.

uzi18, 2018-11-08:
You can save configuration and copy to another module

jimmys01, 2018-11-08:
Obviously this a minimum priority feature to be implemented

closed 2023-09-04 by TD-er