letscontrolit/ESPEasy #2027: Ir Receive results should be able to transmited back to a condroller

jimmys01, 2018-11-08:
So far the IR receive is able to read IR messages but does nothing with them, does not send them in a condroller and as far as i know they cant be used in the rules. Perhaps there is a way to send bac to the condrollers as mqtt messages the output of the IR receiver rather than just sending it to the log.

I propose for 3 strings to be transmited
The first one is the IR code that was received, like "NEC,566C8E" and the second is the IR2 code that was received if the encoding was sucessful. Perhaps also a third string witch will be the extended description given by the Ir library in case the IR code is comming from an AC unit

Examples:
First String: "GREE,0800205014200050" or "Unknown" if the decoding was unsucessfull
Second String: IR2, or "IR2,Unknown" if the decoding was unsucessfull
Thired String: "Power: Off, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Turbo: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Manual"
or nothing if this was not a message from an AC

Hope this makes sence

TD-er, 2018-11-08:
It makes sense, but it may be quite an extensive request to implement.
That's why I asked you to create an issue for it.
At this moment we do not have support for strings as element to transfer between plugins and controllers.

ghtester, 2018-11-09:
@jimmys01 You are wrong because IR input can be used easily in rules. For instance:

On Rules#Timer=1 do

if [IR#input]=3772833823
   gpio,12,1
 else 
     if [IR#input]=3772829743
         gpio,12,0
     endif
endif
timerSet,1,2
endon

jimmys01, 2019-07-15:
@TD-er any considerations about this feature request?

TD-er, 2019-07-15:
At this moment I am making a plugin to interact with the serial port and dump that value to a controller (MQTT at first)
This will be the first (initial) plugin to use binary data as input/output.
Using this one, we may find out what is needed for handling the non-float values and what obstacles we may encounter.

I do need something like this for another project also, which means it will be done in quite a short timeframe, and since it is a long awaited feature request to handle data/strings, I thought we just should start making it instead of pushing it forward.

As soon as this plugin is capable of sending strings to a controller, then there is no reason not to include the IRRX like plugin to do the same.

jimmys01, 2019-10-06:
fyi #2604

closed 2019-11-07 by jimmys01