letscontrolit/ESPEasy #2028: SONOFF BASIC / T1 goes into loop switching on and off

Lahim26, 2018-11-08:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

I do not understand, why it hapens only when rules "lamp_on" and "lamp_off" are active. If I delete them, there is no problem.
Also, the log looks like the action takes time twice

### Expected behavior

Should just switch on or off the light.

By the way 
- the newest firmware (mega-20181108) do not change the switch value - it is still "1", while for example 20181011 does it.
- there is nothing in Log (in the Tools -> Log). The field remains empty.

### Actual behavior

Sometimes, usualy wife and children have this problem - they push the switch and it goes into the loop.

### Steps to reproduce

1. push the switch fast twice.
2. 
3. 

### System configuration

Hardware:
SONOFF TOUCH and SONOFF T1
Build:⋄
20102  - Mega
Libraries:⋄
ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
GIT version:⋄
mega-20181108
Plugins:⋄
46 [Normal]
Build time:⋄
Nov  8 2018 03:22:27
Binary filename:⋄
ESP_Easy_mega-20181108_normal_ESP8266_1024.bin

ESP Easy version: 
ESP_Easy_mega-20181108_normal_ESP8266_1024.bin
ESP_Easy_mega-20181011_normal_ESP8266_1024.bin
but also on older firmware like Esp Easy 148

ESP Easy settings/screenshots: 
[image]
[image]

### Rules or log data

Rules
```
on lamp_on do
    gpio,12,1
    gpio,13,1
endon

on lamp_off do
    gpio,12,0
    gpio,13,0
endon

on SW#Switch do
  if [SW#Switch]=1
    gpio,12,1
    gpio,13,1
  else
    gpio,12,0
    gpio,13,0
  endif
endon
```
Log
```
1180575: SW : Switch state 1 Output value 1
1180580: EVENT: sw#Switch=1.00
1180602: ACT : gpio,12,1
1180604: SW : GPIO 12 Set to 1
1180608: ACT : gpio,13,1
1180609: SW : GPIO 13 Set to 1
1180627: SW : Switch state 1 Output value 1
1180631: EVENT: relay#Switch=1.00
1180977: EVENT: lamp_on
1180984: ACT : gpio,12,1
1180986: SW : GPIO 12 Set to 1
1180989: ACT : gpio,13,1
1180991: SW : GPIO 13 Set to 1
```

uzi18, 2018-11-08:
it is actuall behavior of rules, they are little. bit delayed.

v-a-d-e-r, 2018-11-08:
set internal pullup and push button active low, delete task relay and take this rule:

on SW#Switch do
if [SW#Switch]=1
gpio,12,1
gpio,13,0
else
gpio,12,0
gpio,13,1
endif
endon

v-a-d-e-r, 2018-11-08:
1st: set internal pullup and push button active low, delete task relay (if only local use) and take this rule:

on SW#Switch do
if [SW#Switch]=1
gpio,12,1
gpio,13,0
else
gpio,12,0
gpio,13,1
endif
endon

Lahim26, 2018-11-08:
It is used in/by Domoticz as well.
Without "lamp_on/off" it does not happens. 
I will try pull up for relay?

v-a-d-e-r, 2018-11-09:
No, the relay need no pull-up, because the GPIO for it is an output.

TD-er, 2018-11-11:
Is this still an issue?

Lahim26, 2018-11-13:
So I added pull up and Push button active low - did not change anything.
@TD-er - yes, it is problem. Also I am not able to use it with doubleclick or longpress unfortunatelly.

closed 2019-10-11 by Grovkillen