letscontrolit/ESPEasy #2029: Possible bug/misconception: Values without name are submitted by controller

Tom-Bom-badil, 2018-11-08:
Szenario: 2 devices
- BME280 measuring Temp/Hum/Pres
- Generic Dummy Device name 'Calculations' with one defined value 'Dewpoint' (getting its data from a rule) AND 3 unused values. The data are sent to smarthomeNG system through a Generic UDP controller.

All working well since weeks, but:
smarthomeNG log has weird errors every now and then. Found out that the 3 undefined values (=without name) are transferred through UDP controller as well, along with the one used devices data. This is causing the errors, as the values don't even have a name. So in fact the controller sends one after the other:

- Calculations.Dewpoint  12.9   (#1)
- Calculations."" (empty)   0.0   (#2 / undefined)
- Calculations."" (empty)   0.0   (#3 / undefined)
- Calculations."" (empty)   0.0   (#4 / undefined)

I doubt this is inteded behaviour, as those values dont even have a name enabling the destination system to properly identify it, nor a 'proper' value in it. Suggestion is to check if value has a name defined, and if not to skip it when sent to controller.

/tom

TD-er, 2018-11-09:
The controllers send as many values as exist in the plugin.
I guess we should add a selection option to mark which variable to send, at least for the dummy variable.

Since a few days, the sysinfo plugin may have similar issues. It now supports up-to 4 variables.

Tom-Bom-badil, 2018-11-09:
Thanks TD! Just wondering - wouldnt it be easier and less code  / less coding required to just check if an item has a 'name'? The mentioned selection option sounds tempting, but still doesn't remove the root cause of the problem. I am not an expert in other controllers, but I'd expect most systems like e.g. MQTT/HTTP to fail or report errors when 3 identical items with 'empty' names arrive ...

TD-er, 2018-11-09:
We have several different controllers, each with their own perks.
But maybe the simplest way could be to add some settings to the controllers when to ignore values.
- For invalid values (NaN)
- empty names

For example Domoticz controller is also a bit tricky.
For some sensor type of data it really needs some values to be at a specific position or else it will not be able to handle/interpret the values. Domoticz needs all values at once in the same message, indicating a single IDX value to signal its destination.

thomastech, 2018-11-11:
> But maybe the simplest way could be to add some settings to the controllers when to ignore values.
    For invalid values (NaN)
    empty names

@TD-er: +1. I just installed the latest firmware and was surprised to see the "empty" device values being displayed.  It would be great if these were hidden from the web interface's device list.  This would reduce the screen clutter.

Here's an example screenshot that shows this issue at each System Info device:
[image]

TD-er, 2018-11-11:
The system info plugin needs to check for "unset" and lower the number of items of that plugin.

thomastech, 2018-11-11:
FWIW, the three newly added values in the system info is a nice touch. But masking any empty (unnamed) values from the device list would be great.

I noticed that the "empty" values are all defaulting to the Uptime minutes. Perhaps that's not the best behavior, maybe zero or some other fixed value would be better. 

Attached is a new screenshot demonstrates that the empty values are copies of the Uptime minutes (confirmed by reviewing the Uptime data in the Main tab:
[image]

- Thomas

Grovkillen, 2018-11-11:
I'm working on the next generation GUI of ESP Easy and that sort of things will be fixed through that update. @TD-er we would still want those values to be in the JSON so a fix of the current GUI would have to take place in the browser?

thomastech, 2018-11-11:
Probably known to you developer guys, but with a little research I found  that the System Uptime was chosen as the default. So that explains why it appears in the empty value fields.
https://github.com/letscontrolit/ESPEasy/pull/2010

closed 2019-05-27 by Tom-Bom-badil