letscontrolit/ESPEasy #2031: Redesign the data structures for GPIO state monitoring and tracking

giig1967g, 2018-11-09:
Hi @TD-er ,
just to let you know what I am currently working on.
I am currently rewriting the code of PLUGIN 001, 009, 019 in order to remove the following fixed size data structures:
- 3X switchstate[TASKS_MAX]
- outputstate[TASKS_MAX]
- PinMonitor[GPIO_MAX] 
- PinMonitorState[GPIO_MAX]
- structure pinStates[32]

and change them with a dynamic single `std::map` structure.

This will reduce (1) memory usage, (2) will remove all the loops related to the "monitor" command that run 10 times per second, (3) will avoid synch issues among the device page/pin state page/actual state and (4) will allow to add the "monitor" command to plugin 009 and 019 that currently is unsupported and cannot be supported with the current logic.
Also, it will allow to add other plugins (maybe 013 and 021, but I have to check) to this data structure.

Finally it will allow to print in the log or to a JSON, with a single new command, the PIN status of all pins on request.

I know that that there are others threads talking about this feature, but I decided to open a new one to keep track of the progress.

TD-er, 2018-11-09:
Just one remark.
If storing bits, please use an unsigned int, or uint_32 like time and just use OR/XOR/AND to address bits.
std::map is a great container, but please don't use it for quick access to bits.
It is just too much overhead

giig1967g, 2018-11-09:
No it will not store bits.
"state" can have more than 2 values: -1,0,1.
So I need bytes.

here the std::map value structure:
```
struct portStatusStruct
{
  byte state; //current state (-1,0,1)
  byte output; //current output (-1,0,1) (for push buttons only)
  byte mode; //pin mode (undefined, input, output, servo, etc...)
  bool monitor; //true if port needs to be monitored and displayed in the pinstate page
  byte task; //number of devices/tasks in which the port is present in order to display in the pinstate page
  bool command; //true if port has been called from a command in order to display in the pinstate page
};
```

uzi18, 2018-11-09:
remember this struct as a part of settings is stored in file as is.

TD-er, 2018-11-09:
A few remarks:
- sort the members of the struct by size, largest first.
- Try to make some smaller members defining them in number of bits (sorting advice still applies)

In your suggested implementation, the struct takes 6 bytes. (you can check using `sizeof()` )
`state`, `output` only need 2 bits and both booleans need 1.
So that would fit in a single byte, which leaves 3 bytes for the struct.

Also I am not sure what `command` will be used for.

TD-er, 2018-11-09:
@uzi18 makes a valid point. 
I thought it would be data stored in memory.
If it is stored in settings, you will invalidate existing settings files.
Also a std::map stored values on the heap, so you cannot store a std::map in a settings file by simply taking the memory space occupied by the map object.

giig1967g, 2018-11-09:
It's not stored in Settings, of course. It's a global variable defined in 
It will have it's own functions to load,save etc.

Regarding the size of the struct, I would rather prefer to ptimize speed than space usage.
Consider that with `std::map` the number of elements in the map will be just the number of devices that have a task, as opposed to previous design where the struct size was fixed = to 12 for every plugin.
So grouping more variables together to save 1 byte but that will require more complexity to retrieve them in functions like 10_per_second, in my opinion is not the best choice.
We can afford few more bytes, if this reduce complexity and increase speed.

command: In current implementation, if GPIO1 is not part of any device, after sending for example a `gpio,1,1` command, GPIO1 will be visible in the pinstate page. So this flag is used to keep trace of the fact that gpio number 1 should be displayed in the pinstate page even if is not in any device page.
But I think that I can remove it and save more space.

TD-er, 2018-11-09:
No keep it, the command flag may be useful.

Also the complexity hardly increases when defining members like this `unsigned blah:2;`
If you sort them by size and add the packed attribute, it will be smaller and still be accessible like normal members.

TD-er, 2018-11-09:
If you define it as a union, it is easier to read.
But that's almost the same as the packed struct.
Using an uint32 as key sounds good to me.

You may also need to think about when to delete an item from the map.

uzi18, 2018-11-09:
ok so maybe not this struct ;)

TD-er, 2018-11-09:
Also try to add the get/set functions inside the struct. They do not add up to the sizeof the struct, but make it much more clear how to use it.
And a proper constructor initializing to proper default values.

giig1967g, 2018-11-09:
> If you define it as a union, it is easier to read.

can you show me what do you suggest for `union` in the declaration?

> Using an uint32 as key sounds good to me.
 
ok

> You may also need to think about when to delete an item from the map.

already thought about it. 
The item will be deleted if (task=0 && monitor=0 && command=0).
The logic is the following: 
in PLUGIN_INIT: task++
in PLUGIN_EXIT task--
in PLUGIN_SAVE: task-- on previous port and task++ on new port
in PLUGIN_WRITE: set command=1
in PLUGIN_WRITE (command= monitor): set monitor=1
in PLUGIN_WRITE (command= unmonitor [new command]): set monitor=0
whenever those values are changed if all of them are = 0 the map item is deleted.

TD-er, 2018-11-09:
What is the considered GPIO name for GPIO extenders?
Does this map have to be part of the plugins, or is it something more central?

giig1967g, 2018-11-09:
can this be the struct declaration?

```
struct name {
uint16_t state : 2;
uint16_t output : 2;
uint16_t mode : 2;
uint16_t task : 2;
uint16_t monitor : 1;
uint16_t command : 1;
uint16_t spare1 : 2;
uint16_t spare2 : 2;
uint16_t spare3 : 1;
uint16_t spare4 : 1;
}
```

Barracuda09, 2018-11-09:
@giig1967g 

Hello, using an union for type punning  in C++ is most likely "undefined behavior". So be careful!

see: https://stackoverflow.com/questions/11373203/accessing-inactive-union-member-and-undefined-behavior

giig1967g, 2018-11-09:
thanks for pointing out it.
I am still working in the final structure.
For now I am coding the logic.

TD-er, 2018-11-09:
Please also test with `sizeof` to see if the members are packed.
And if these are already 2 bytes, then there is no need to use an union.

giig1967g, 2018-11-13:
I have finished the development for the Switch plugin.
Everything works fine.
Will start changing the other plugins that use pinstate/switchstate structures.

uzi18, 2018-11-13:
@giig1967g is it possible for you to create PR with these changes so we can discuss it during development?

uzi18, 2018-11-13:
we will also need helper functions for compatibility with old api

giig1967g, 2018-11-13:
I have also added a new option in the Switch plugin: SafeButton.
When this feature is enabled, the EVENT of a switch change is created only if the switch remains in the new position for at least 2 cycles of the 10xsecond function.
This should resolve the cases where there are false positive switch events due to interferences.
I just finished coding this new feature and couldn't test it properly yet. So it may still contain some errors.

TD-er, 2018-11-14:
One remark for now....
Can you do comments in the code in English? ;)

giig1967g, 2018-11-14:
Absolutely.
It's still work in progress. I still have to clean a lot of debug messages and code.

giig1967g, 2018-11-14:
please see #2057

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er