letscontrolit/ESPEasy #2039: Request: _P059_Encoder.ino, push for reset

Domosapiens, 2018-11-11:
Use case:
Show current set-point Temp on LCD.
Rotate the encoder and show additional the new Temp.
Push to acknowledge the new set-point Temp and Zero the Rotary encoder.
Zeroing the encoder is needed for the next adjustment.

Problem:
Values are kept internally  and can not be zero'd
Tested on mega-20181101 normal, Wemos D1 mini.

Suggestion:
Cheap encoders like these: [https://www.letscontrolit.com/wiki/images/thumb/c/cb/Rotary_encoder_001.jpg/520px-Rotary_encoder_001.jpg](url)
do have a push button (active low, don't forget the Pull-Up)
The Encoder GUI has a third GPIO available for the Push button (currently used for encoders with "I" interface??)
Please add a checkbox "zero on push" (standard NOT checked for backward compatibility)
When checkbox is selected and Push is detected, zero the internal values.

Plugin originator @JK-de 
Last commits @TD-er #1989 and #1486 

Rather complicated to use rules for this, see:
[https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=6043&p=33034#p33034](url)

Domosapiens, 2018-11-13:
Thanks @TD-er 
Yes I understand that you need a shadow copy of the value before reset.

Could there be a difference between the SW pin of my cheap encoder and the I (=Index) pin of more professional ones?
[http://www.ni.com/white-paper/4763/en/](url):
> Some quadrature encoders also include a third output channel, called a zero or index or reference signal, which supplies a single pulse per revolution. This single pulse is used for precise determination of a reference position.

If I look at the code (that I hardly understand!):
```

pinMode(_pinA, INPUT_PULLUP);
	pinMode(_pinB, INPUT_PULLUP);
	pinMode(_pinI, INPUT_PULLUP);

	if (_pinA >= 0)
		attachInterrupt(digitalPinToInterrupt(_pinA), ISR, CHANGE);
	if (_pinB >= 0)
		attachInterrupt(digitalPinToInterrupt(_pinB), ISR, CHANGE);
}
```
So I see only an interrupt subscription for A and B signal.
Not for the _pinI !!!

````
if (_pinI >= 0 && _bIndexTrigger && bCounterChange && digitalRead(_pinI)) {   // is index pin triggered?

			_bIndexTrigger = false;
			_counter = 0;
                         DEB("I");
````

I understand from this that if the I pin is in use (=GPIO in GUI defined)  and _bIndexTrigger (where does that come from?) and a bCounterChange (position change) and _PinI must be high. ...
Then the counter is zero'd.

I see in the log a line
OEI GPIO 14,12,13 (that is correct)
OEI values ... correct
But no zero actions

Or .. only 1 time a zero action at task creation ???
Difficult to reproduce .. is _bIndexTrigger one time true at startup??

TD-er, 2018-11-14:
At plugin_init the `setIndexTrigger` is called, which enables the index trigger.
This init is only called when saving the settings or loading it (at boot for example).
After the index is found, its trigger will be disabled.

Domosapiens, 2018-11-14:
Sorry to bother you @TD-er 
> It looks like the "I" pin is already pulled up, so if you ground the pin, it will perform a reset.
I assume, that is what the switch already does?
So you can use the switch as reset, but then the value is lost.

It looks like the "I" pin is already pulled up, ....YES
so if you ground the pin, it will perform a reset. ....NO
I assume, that is what the switch already does? ....YES, cheap encoder SW is active LOW
So you can use the switch as reset, but then the value is lost. ....NO, value count goes on

Is this the correct conclusion?
- plugin works as expected ?
- cheap rotary encoder "SW" switch is not compatible with "I" pin ?
- further testing is not needed ?
- adaptation of plugin is needed for zero-ing values ?

TD-er, 2018-11-14:
Conclusion looks right in my opinion.
We can also add a command to the plugin to accept zero.
Then you can 'connect' a switch input via rules to that new command.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er