letscontrolit/ESPEasy #2046: Double queries on HTTP Generic

JG24, 2018-11-12:
I just put together ESP with BME280 and I have a problem with double queries to my webserver through HTTP Generic function. ESP Easy always send two (sometimes more) queries with the same results. How to avoid that?

```
meteo01 - - [11/Nov/2018:22:30:07 +0100] "GET /espeasy/index.php?name=Meteo01&task=bme280&n1=Temperature&v1=22.69&n2=Humidity&v2=35.71&n3=Pressure&v3=1020.79 HTTP/1.1" 200 170 "-" "ESP Easy/20102/Nov 11 2018 03:25:02"
meteo01 - - [11/Nov/2018:22:30:07 +0100] "GET /espeasy/index.php?name=Meteo01&task=bme280&n1=Temperature&v1=22.69&n2=Humidity&v2=35.71&n3=Pressure&v3=1020.79 HTTP/1.1" 200 170 "-" "ESP Easy/20102/Nov 11 2018 03:25:02"
```

Here's my config:
https://i.imgur.com/pSBj07d.png
https://i.imgur.com/sLXZpsb.png

ESP Easy Mega version: mega-20181111

jimmys01, 2018-11-12:
Perhaps related to #2022

TD-er, 2018-11-12:
You're defining you own http request including all values.
It is sending every value separate to the controller.

Can you give the whole publish line?
And I should add an option to set in this controller if it should send all values at once or per value.

JG24, 2018-11-12:
> Can you give the whole publish line?

Here's my line:
`espeasy/index.php?name=%sysname%&task=%tskname%&n1=%vname1%&v1=%val1%&n2=%vname2%&v2=%val2%&n3=%vname3%&v3=%val3%`

BME280 returns 3 values: temperature, humidity, pressure. Sending values at the same time is a great thing.

uzi18, 2018-11-12:
@TD-er maybe just check publish line for val3 and if found send max 3 values

tonhuisman, 2022-04-18:
This seems to have been resolved, so probably the issue can be closed.

closed 2022-04-18 by TD-er